zephyrproject-rtos/zephyr #72914: boards: st: `vbat` node enabled without enabling associated `adc` node

ycsin, 2024-05-17:
**Describe the bug**
The ST vbat driver expects the `adc` node to be enabled in the devicetree (due to the usage of `DEVICE_DT_GET`), but the `vbat` nodes in multiple boards are currently being enabled without also enabling its associated `adc` node

https://github.com/zephyrproject-rtos/zephyr/blob/adeb19b30cbfb3552cb2d19c6c6676a872afe6c0/drivers/sensor/st/stm32_vbat/stm32_vbat.c#L119-L123

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b nucleo_h753zi -p auto zephyr/samples/sensor/sensor_shell`

**Expected behavior**
`vbat` probably shouldn't be enabled if the `adc` that it is using isn't enabled

**Impact**
Causing CI to fail in https://github.com/zephyrproject-rtos/zephyr/pull/72833

**Logs and console output**
See 72910

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: 0.16.5-1
 - Commit SHA or Version used: `main` branch

**Additional context**
The proposed fix in #72911 only fixes the driver side of the issues, the `vbat` nodes are still being enabled in boards where it shouldnt

FRASTM, 2024-05-21:
@ycsin 
 Adding a `depends on DT_HAS_ST_STM32_ADC_ENABLED` condition in the Kconfig for the drivers/sensor/st/stm32_xxx 
will make the compilation successful but application is meaningless if no ADC node is enabled

FRASTM, 2024-05-21:
> > @ycsin Adding a `depends on DT_HAS_ST_STM32_ADC_ENABLED` condition in the Kconfig for the drivers/sensor/st/stm32_xxx will make the compilation successful but application is meaningless if no ADC node is enabled
> 
> The boards either fail to build the test in main branch's CI and fail, or becomes meaningless. I chose the latter as it seems less evil (and unblocks my PRs).
> 
> #72911 fixes half of the issue, the other half IMHO is to disable the vbat nodes in the board dts that are 'enabled' without an adc, or enable the adc properly with the correct configurations.
> 
> If there's usually only one vbat, maybe?:
> 
> ```c
> #if !IS_ENABLED(DT_NODE_HAS_STATUS(DT_INST_IO_CHANNELS_CTLR(0), okay))
> #warning "No ADC enabled for vbat 0"
> #endif
> ```

`#warning` or `#error "No ADC node enabled"`

PR #76180 (2024-08-01): STM32: handle disabled ADCs more cleanly in internal sensor drivers
This PR intends to fix #72914 and make detecting such mistakes easier.

As piggybacking changes, ba90cd530adf17cbfb4c290a1204c6d79473c87b also makes the `pinctrl-0` property of STM32 ADC no longer required; indeed, it is possible to enable the ADC only for usage of internal sensors - in this case, no GPIO pin needs to be switched to ANALOG alternate function.
closed 2024-08-01 by carlescufi