zephyrproject-rtos/zephyr #72957: MPU fault while trying to call symbol in LLEXT module

joelguittet, 2024-05-17:
Hello,

I'm working with nucleo_l4a6zg board from st.

I have build hello-world LLEXT example with the following command:
```
west build -d build-llext -b nucleo_l4a6zg tests/subsys/llext/simple/
```
and then retrieved `hello-world.llext` file in the build dir.

Then I have loaded the file on the board using the llext loader (not using the shell).

Then with the shell I would like to test it, so some simple commands on the uart interface:

```
uart:~$ llext list
| Name             | Size         |
|      hello-world |          949 |
```
Looks it's properly loaded 🎉 

```
uart:~$ llext list_symbols hello-world 
Extension: hello-world symbols
| Symbol           | Address    |
```
Seems there is no symbol 😔 

And finally:
```
uart:~$ llext call_fn hello-world test_entry
[00:00:58.481,000] <err> os: ***** MPU FAULT *****
[00:00:58.486,000] <err> os:   Instruction Access Violation
[00:00:58.493,000] <err> os: r0/a1:  0x20026241  r1/a2:  0x20004853  r2/a3:  0x00000000
[00:00:58.501,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0803dccf
[00:00:58.510,000] <err> os:  xpsr:  0x61000000
[00:00:58.515,000] <err> os: Faulting instruction address (r15/pc): 0x20026240
[00:00:58.523,000] <err> os: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
[00:00:58.531,000] <err> os: Current thread: 0x200010c0 (shell_uart)
[00:00:58.538,000] <err> os: Halting system
```

What could be the reason the `test_entry` function is not available ??

I activated LLEXT debug log level and I have the following logs when the module is loaded in my application:
```
[00:00:27.445,000] <dbg> llext: llext_load: Loading relocatable or shared elf
[00:00:27.455,000] <dbg> llext: do_llext_load: Finding ELF tables...
[00:00:27.464,000] <dbg> llext: llext_find_tables: section 0 at 2c4: name 0, type 0, flags 0, addr 0, size 0
[00:00:27.476,000] <dbg> llext: llext_find_tables: section 1 at 2ec: name 31, type 1, flags 6, addr 0, size 36
[00:00:27.489,000] <dbg> llext: llext_find_tables: section 2 at 314: name 27, type 9, flags 40, addr 0, size 24
[00:00:27.502,000] <dbg> llext: llext_find_tables: section 3 at 33c: name 37, type 1, flags 3, addr 0, size 0
[00:00:27.514,000] <dbg> llext: llext_find_tables: section 4 at 364: name 43, type 8, flags 3, addr 0, size 0
[00:00:27.527,000] <dbg> llext: llext_find_tables: section 5 at 38c: name 48, type 1, flags 2, addr 0, size 47
[00:00:27.539,000] <dbg> llext: llext_find_tables: section 6 at 3b4: name 60, type 1, flags 2, addr 0, size 8
[00:00:27.552,000] <dbg> llext: llext_find_tables: section 7 at 3dc: name 56, type 9, flags 40, addr 0, size 16
[00:00:27.564,000] <dbg> llext: llext_find_tables: section 8 at 404: name 74, type 1, flags 30, addr 0, size 33
[00:00:27.577,000] <dbg> llext: llext_find_tables: section 9 at 42c: name 83, type 1879048195, flags 0, addr 0, size 48
[00:00:27.590,000] <dbg> llext: llext_find_tables: section 10 at 454: name 1, type 2, flags 0, addr 0, size 272
[00:00:27.603,000] <dbg> llext: llext_find_tables: symtab at 10
[00:00:27.612,000] <dbg> llext: llext_find_tables: section 11 at 47c: name 9, type 3, flags 0, addr 0, size 65
[00:00:27.624,000] <dbg> llext: llext_find_tables: strtab at 11
[00:00:27.633,000] <dbg> llext: llext_find_tables: section 12 at 4a4: name 17, type 3, flags 0, addr 0, size 99
[00:00:27.645,000] <dbg> llext: llext_find_tables: shstrtab at 12
[00:00:27.654,000] <dbg> llext: do_llext_load: Allocate and copy strings...
[00:00:27.664,000] <dbg> llext: do_llext_load: Mapping ELF sections...
[00:00:27.673,000] <dbg> llext: llext_map_sections: section 0 name 
[00:00:27.682,000] <dbg> llext: llext_map_sections: Not copied section 
[00:00:27.691,000] <dbg> llext: llext_map_sections: section 1 name .text
[00:00:27.700,000] <dbg> llext: llext_map_sections: section 2 name .rel.text
[... 64 lines cut ...]
[00:00:28.433,000] <inf> llext: Loaded extension hello-world
```
This looks good and we can see the `test_entry`symbol is properly read.

I'm on top of the main branch + 1 modification to get the LLEXT shell building properly (looks strange nobody is reporting this isn't it ? I can do a PR):
```
diff --git a/subsys/llext/CMakeLists.txt b/subsys/llext/CMakeLists.txt
index b129dc7f943..68a9e83df08 100644
--- a/subsys/llext/CMakeLists.txt
+++ b/subsys/llext/CMakeLists.txt
@@ -1,5 +1,6 @@
 if(CONFIG_LLEXT)
   zephyr_library()
+  zephyr_library_compile_definitions(-D_POSIX_C_SOURCE=200809L)
   zephyr_library_sources(llext.c llext_export.c buf_loader.c)
   zephyr_library_sources_ifdef(CONFIG_LLEXT_SHELL shell.c)
 endif()
```

So at the end several questions:
- what is the proper way to build an LLEXT module ? I didn't find information in the docs.
- Why `test_entry` symbol is not found ? Could it be related to my platform ? I tested with `CONFIG_ARM_MPU=n` after discussion on discord, same result.
- What about the modification I had to do to build the LLEXT shell ?

Thanks for any advice!

joelguittet, 2024-05-19:
Additional informations.

In my code after loading the module with `llext_load`, I added the following:

```
void (*test_entry_fn)() = llext_find_sym(&ext->exp_tab, "test_entry");
LOG_INF("test_entry_fn='%p'", test_entry_fn);
if (NULL != test_entry_fn) {
    LOG_INF("calling test entry");
    test_entry_fn();
    LOG_INF("called ok");
}
```

Result is the following:

```
[00:00:28.434,000] <inf> mender_stm32l4a6_zephyr_example: test_entry_fn='0x20026241'
[00:00:28.445,000] <inf> mender_stm32l4a6_zephyr_example: calling test entry
[00:00:28.454,000] <err> os: ***** MPU FAULT *****
[00:00:28.462,000] <err> os:   Instruction Access Violation
[00:00:28.470,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000005  r2/a3:  0x20004248
[00:00:28.481,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0800f97b
[00:00:28.492,000] <err> os:  xpsr:  0x010e0000
[00:00:28.499,000] <err> os: Faulting instruction address (r15/pc): 0x20026240
[00:00:28.510,000] <err> os: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
[00:00:28.520,000] <err> os: Current thread: 0x20004248 (mender_rtos_work_queue)
[00:00:28.530,000] <err> os: Halting system
```

So the function is found but execution failed as seen previously in the shell.

Also done some basics, but not sure it helps:

```
(.venv) ubuntu@ubuntu-VirtualBox:~/zephyrproject/zephyr$ objdump -s build-llext/zephyr/hello_world.llext 

build-llext/zephyr/hello_world.llext:     file format elf32-little

Contents of section .text:
 0000 80b500af 0448054b 98472a23 19460448  .....H.K.G*#.F.H
 0010 024b9847 00bf80bd 04000000 00000000  .K.G............
 0020 14000000                             ....            
Contents of section .rodata:
 0000 2a000000 68656c6c 6f20776f 726c640a  *...hello world.
 0010 00000000 41206e75 6d626572 20697320  ....A number is 
 0020 25750a00 74657374 5f656e74 727900    %u..test_entry. 
Contents of section .exported_sym:
 0000 24000000 00000000                    $.......        
Contents of section .comment:
 0000 00474343 3a20285a 65706879 72205344  .GCC: (Zephyr SD
 0010 4b20302e 31362e31 29203132 2e322e30  K 0.16.1) 12.2.0
 0020 00                                   .               
Contents of section .ARM.attributes:
 0000 412f0000 00616561 62690001 25000000  A/...aeabi..%...
 0010 0537452d 4d00060d 074d0902 12041401  .7E-M....M......
 0020 15011703 18011901 1a011e06 22012601  ............".&.
```

```
(.venv) ubuntu@ubuntu-VirtualBox:~/zephyrproject/zephyr$ readelf -a build-llext/zephyr/hello_world.llext
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              REL (Relocatable file)
  Machine:                           ARM
[... 949 lines cut ...]
===================================================================
START - test_printk_exported
 PASS - test_printk_exported in 0.001 seconds
===================================================================
TESTSUITE llext succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [llext]: pass = 8, fail = 0, skip = 0, total = 8 duration = 3.858 seconds
 - PASS - [llext.test_ext_syscall_fail] duration = 0.001 seconds
 - PASS - [llext.test_load_unload_hello_world] duration = 0.403 seconds
 - PASS - [llext.test_load_unload_logging] duration = 0.683 seconds
 - PASS - [llext.test_load_unload_object] duration = 0.532 seconds
 - PASS - [llext.test_load_unload_relative_jump] duration = 0.879 seconds
 - PASS - [llext.test_load_unload_syscalls] duration = 0.429 seconds
 - PASS - [llext.test_load_unload_threads_kernel_objects] duration = 0.930 seconds
 - PASS - [llext.test_printk_exported] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

So looks it's something related maybe to my project configuration ?

joelguittet, 2024-05-21:
@teburd hello, I provided these information in the first message of this issue, with details.

joelguittet, 2024-05-22:
Redone a new test rebuilding and flashing my target with `CONFIG_ARM_MPU=n` and it works! 🎉 🎉 🎉 
Not sure why it wasn't before. So it appears this is a mandatory condition to get the LLEXT modules working, isn't it ?

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/llext/shell_loader/README.rst indicates: "You may need to disable memory protection for the sample to work (e.g. CONFIG_ARM_MPU=n)."
So when it is required to disable MPU and when is it possible to not disable MPU on the device ? This readme is not enough explicit, and moreover I think this is an information that should be indicated in LLEXT doc (https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/services/llext/index.rst).

If disabling MPU is a requirement to LLEXT, how is it possible to unselect `CONFIG_ARM_MPU` when `CONFIG_LLEXT` is enabled ? (what should be the proper ay to achieve it ? I can do a PR for that :-) )

teburd, 2024-05-22:
Its not required that CONFIG_ARM_MPU=n, but if you want to use the MPU and loadable extensions, we need to be able to setup memory region permissions. That today requires CONFIG_USERSPACE=y

So the options are

CONFIG_ARM_MPU=n || CONFIG_USERSPACE=y

Note that using userspace may not be enough, if you look at the test I setup a thread (both in supervisor mode and user mode) before calling into the extension with the regions setup. That's because some arm parts (nxp kinetis for example) don't have many mpu regions to work with, and the thread context to run from needs to have access to the .text of the extension today.

There's some other ways of solving the problem that I didn't explore because the tools to setup regions with permissions was already there with user space (memory domains and partitions)

joelguittet, 2024-05-22:
Okkkk good, thanks for the details @teburd !!!
Considering this issue is closed so, thanks for your attention on this.
Regards

closed 2024-05-22 by joelguittet