zephyrproject-rtos/zephyr #72959: drivers: video: API enhancement

josuah, 2024-05-17:
## Introduction

Pursue the work on the video API past the [Video 4 Zephyr](https://docs.google.com/presentation/d/1j44YHUqynN-Vw67NgHHx741yzLIbjbc76tcObi8J3Zc/edit#slide=id.p) added  to Zephyr in #17194.

### Problem description

I would like to implement UVC and enable more complex video pipelines to be written without having to rewrite from scratch a new application every time the devicetree is updated.

### Proposed change

Incrementally rework the devicetree and video driver API 

## Detailed RFC

The existing API leaves corners left for future specification, such as:

The goal is to offer:

- A way to fully access the devicetree video content from the application
- allow to effectively connect video elements together in code, like specified in the devicetree without having to do it manually.
- Define all semantics to cover the most common cases completely
- Reuse ideas and code widespread through Zephyr drivers

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

- `remote-endpoint` [#72311](https://github.com/zephyrproject-rtos/zephyr/issues/72311) which is not used by Zephyr drivers
  It is possible to interconnect video devices directly by leveraging RTIO, but needs to be investigated.

- [`video_endpoint_id`](https://docs.zephyrproject.org/latest/hardware/peripherals/video.html#c.video_endpoint_id) has unclear semantics in some cases
  - `VIDEO_EP_IN` and `VIDEO_EP_OUT` does not work with drivers with more than 2 INs or 2 OUTs port (such as a source selector).
  - `VIDEO_EP_ANY` is convenient to skip implementation details, but from the point of view of a driver, it is difficult to decide what to select. Maybe `ALL` is intended instead?
  - `VIDEO_EP_NONE` looks like a NO-OP, and not used in Zephyr source

- [`video_get_caps()`](https://docs.zephyrproject.org/latest/hardware/peripherals/video.html#group__video__interface_1ga4d5237607c858708380955705a2023bd) fills a struct with all the video caps filled at once.
  This works well, but makes it more difficult than using an enumerator-style API such as what [#72254](https://github.com/zephyrproject-rtos/zephyr/pull/72254) proposes which allows drivers to be a bit more generic: a device doing software video processing can filter the format capabilities of its sources this way.

- [`video_enqueue()`](https://docs.zephyrproject.org/latest/hardware/peripherals/video.html#group__video__interface_1gac19d14a5875d48c96bd66a8bb65e8a51): Quoting [#72827](https://github.com/zephyrproject-rtos/zephyr/pull/72827): "an addition to the video API to enable streaming partial frames within one video buffer" is missing. For instance, and "entire frame" could have both a flag "start of frame" and "end of frame" for the same buffer, and a partial frame would have the flag only if it starts or finishes a frame (then no special case for full frames).

- [`rtio`](https://github.com/zephyrproject-rtos/zephyr/issues/72311#issuecomment-2106400935) can be leveraged to replace the many FIFOs that are typically defined in each driver [1](https://github.com/zephyrproject-rtos/zephyr/pull/72827/files#diff-2b616db5ead9f883465ba69f30da1148d92469f3682c9f894ae324e991fb7555R199) [2](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/video/video_sw_generator.c#L192), [3](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/video/video_stm32_dcmi.c#L334) and provide a way to directly connect drivers while also allowing the application to be interleaved in the pipeline. This would act as a [splice(2)](https://linux.die.net/man/2/splice) for Zephyr video drivers, triggered by the devicetree `remote-endpoint`. More about it here [#72311](https://github.com/zephyrproject-rtos/zephyr/issues/72311#issuecomment-2106400935)

- Add a `drivers/video_skeleton.c` in the same style as [`drivers/usb/udc/udc_skeleton.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/usb/udc/udc_skeleton.c) that can speed-up development and help with understanding the semantics.

### Dependencies

- https://github.com/zephyrproject-rtos/zephyr/pull/72950
- https://github.com/zephyrproject-rtos/zephyr/pull/72254
- https://github.com/zephyrproject-rtos/zephyr/pull/72623
- https://github.com/zephyrproject-rtos/zephyr/pull/72827
- (more to come over the upcoming days)

### Concerns and Unresolved Questions

- A few modification to existing drivers are to be introduced to support all proposed changes.
- Many of the "rework" proposal are a matter of deciding what is most comfortable to work with, and a call to discussion more than a fixed solution proposal.
- Using a FIFO like done now is widely understood and simple solution.

## Alternatives

Leaving-up the application developer resolve these issues is the current approach, and demos and samples submitted #71463 show that it effectively works end-to-end with the current API.

For the case of RTIO, it is also possible to introduce a "genderless" interconnect between source and sinks (so that the application or another driver can use it) by replacing `enqueue()`/`dequeue()` by a public FIFO integrated directly in the API.

josuah, 2024-05-17:
Extra material to investigate the rationale of the existing APIs, so that their semantics can be kept: https://static.linaro.org/connect/san19/presentations/san19-503.pdf

josuah, 2024-05-18:
Thanks to @loicpoulain I [rediscovered Gstreamer,](https://discuss.96boards.org/t/mpph264enc-is-too-slow/8902) and it is great!

The above is also the foundation for a zephyr-shell pipeline, as it is not possible to do it if rewriting the application every time is required. The difficulty, though, is that getting `<&phandles>` from `remote-endpoint` at compile-time will not allow to reconnect them later with zephyr-shell.

Maybe it is enough to have "gst-launch, but all devicetree" as a first step...

teburd, 2024-05-20:
Right you could treat an rtio_iodev as an endpoint (input or output stream), which is what sensors do today. There's effectively two output streams for sensors, one which polled, and one which is driven by the sensor itself with an event.

I could imagine video input/output endpoints being similar, though perhaps needing some additional information about image encoding and framesize to describe the metadata.

josuah, 2024-05-21:
So let's give it a try! :) I will pick some hardware known to work first and move from here.

To everyone working on a sample or a driver, I wish to avoid disturbing any ongoing effort, so in case I missed an issue, do feel free to ping me about it, I can provide a patch to merge on the PR or delay another PR so that actual driver contribution gets merged first!

teburd, 2024-08-12:
@josuah any progress on doing video with RTIO? Any help needed? I have an interest in seeing this work

josuah, 2024-08-12:
@teburd many thanks!

Not direct progress, but still interesting insights:

- USB Video Class with a slow CPU: context-switching seems to impact FPS.
- There are independent parts to `video.h`: control (no change needed?), data (to integrate with RTIO).
- RTIO could be built as a separate API acting as a replacement for enqueue()/dequeue()
- RTIO could be built top of enqueue()/dequeue() at the cost of extra FIFO -> RTIO -> FIFO traffic but useful for migration

Where the most help is needed is probably in designing and discussing an API that gracefully fits the `<video.h>` model.
Then I'd be quite glad to try to convert the existing drivers. I started gathering boards for that (NXP's FRDM, Some ST's...).

teburd, 2024-08-12:
> @teburd many thanks!
> 
> Not direct progress, but still interesting insights:
> 
>     * USB Video Class with a slow CPU: context-switching seems to impact FPS.

This is pretty common, if interrupts are involved you can reduce the cost by using direct interrupts and controlling more when the scheduler gets called into only when you know things need to be rescheduled. E.g. completed entire transfer rather than a little peripheral FIFO ding to do some refilling.

> 
>     * There are independent parts to `video.h`: control (no change needed?), data (to integrate with RTIO).
> 
>     * RTIO could be built as a separate API acting as a replacement for enqueue()/dequeue()

Those APIs could be replaced with the read/write ops for RTIO. Ideally we'd benefit then from some common infrastructure around tracing/statistics/timeouts/etc.

> 
>     * RTIO could be built top of enqueue()/dequeue() at the cost of extra FIFO -> RTIO -> FIFO traffic but useful for migration

It can be, but it'd be an added cost instead of benefit perhaps, but I don't know enough of the video API to make a judgement on that. It's quite possible the API and implementations already do whats needed.

> 
> 
> Where the most help is needed is probably in designing and discussing an API that gracefully fits the `<video.h>` model. Then I'd be quite glad to try to convert the existing drivers. I started gathering boards for that (NXP's FRDM, Some ST's...).

I don't have many boards that support video other than the imxrt1060 evk, which I do have the camera and screen for.

josuah, 2024-08-13:
> completed entire transfer rather than a little peripheral FIFO ding to do some refilling

Precious insight!

> Those APIs could be replaced with the read/write ops for RTIO.

The API [introduced](https://resources.linaro.org/en/resource/EksPTo3AMEdbbte3gppX4R) by @loicpoulain looks similar to RTIO in practice:

- [`video_enqueue()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/video.h#L170-L172) submits to a `fifo_in` (like a submission queue).
- [`video_dequeue()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/video.h#L180-L183) pulls from a `fifo_out` (like a completion queue) with a timeout parameter.
- [`video_set_signal()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/video.h#L248-L250) permits to watch multiple video devices like POSIX `select(2)`.
- Some drivers ([1](https://github.com/zephyrproject-rtos/zephyr/blob/5368d7d0df3571cea21fbb520c45015df27f576f/drivers/video/video_sw_generator.c#L22-L23), [2](https://github.com/zephyrproject-rtos/zephyr/pull/66994/files#diff-a6394ee84ba37d741b1aecea29085811d0f447fe128a655c3840eb59c3e86604R104), [3](https://github.com/zephyrproject-rtos/zephyr/pull/76798/files#diff-8741d252cf20802cf806752bfd74a7e120f74445aa181cfd85ff04c9e2dc8ec0R49)) have an extra work queue

7 drivers do I/O in total:
- https://github.com/zephyrproject-rtos/zephyr/blob/5368d7d0df3571cea21fbb520c45015df27f576f/drivers/video/video_mcux_csi.c#L31-L32
- https://github.com/zephyrproject-rtos/zephyr/blob/5368d7d0df3571cea21fbb520c45015df27f576f/drivers/video/video_stm32_dcmi.c#L39-L40
- https://github.com/zephyrproject-rtos/zephyr/blob/5368d7d0df3571cea21fbb520c45015df27f576f/drivers/video/video_sw_generator.c#L20-L21
- https://github.com/zephyrproject-rtos/zephyr/pull/72827
- https://github.com/zephyrproject-rtos/zephyr/pull/75331
- https://github.com/zephyrproject-rtos/zephyr/pull/66994
- https://github.com/zephyrproject-rtos/zephyr/pull/76798

josuah, 2024-08-14:
Another challenge is to combine the  RTIO and video buffers together.

https://github.com/zephyrproject-rtos/zephyr/blob/6b7558a9bf96fc5569d1b9b73a01f78ce6affe58/include/zephyr/rtio/rtio.h#L232-L260

https://github.com/zephyrproject-rtos/zephyr/blob/6b7558a9bf96fc5569d1b9b73a01f78ce6affe58/include/zephyr/drivers/video.h#L105-L119
with also `uint32_t flags;` and `uint32_t bytesframe` introduced in https://github.com/zephyrproject-rtos/zephyr/pull/66994

Is it planned to just wrap the current video struct?
```c
rtio_sqe.userdata = &video_buffer;
```

Or use RTIO instead of video_buffer mapping the fields like this maybe?
```c
rtio_sqe.buf = video_buffer_buffer;
rtio_sqe.buf_len = video_buffer_size; /* if OP_RX */
rtio_sqe.buf_len = video_buffer_bytesused; /* if OP_TX */
rtio_sqe.userdata = video_buffer_driver_data;
(void *)video_buffer_bytesframe; /* does not fit */
(void *)video_buffer_timestamp; /* does not fit */
```

Or change the video buffer to only contain what does not fit into RTIO buffers to avoid too much copy?
```c
struct video_buffer_extra_info {
    	uint32_t timestamp;
	uint32_t bytesframe;
} video_buffer;

rtio_sqe.userdata = &video_buffer;
```

Or modify RTIO to support the extra fields so that `rtio_sqe` can replace `video_buffer` completely?

- https://github.com/zephyrproject-rtos/zephyr/issues/76972 for replacing `timestamp`
- something else replacing `bytesframe`

If RTIO replaces `struct video_buffer`, then there would need to handle the allocation too.

Tagging @loicpoulain who introduced the API and contributors/reviewers @ArduCAM @CharlesDias @danieldegrasse @decsny @epc-ake @erwango @ngphibang (alphabetical order) in case anyone is interested in seeing this happen.

teburd, 2024-08-14:
The way sensing deals with this is by encoding the extra info in the buffer itself and providing functions to get it back. Video can do something similar perhaps. It would require over allocating the buffer itself to account for the metadata and not just the frame data then.

XenuIsWatching, 2024-08-16:
I have worked quite a bit with cameras to the point where we made our own "internal" API with cameras due to some short comings with the 'current' implemention of the video api. It's good to see some work on this now to improve it.

Some APIs that I've noticed that are lacking within the current implemention are ways of setting the Physical Link Rate and Type such as CPHY or DPHY along with their rate in sps for CPHY and bps for DPHY. This would be a good feature that could be implemented somehow.
Some Cameras may be a bit 'advanced', and be able to stream out multiple frame types out at once such as a IR image along with a RGB image, I'm not sure how that would be currently implemented to select which frame type to enable or set a ctrl for.

josuah, 2024-08-16:
@XenuIsWatching

> ways of setting the Physical Link Rate and Type such as CPHY or DPHY along with their rate in sps for CPHY and bps for DPHY

I think this PR is addressing this, leveraging `port { endpoint { link-properties-here; }; };`:

- https://github.com/zephyrproject-rtos/zephyr/pull/74415

> stream out multiple frame types out at once such as a IR image along with a RGB image

This PR would allow specify endpoint numbers in addition to the generic `VIDEO_EP_IN/OUT/ANY/NONE`.

- https://github.com/zephyrproject-rtos/zephyr/pull/73009

I will comment on each PR...

teburd, 2024-08-16:
Sending deals with multiplexed fifo buffers today and could be used as a model to follow perhaps. Or you could have multiple output streams, each one acting like an IO device (struct rtio_iodev) by moving to that API.

josuah, 2024-08-16:
@teburd

I did not think about placing the metadata in the buffer! I think all elements are in place and I will be able to test now.

> It would require over allocating the buffer itself to account for the metadata and not just the frame data then.

This could be integrated into the video allocation functions then.

> Sensing deals with multiplexed fifo buffers today and could be used as a model to follow perhaps.
> Or you could have multiple output streams, each one acting like an IO device (struct rtio_iodev) by moving to that API.

Time for experimentation. Thank you.

teburd, 2024-08-16:
> I have worked quite a bit with cameras to the point where we made our own "internal" API with cameras due to some short comings with the 'current' implemention of the video api. It's good to see some work on this now to improve it.
> 
> Some APIs that I've noticed that are lacking within the current implemention are ways of setting the Physical Link Rate and Type such as CPHY or DPHY along with their rate in sps for CPHY and bps for DPHY. This would be a good feature that could be implemented somehow. Some Cameras may be a bit 'advanced', and be able to stream out multiple frame types out at once such as a IR image along with a RGB image, I'm not sure how that would be currently implemented to select which frame type to enable or set a ctrl for. Maybe `enum video_endpoint_id ep` could be used for it?

Right, this is general an issue in Zephyr today because we don't have a way of doing device specific behaviors akin to ioctl. Sensing sort of deals with this with its attributes API but there are still oddities with it. Because each call to set attr is a partial reconfiguration of the device, and ordering can matter, there can be invalid configurations.

For example many devices offer low power modes which will toggle on/off the mems device at the cost of noise. Usually these modes are limited in sampling rate. Sometimes the sample rate overlaps, but frequently some sample rates only work in a low noise (always on) mode.

So now you have this quirk of ordering where you may wish to change both the power mode and sample rate at the same time, but the API has no way of allowing this. So drivers then have to work out what the implied meaning of a sample rate setting may mean.

There's also the dai interface which fully gave up on trying to provide structured configuration and takes a ```void *config``` defined by each dai implementation.

Sambsc19, 2025-02-04:
Please add support for STM32H7S78-DK

josuah, 2025-02-06:
> Please add support for STM32H7S78-DK DCMIPP. (ov5640 in my case)

@Sambsc19 I believe this got merged 2h from now: https://github.com/zephyrproject-rtos/zephyr/pull/76124#event-16201686749

Assuming you meant a parallel port with the OV5640.