zephyrproject-rtos/zephyr #72963: Build Process Hangs on Subsequent Builds After First Successful Run in VSCode

schererant, 2024-05-17:
---

**Describe the bug**
The build process stops at the last line during the execution of the build command using west in VSCode. The issue occurs after the first successful build, and subsequent builds halt without completing. Restarting VSCode temporarily resolves the issue.

Please also mention any information which could help others to understand the problem you're facing:
 - **Target platform**: ARM Cortex-M based custom board (customboard)
 - **Diagnostic steps**: 
   - Monitored resource usage.
   - Cleaned and rebuilt the build directory.
   - Ran the build process in an external terminal.
   - Disabled extensions in VSCode to identify potential conflicts.
 - **Regression**: Not confirmed. The issue was observed recently but hasn't been bisected to a specific commit.

**To Reproduce**
Steps to reproduce the behavior:
1. Execute the build command in VSCode:
   ```sh
   /path/to/venv/bin/west build -p -b customobc -- -DBOARD_ROOT=/path/to/project/root -DCONFIG_DEBUG_THREAD_INFO=y -DCONFIG_INIT_STACKS=y -DCONFIG_THREAD_STACK_INFO=y
   ```
2. The build process succeeds initially but fails on subsequent attempts, halting at the following line:
   ```
   -- Found GnuLd: /path/to/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
   ```

**Expected behavior**
The build process should complete successfully on subsequent attempts without needing to restart VSCode.

**Impact**
This issue is a showstopper as it requires restarting the development environment frequently, disrupting the workflow and slowing down development progress.

**Logs and console output**
```
 *  Executing task in folder app: /path/to/workspace/.venv/bin/west build -p -b customboard -- -DBOARD_ROOT=/path/to/workspace/project -DCONFIG_DEBUG_THREAD_INFO=y -DCONFIG_INIT_STACKS=y -DCONFIG_THREAD_STACK_INFO=y 

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /path/to/workspace/project/app
-- CMake version: 3.23.3
-- Found Python3: /path/to/workspace/.venv/bin/python3.12 (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /path/to/Library/Caches/zephyr
-- Zephyr version: 3.6.0 (/path/to/workspace/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: customboard
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/path/to/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/path/to/zephyr-sdk-0.16.5)
-- Found Dtc: /opt/homebrew/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /path/to/workspace/project/boards/arm/customboard/customboard.dts
'label' is marked as deprecated in 'properties:' in /path/to/workspace/project/dts/bindings/dac/adi,ad5254.yaml for node /soc/i2c@40005400/i2c_mux1@71/mux_i2c@7/ad5254@2d.
-- Generated zephyr.dts: /path/to/workspace/project/app/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /path/to/workspace/project/app/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /path/to/workspace/project/app/build/zephyr/dts.cmake

warning: Deprecated symbol CPLUSPLUS is enabled.

warning: Deprecated symbol LIB_CPLUSPLUS is enabled.

Parsing /path/to/workspace/project/app/Kconfig
Loaded configuration '/path/to/workspace/project/boards/arm/customboard/customboard_defconfig'
Merged configuration '/path/to/workspace/project/app/prj.conf'
Merged configuration '/path/to/workspace/project/app/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/path/to/workspace/project/app/build/zephyr/.config'
Kconfig header saved to '/path/to/workspace/project/app/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /path/to/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
```

**Environment (please complete the following information):**
 - **OS**: macOS 14.5 (23F79)
 - **Toolchain**: Zephyr SDK 0.16.5-1
 - **Commit SHA or Version used**: Zephyr version 3.6.0
 - **VSCode**: Version: 1.89.1

**Additional context**
- Using VSCode with the following extensions (all up to date):
  - ms-vscode.cpptools-extension-pack
  - ms-python.python
  - ms-vscode.vscode-embedded-tools
  - ms-vscode.vscode-serial-monitor
  - marus25.cortex-debug
  - donjayamanne.python-environment-manager
  - aaron-bond.better-comments
  - twxs.cmake
  - eamodio.gitlens
  - mcu-debug.rtos-views
  - ms-vscode-remote.vscode-remote-extensionpack
  - nordic-semiconductor.nrf-devicetree

- The issue appears to be related to VSCode or one of its extensions, as restarting VSCode resolves the issue temporarily.

---

henrikbrixandersen, 2024-05-18:
As far as I know, we do not provide anything related to vscode?

Can you reproduce this problem on the command line? If not, I'd advice you to report this issue against vscode instead.

nashif, 2024-05-21:
We do not provide any vscode extensions, please report the issue wherever you got the extensions from.

closed 2024-05-21 by nashif

cdwilson, 2024-10-31:
@schererant FYI, I'm seeing the exact same behavior as well (build command hangs in VS Code terminal, but the same command works fine in iTerm)

I found https://devzone.nordicsemi.com/f/nordic-q-a/110023/getting-stuck-at-ld-bfd-tool-when-generating-configuration-from-nrf-connect-sdk-for-vs-code which suggested that the issue is a VS Code bug introduced in version 1.88 onwards.

If you try reverting to VS Code 1.87 (https://code.visualstudio.com/updates/v1_87) does this issue still occur for you? When I revert to VS Code 1.87 the issue goes away for me (i.e. build command completes without hanging). Obviously this isn't a good long term solution, but I just wanted to see if you can reproduce this "fix" by reverting VS Code on your end to 1.87.

cdwilson, 2024-10-31:
> please report the issue wherever you got the extensions from.

BTW, this isn't an issue with any VS Code extension. I can reproduce this hang behavior on a fresh VS Code without any extensions installed.

RobertDaleSmith, 2025-07-28:
yea i am seeing the same thing and have been going in circles trying to resolve it. restarting vscode allows the build to work but hangs at same spot every subsequent west build command after the initial successful build. everything works flawlessly outside of Cursor/VS Code's terminal though.

tohidg-fcs, 2026-02-13:
Same issue, MACOS,VS code .

YuSalvatore, 2026-02-25:
Same issue, MacOS VSCode 
Ver1.109.5(Universal)