zephyrproject-rtos/zephyr #72979: `nrf54h20dk/nrf54h20/cpuppr` fails `tests/kernel/gen_isr_table/arch.interrupt.gen_isr_table.riscv_no_direct`

ycsin, 2024-05-18:
**Describe the bug**
`nrf54h20dk/nrf54h20/cpuppr` fails to build the `tests/kernel/gen_isr_table/arch.interrupt.gen_isr_table.riscv_no_direct` test and is currently failing in the CI of #66877

**To Reproduce**
Steps to reproduce the behavior:
1. Check out `main`
2. `west build -b nrf54h20dk/nrf54h20/cpuppr -p auto zephyr/tests/kernel/gen_isr_table -T arch.interrupt.gen_isr_table.riscv_no_direct`
3. See failure

**Expected behavior**
Build should pass in `main` branch

**Impact**
Blocking #66877

**Logs and console output**

```
/Users/ycsin/zephyr-sdk-0.16.5-1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(vector.S.obj): in function `.L0 ':
/Users/ycsin/zephyrproject/zephyr/soc/common/riscv-privileged/vector.S:40: undefined reference to `_irq_vector_table'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /Users/ycsin/homebrew/bin/cmake --build /Users/ycsin/zephyrproject/build
```

See the [CI output](https://github.com/zephyrproject-rtos/zephyr/actions/runs/9121865587/job/25081815350?pr=66877) of #66877

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: 0.16.5-1
 - Commit SHA or Version used: Current `main` branch

ycsin, 2024-05-18:
cc @anangl

masz-nordic, 2024-05-20:
This subtest should be skipped for VPR cores - their CLIC supports vectored mode only.

closed 2024-05-25 by jhedberg