zephyrproject-rtos/zephyr #73000: Device tree error with nice_nano_v2

ElectricS01, 2024-05-20:
I am getting the following error when trying to build keyboard firmware for the nice_nano_v2 target 
```devicetree error: child specifier for <Node /kscan in '/tmp/zmk-config/zephyr/misc/empty_file.c'> (b'\x00\x00\x00\x0b\x00\x00\x00 ') does not appear in <Property 'gpio-map' at '/connector' in '/tmp/zmk-config/zephyr/misc/empty_file.c'>```

I'm using GitHub Actions to build it in this repo https://github.com/dp2305/zmk-config-tubetest/
I've made sure that I'm using the latest version of zephyr

Here's the full error log: 
```Run west build -s zmk/app -d "/tmp/tmp.qwmJNd36X2" -b "nice_nano_v2"  -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD="TubeTest" -DZMK_EXTRA_MODULES='/__w/zmk-config-tubetest/zmk-config-tubetest' 
+ west build -s zmk/app -d /tmp/tmp.qwmJNd36X2 -b nice_nano_v2 -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=TubeTest -DZMK_EXTRA_MODULES=/__w/zmk-config-tubetest/zmk-config-tubetest
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /tmp/zmk-config/zmk/app
-- CMake version: 3.28.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /tmp/zmk-config/zephyr/.cache
-- Zephyr version: 3.5.0 (/tmp/zmk-config/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Adding ZMK config directory as board root: /tmp/zmk-config/config
-- Adding /tmp/zmk-config/config/boards/shields/TubeTest
-- ZMK Config directory: /tmp/zmk-config/config
-- Using keymap file: /tmp/zmk-config/config/boards/shields/TubeTest/TubeTest.keymap
-- Board: nice_nano_v2
-- Shield(s): TubeTest
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found Dtc: /opt/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /tmp/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /tmp/zmk-config/config/boards/shields/TubeTest/TubeTest.overlay
-- Found devicetree overlay: /tmp/zmk-config/config/boards/shields/TubeTest/TubeTest.keymap
devicetree error: child specifier for <Node /kscan in '/tmp/zmk-config/zephyr/misc/empty_file.c'> (b'\x00\x00\x00\x0b\x00\x00\x00 ') does not appear in <Property 'gpio-map' at '/connector' in '/tmp/zmk-config/zephyr/misc/empty_file.c'>
CMake Error at /tmp/zmk-config/zephyr/cmake/modules/dts.cmake:279 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/cmake/modules/zephyr_default.cmake:[12](https://github.com/dp2305/zmk-config-tubetest/actions/runs/9152470279/job/25159966567#step:11:13)9 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- In: /tmp/tmp.qwmJNd36X2/zephyr, command: /usr/bin/python3;/tmp/zmk-config/zephyr/scripts/dts/gen_defines.py;--dts;/tmp/tmp.qwmJNd36X2/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/tmp/zmk-config/zmk/app/module/dts/bindings;/tmp/zmk-config/zmk/app/dts/bindings;/tmp/zmk-config/zephyr/dts/bindings;--header-out;/tmp/tmp.qwmJNd36X2/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/tmp/tmp.qwmJNd36X2/zephyr/zephyr.dts.new;--edt-pickle-out;/tmp/tmp.qwmJNd36X2/zephyr/edt.pickle;--vendor-prefixes;/tmp/zmk-config/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/tmp/zmk-config/zephyr/dts/bindings/vendor-prefixes.txt
-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.qwmJNd36X2 -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=TubeTest -DZMK_EXTRA_MODULES=/__w/zmk-config-tubetest/zmk-config-tubetest -S/tmp/zmk-config/zmk/app
Error: Process completed with exit code 1.```

henrikbrixandersen, 2024-05-21:
This seems like something that should be reported against ZMK, not Zephyr?

nashif, 2024-05-21:
Closing, as this is not a zephyr bug and the board does not exist in zephyr. Please report against ZMK.

TylerBarnes, 2025-01-04:
If anyone else finds this via google: your `row-gpios` and `col-gpios` are probably using incorrect pin numbers. For ex if it's a nicenano don't write `06` (for `P0.06`) - write `1` (as in `D1`).

luckyunlock, 2025-12-26:
I ran into a similar issue and eventually figured out the cause, so sharing here in case it helps others.

I was trying to update my ZMK firmware for the **nice!nano**. Since the last time I generated a firmware package (about a year ago), the naming convention for board/shield revisions changed:

<https://zmk.dev/blog/2025/12/09/zephyr-4-1#board-revisions>

Because revision **2.0** is now the default for the nice!nano, the board name `nice_nano_v2` no longer works.  
To fix the build, I simply replaced `nice_nano_v2` with `nice_nano`.