zephyrproject-rtos/zephyr #73003: tests: drivers: uart: uart_mix_fifo_poll: hags at nrf54h20 and nrf54l15

nordic-piks, 2024-05-20:
**Describe the bug**
Tests:
- tests/drivers/uart/uart_mix_fifo_poll/drivers.uart.uart_mix_poll_fifo
- tests/drivers/uart/uart_mix_fifo_poll/drivers.uart.uart_mix_poll_fifo_with_ppi
- tests/drivers/uart/uart_mix_fifo_poll/drivers.uart.uart_mix_poll_async_api_with_ppi

Observed for
- nrf54l15pdk/nrf54l15/cpuapp
- nrf54h20dk/nrf54h20/cpuapp
- nrf54h20dk/nrf54h20/cpurad

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54l15pdk/nrf54l15/cpuapp connected
2. go to your zephyr dir
3. call ./scripts/twister -T  tests/drivers/uart/uart_mix_fifo_poll -p nrf54l15pdk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-4231-g7046c9030e70 ***
Running TESTSUITE uart_mix_fifo_poll
===================================================================
START - test_mixed_uart_access
```
Test hangs

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: v3.6.0-4231-g7046c9030e70

nordic-piks, 2024-05-20:
@nordic-krch Found after #72548

nordic-krch, 2024-05-21:
It hangs after d599e2b6705eb3f3cc26fc5c02f0636660da3e6 @aescolar @adamkondraciuk any ideas?
Test is extensively using timers and there are 2 interrupt levels (GRTC runs from lower priority timer than UARTE, TIMER).

PR #73117 (2024-05-24): drivers: timer: nrf_grtc_timer: Fix for random system hangs
d599e2b6705eb removed early return from sys_clock_timeout_handler if current counter value is less than cc_val. It seems that this return is needed as after the removal some stress tests which were using system timers heavily started to hang.

Fixes #73003.
closed 2024-05-24 by nashif