zephyrproject-rtos/zephyr #73004: samples: subsys: usb: Failed to enable USB at nrf54h20

nordic-piks, 2024-05-20:
**Describe the bug**
Tests:
- samples/subsys/usb/cdc_acm/sample.usb.cdc-acm.comp
- samples/subsys/usb/audio/headset/sample.usb.audio.headset
- samples/subsys/usb/audio/headphones_microphone/sample.usb.audio.headphones_microphone
- samples/subsys/usb/mass/sample.usb.mass_ram_none
- samples/subsys/usb/mass/sample.usb.mass_ram_fat 
- samples/subsys/usb/mass/sample.usb_device_next.mass_ram_none
- samples/subsys/usb/cdc_acm_composite/sample.usb.cdc-acm-composite
- samples/subsys/usb/cdc_acm/sample.usb_device_next.cdc-acm
- samples/subsys/usb/cdc_acm/sample.usb.cdc-acm

Observed for
- nrf54h20dk/nrf54h20/cpuapp

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk/nrf54h20/cpuapp connected
2. go to your zephyr dir
3. call ./scripts/twister -T  samples/subsys/usb -p nrf54h20dk/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
samples/subsys/usb/cdc_acm/sample.usb.cdc-acm.comp
```
*** Booting Zephyr OS build v3.6.0-4231-g7046c9030e70 ***
[00:00:00.080,617] <err> usb_dc_dw: USB reset HANG! AHB Idle GRSTCTL=0x00000000
[00:00:00.080,621] <err> cdc_acm_echo: Failed to enable USB
```
samples/subsys/usb/audio/headset/sample.usb.audio.headset 
```
*** Booting Zephyr OS build v3.6.0-4231-g7046c9030e70 ***
[00:00:00.011,806] <inf> main: Entered main
[00:00:00.011,811] <inf> main: Found USB Headset Device
[00:00:00.011,828] <err> usb_descriptor: Failed to validate endpoints
[00:00:00.011,831] <err> usb_descriptor: Failed to fixup USB descriptor
[00:00:00.011,834] <err> usb_device: Failed to configure USB device stack
[00:00:00.011,840] <err> main: Failed to enable USB
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: v3.6.0-4231-g7046c9030e70

nordic-piks, 2024-05-20:
@jfischer-no Most probably after this #72774

jfischer-no, 2024-05-20:
These are for the "legacy" stack, maybe we need a new feature to distinguish them:
```yaml
supported:
  - usb_device
  - usbd
```

https://docs.zephyrproject.org/3.6.0/develop/test/twister.html#board-configuration

PR #73269 (2024-06-11): boards: add new test supported feature usbd 
We have recently added a new platform where the USB controller is only
supported in the new USB device support. This forces us to introduce a
new supported feature tag to distinguish the dependency. Use "usbd",
which is also the prefix of the new device support.

There are many boards that already have new driver support, but we will
replace the "usb_device" feature when the old device support is
deprecated.

Fixes: #73004
closed 2024-06-11 by fabiobaltieri