zephyrproject-rtos/zephyr #73038: IOT Devices disconnects if the Gateway is replaced with new gateway

SadanandPenore-eaton, 2024-05-20:
**Describe the bug**
We have multiple IOT devices connected in lab network. The Gateway is helping route the IOT communication to external hub.
But if the Lab team replaces the Gateway all the device are disconnected and we need to to do manual reset of the devices.
With the change of the Gateway the ip address is kept same but the mac address gets changed so I think the device are not able to communicate through new Gateway device.

We are using NucleoH743 devices as IOT nodes connecting to Azure IOT hub.

**To Reproduce**
Steps to reproduce the behavior:
1. Establish IOT agent and IOT hub communication.
2. Change the Lab Gateway with same Gateway address. ( mac will be obviously different for different unit)
3. The IOT agent and Hub communication stops.
4. Restart the device the communication establishes.

**Expected behavior**
The reset of the device should not be required when gateway changes.
We have hundreds of devices connected in lab network where manual reset is not possible.

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: - NA
 - Toolchain - zephyr-sdk-0.15.2 
 - Zephyr RTOS version - v3.2.0

**Additional context**
target configuration - STM32H743

SadanandPenore-eaton, 2024-05-21:
Here is a normal setup.
[image]

We implemented IOT agent, MQTT is used. We are using azure-sdk-for-c for implementation of IOT.
 The messages are routed to IOT hub through Gateway ( Usually a router). But if gateway is changed the device's IOT communication stops as the mac address is different for the new Gateway.
We did not try it on latest zephyr 3.6 as will need lot of changes our team is already in the process of doing that but facing lot of challenges.

SadanandPenore-eaton, 2024-05-22:
Requesting the raise the priority to high or more.

rlubos, 2024-05-22:
This sounds like a stale ARP entry on the Zephyr device side.

Is `CONFIG_NET_ARP_GRATUITOUS` enabled in your project? Is the gateway device sending gratuitous ARP requests? If any of these is not true, then the ARP entry won't be updated automatically, and you'd need to clear the ARP cache manually from the application.

Try to call `net_arp_clear_cache()` whenever communication fails, and see if that helps.

SadanandPenore-eaton, 2024-05-24:
CONFIG_NET_ARP_GRATUITOUS is enabled.
We are receiving the GARP query. But we observed that the destination mac address in that query as 00:00:00:00:00:00.
Zephyr's arp.c file only handles FF:FF:FF:FF:FF:FF and so it ignores the request.
Refer the attached log
[image]

If we handle all zeros it works. Here is what I tried which worked. ( **Bold code is newly added code in arp.c** )
		if (IS_ENABLED(CONFIG_NET_ARP_GRATUITOUS)) {
			if (memcmp(&eth_hdr->dst, net_eth_broadcast_addr(), sizeof(struct net_eth_addr)) == 0 &&
			    memcmp(&arp_hdr->dst_hwaddr, net_eth_broadcast_addr(), sizeof(struct net_eth_addr)) == 0 &&
			    memcmp(&arp_hdr->dst_ipaddr, &arp_hdr->src_ipaddr, sizeof(struct in_addr)) == 0)
			{
				/* If the IP address is in our cache,
				 * then update it here.
				 */
				NET_INFO("CONFIG_NET_ARP_GRATUITOUS FF");
 
				arp_update(net_pkt_iface(pkt),
					   (struct in_addr *)arp_hdr->src_ipaddr,
&arp_hdr->src_hwaddr,
					   true, false);
				break;
			}
 
			**if (memcmp(&eth_hdr->dst, net_eth_broadcast_addr_00(), sizeof(struct net_eth_addr)) == 0 &&
			    memcmp(&arp_hdr->dst_hwaddr, net_eth_broadcast_addr_00(), sizeof(struct net_eth_addr)) == 0 &&
			    memcmp(&arp_hdr->dst_ipaddr, &arp_hdr->src_ipaddr, sizeof(struct in_addr)) == 0)
			{
				/* If the IP address is in our cache,
					* then update it here.
					*/
				NET_INFO("CONFIG_NET_ARP_GRATUITOUS 00");
 
				arp_update(net_pkt_iface(pkt),
						(struct in_addr *)arp_hdr->src_ipaddr,
&arp_hdr->src_hwaddr,
						true, false);
				break;
			}**				
		}

**Below code added at other respective locations**
static const struct net_eth_addr broadcast_eth_addr_00 = {
    { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 } };
 
const struct net_eth_addr *net_eth_broadcast_addr_00(void)
{
    return &broadcast_eth_addr_00;
}

Here is the more description of it
https://wiki.wireshark.org/Gratuitous_ARP
[image]

Can you please confirm if this is right solution.

SadanandPenore-eaton, 2024-05-24:
net_arp_clear_cache() also resolves but right now we are not sure about the impact of that.
I think if the protocols sends zeros it is better to handle that in the arp input function.

jukkar, 2024-05-24:
Gratuitous arp uses all ff in dest address, but auto ip uses all 00. See https://wiki.wireshark.org/Gratuitous_ARP for details.
So if you enable auto ip, do things start to work?

rlubos, 2024-05-24:
I think we need to differentiate "destination HW address" from the L2 header, and "target HW address" from the ARP header.
The latter indeed is said to be set to all-zeroes for ARM probes (https://datatracker.ietf.org/doc/html/rfc5227#section-2.1.1 also confirms that), so we may indeed have a bug in our ARP implementation.

rlubos, 2024-05-24:
I've opened a PR with a quick fix: https://github.com/zephyrproject-rtos/zephyr/pull/73276. Please give it a try and review.

In terms of Zephyr 3.2 however you'll need to cherry-pick this on your own - as stated before that Zephyr version is no longer supported.

closed 2024-05-28 by aescolar