zephyrproject-rtos/zephyr #73041: Unaligned access fault in net_if.c for pi_pico_w.

foldedtoad, 2024-05-20:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
When building & executing wifi driver for Raspberry Pi Pico-W, an Unaligned Access Fault occurs.

**To Reproduce**
This wifi driver and app are out of tree and can be downloaded at https://github.com/foldedtoad/pico_wifi.git
1) cd into app subdirectory 
2) set contents of "local.conf" to said and password.
3) run "west build -b rpi_pico_w ."
4) connect JTAG/SWD to target rpi_pico_w.
5) download firmware to rpi_pico_w:  I use Ozone debugger, but OpenOCD also works.
6) start/run the firmware.
7) wait for broadcast message to be received: usually under 2 or 3 minutes.

The unaligned access fault occurs during the processing of the broadcast frame.

**Expected behavior**
No unaligned fault should happen.
The rpi_pico_w's rp2040 SoC is built on cortex-m0 technology on which all word-access instructions require data to be 32-bit aligned.

**Impact**
I have found a fix for this issue which occurs in ./subsys/net/ip/net_if.c (see below).
But this fix or something similar needs to be in main Zephyr.

```
diff --git a/subsys/net/ip/net_if.c b/subsys/net/ip/net_if.c
index e7306acd87..c41751830e 100644
--- a/subsys/net/ip/net_if.c
+++ b/subsys/net/ip/net_if.c
@@ -3441,7 +3441,7 @@ static bool ipv4_is_broadcast_address(struct net_if *iface,
                bcast.s_addr = ipv4->unicast[i].ipv4.address.in_addr.s_addr |
                               ~ipv4->unicast[i].netmask.s_addr;
 
-               if (bcast.s_addr == addr->s_addr) {
+               if (bcast.s_addr == UNALIGNED_GET(&addr->s_addr)) {
                        ret = true;
                        goto out;
                }

```

**Environment (please complete the following information):**
 - OS: Linux, MacOS
 - Toolchain: Zephyr SDK 0.16.5
 - Zephyr V3.6.99

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

jukkar, 2024-05-21:
@foldedtoad your proposed fix is good, please send a PR for it.

rlubos, 2024-05-21:
Seems to be fixed already in https://github.com/zephyrproject-rtos/zephyr/pull/72953

jukkar, 2024-05-21:
We can close this as the issue is already fixed.

closed 2024-05-21 by jukkar