zephyrproject-rtos/zephyr #73082: DOC: Generation: Win10: Could not import extension zephyr.kconfig (exception: No module named 'sphinx.util.display')

butok, 2024-05-21:
**Describe the bug**

Zephyr Documentation generation on Windows is causing the Sphinx error:
_sphinx.errors.ExtensionError: Could not import extension zephyr.kconfig (exception: No module named 'sphinx.util.display')_

**To Reproduce**
Steps to reproduce the behavior:
1. Use Windows OS.
2. Install all documentation tools and requirements according to https://docs.zephyrproject.org/latest/contribute/documentation/generation.html
3. Go to the zephyr doc directory
`cd \zephyr\doc`
4.  Use cmake to configure a Ninja-based build system:
`cmake -GNinja -B_build .`
5. Enter the build directory
`cd _build`
6. To generate HTML output, run ninja on the generated build system:
`ninja html`
5. See error in the command log: _sphinx.errors.ExtensionError: Could not import extension zephyr.kconfig (exception: No module named 'sphinx.util.display')_:

````
C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build>ninja html
[0/2] Generating Devicetree bindings documentation...
[1/2] Running Sphinx HTML build...Running Sphinx v5.3.0

Traceback (most recent call last):
  File "C:\Python311\Lib\site-packages\sphinx\registry.py", line 459, in load_extension
    mod = import_module(extname)
          ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_extensions\zephyr\kconfig\__init__.py", line 49, in <module>
    from sphinx.util.display import progress_message
ModuleNotFoundError: No module named 'sphinx.util.display'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Python311\Lib\site-packages\sphinx\cmd\build.py", line 276, in build_main
    app = Sphinx(args.sourcedir, args.confdir, args.outputdir,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\sphinx\application.py", line 223, in __init__
    self.setup_extension(extension)
  File "C:\Python311\Lib\site-packages\sphinx\application.py", line 398, in setup_extension
    self.registry.load_extension(self, extname)
  File "C:\Python311\Lib\site-packages\sphinx\registry.py", line 462, in load_extension
    raise ExtensionError(__('Could not import extension %s') % extname,
sphinx.errors.ExtensionError: Could not import extension zephyr.kconfig (exception: No module named 'sphinx.util.display')

Extension error:
Could not import extension zephyr.kconfig (exception: No module named 'sphinx.util.display')

FAILED: CMakeFiles/html C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/CMakeFiles/html
C:\WINDOWS\system32\cmd.exe /C "cd /D C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build && C:\Users\nxa12205\.mcuxpressotools\cmake-3.28.1-windows-x86_64\bin\cmake.exe -E env DOXYGEN_EXECUTABLE=C:/Strawberry/c/bin/doxygen.exe DOT_EXECUTABLE=C:/ProgramData/chocolatey/bin/dot.exe C:/Python311/Scripts/sphinx-build.exe -b html -c C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc -d C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/doctrees -w C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/html.log -t development -j auto -W --keep-going -T C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/src C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/html"
ninja: build stopped: subcommand failed.
````

**Expected behavior**
No error

**Impact**
Not able to generation the Zephyr documentation.

**Environment (please complete the following information):**
 - OS: Windows
 - Sphinx v5.3.0
 - Commit a2ebe49259d12c7c3375669f65ed5703ddd7f33a

kartben, 2024-05-21:
@butok you seem to be running a pretty old Sphinx version, and that is why it's failing. Please make sure you have up-to-date Python dependencies for the documentation build as per `zephyr/doc/requirements.txt`

gmarull, 2024-05-21:
Please, make sure to install updated dependencies: `pip install -U -r doc/requirements.txt`

closed 2024-05-21 by gmarull

butok, 2024-05-22:
> Please, make sure to install updated dependencies: `pip install -U -r doc/requirements.txt`

Thank you @gmarull, 
I have followed the Zephyr documentation, and I have used `pip install -r doc/requirements.txt` (Without -U).
I'll probably add a fix to the documentation. 

Also, think about how to update the requirements.txt file to avoid this problem on systems with previously installed old modules.