zephyrproject-rtos/zephyr #73089: Zephyr MQTT library -128 error when trying to use unencrypted broker with authentication.

tsotnek, 2024-05-21:
Hello,

I am trying to test the connection to the MQTT broker using nRF9160 running Zephyr. I have so far tested the following combinations: 

- unencrypted, unauthenticated.
- unencrypted, authenticated.
- encrypted, unauthenticated. 
- encrypted, authenticated. 

So far there are no issues with either unencrypted and encrypted as long as I use no user and password (unauthenticated). But as soon as I try to set the client username and password, things go south. This is the code I am using to set the user and password in client_init function. 

`       uint8_t username[] = "user";
	uint8_t password[] = "user";
	struct mqtt_utf8 pass,name;
	
	pass.size = (uint32_t) strlen(password);
	pass.utf8 = password;
	name.size = (uint32_t) strlen(username);
	name.utf8 = username;

	client->password = &pass;
	client->user_name = &name;`

As soon as I use the unauthenticated broker and set both password and user_name to **NULL**, there are no more issues.

I am wondering if this is a bug in the library and if someone else had issues with it?

Thanks

rlubos, 2024-05-21:
>  But as soon as I try to set the client username and password, things go south.

So what happens exactly? Does it crash or simply fail to connect? What Zephyr revision (commit) are you using?

> As soon as I use the unauthenticated broker and set both password and user_name to NULL, there are no more issues.

Not sure how accurate is the code snippet you provide, but please be aware that the user_name/password arrays need to have the same lifespan as the MQTT client context structure you assign them to (I've briefly checked, and most of our example either define them in a global scope, or if defined within a function, mark them as static.

In you example code, it looks as if the `user_name`/`password` were defined within a `client_init()` function, on a stack. That wouldn't work, as that memory could be overwritten just after the function exits.

tsotnek, 2024-05-21:
Thanks for your reply. 

> So what happens exactly? Does it crash or simply fail to connect? What Zephyr revision (commit) are you using?
The application doesn't crash it just fails to connect. I am using Nordic's **NCS toolchain** which uses Zephyr SDK 0.16.1.

> In your example code, it looks as if the user_name/password were defined within a client_init() function, on a stack. That wouldn't work, as that memory could be overwritten just after the function exits.

I am sorry, I should've written this more clearly. I just copied it with the code snippet for visual understanding. 

This is defined in the global scope in the .c file:
`struct mqtt_utf8 pass,name;`

The rest of the code, however, is then defined in **client_init()** function as

```
client_init(){
    ........
    uint8_t username[] = "user";
    uint8_t password[] = "user";
	
    pass.size = (uint32_t) strlen(password);
    pass.utf8 = password;
    name.size = (uint32_t) strlen(username);
    name.utf8 = username;

    client->password = &pass;
    client->user_name = &name;
    ....
}

```

tsotnek, 2024-05-21:
> So username/password are local variables, and pass/name are global. And with  pass.utf8 = password; you assign an address of a local variable (string) to a global pointer. The pointer will remain intact, but the memory it points to may get overwritten. So it's quite probable, the MQTT library just writes some rubbish into the MQTT connect packet instead of a proper username/password.

I haven't thought about that, thanks for that. I will move it to the global scope, test it and will update this thread.

> And just to be sure, you are 100% certain that the MQTT broker you use is configured properly, I. e. to accept client with a given username and password? I guess brokers could simply close the connection right after sending CONNACK if misconfigured, which could in result give -ENOTCONN from the library (never tested such scenario on my own though).

Yes I am sure about that, I have tested it with the same exact user/password using Node-red MQTT client and also MQTT explorer.

tsotnek, 2024-05-22:
> So username/password are local variables, and pass/name are global. And with pass.utf8 = password; you assign an address of a local variable (string) to a global pointer. The pointer will remain [...]

So after moving both username and password arrays to the global scope, there were no more issues. The memories were indeed getting overwritten. 

Thanks for the quick and amazing assistance.

Closing this issue.

closed 2024-05-22 by tsotnek