zephyrproject-rtos/zephyr #73127: Contributor nomination: valeriosetti

valeriosetti, 2024-05-22:
## GitHub User

Provide the following information about the GitHub user:

1. Valerio Setti
1. valeriosetti
1. BayLibre

## Supporting Documents

Ongoing:
https://github.com/zephyrproject-rtos/zephyr/pull/73061
https://github.com/zephyrproject-rtos/zephyr/pull/73049
https://github.com/zephyrproject-rtos/zephyr/pull/72243

Merged:
https://github.com/zephyrproject-rtos/zephyr/pull/72747
https://github.com/zephyrproject-rtos/zephyr/pull/71830
https://github.com/zephyrproject-rtos/zephyr/pull/71827
https://github.com/zephyrproject-rtos/zephyr/pull/51236

tomi-font, 2024-05-22:
@ceolin @d3zd3z @dleach02 @MaureenHelm @carlescufi @aescolar

ceolin, 2024-05-22:
+1 for sure.

dleach02, 2024-05-30:
+1... is this our process?

carlescufi, 2024-06-12:
@valeriosetti wouldn't you rather become a collaborator here:
https://github.com/zephyrproject-rtos/zephyr/blob/41389fbb818616fa1f1127be4a87f1c183a9ab09/MAINTAINERS.yml#L2432

and here:

https://github.com/zephyrproject-rtos/zephyr/blob/41389fbb818616fa1f1127be4a87f1c183a9ab09/MAINTAINERS.yml#L3901

More info here:

https://docs.zephyrproject.org/latest/project/project_roles.html#collaborator

carlescufi, 2024-06-12:
+1 from me too

nashif, 2024-06-12:
https://github.com/zephyrproject-rtos/zephyr/invitations

closed 2024-06-12 by nashif