zephyrproject-rtos/zephyr #73131: Bluetooth: Shell: advertising connection randomly loss

dingste, 2024-05-22:
using bluetooth advertising on shell (USB)
via: **zephyr/tests/bluetooth/shell**
This specific test fails in our CI/CD regularly 

[partly configuration]
Kconfig
```
depends on SOC_SERIES_IMX_RT
select SOC_PART_NUMBER_MIMXRT1062DVL6A
CONFIG_SOC_MIMXRT1062=y
```
defconfig
```
CONFIG_SOC_SERIES_IMX_RT=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_SERIAL=y
CONFIG_GPIO=y
CONFIG_ARM_MPU=y
CONFIG_HW_STACK_PROTECTION=y
CONFIG_PINCTRL=y
CONFIG_POWER_DOMAIN=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
```

using:
* arch linux
* actual zephyr-sdk / latest repository

[pre: bt init etc.] 
_uart:~$ bt advertise on_

```
Advertising started
Connected: 5D:58:E4:2A:E2:B5 (random)
LE data len updated: TX (len: 251 time: 2120) RX (len: 27 time: 328)
Remote LMP version 5.2 (0x0b) subversion 0x27ec manufacturer 0x001d
LE Features: 0x000000000701ff6f
LE conn  param req: int (0x0006, 0x0006) lat 0 to 500
[00:00:09.086,000] <inf> fs_nvs: alloc wra: 0, f78
[00:00:09.086,000] <inf> fs_nvs: data wra: 0, 72
[00:00:09.313,000] <dbg> bt_att: bt_eatt_init:
[00:00:09.314,000] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:09.318,000] <inf> bt_hci_core: Identity: 88:88:88:88:88:88 (public)
[00:00:09.318,000] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x8300, manufacturer 0x0048
[00:00:09.318,000] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0x1052
[00:00:22.997,000] <dbg> bt_att: bt_att_accept: conn 0x20203108 handle 128
[00:00:22.997,000] <dbg> bt_att: att_chan_attach: att 0x2020f280 chan 0x2020f100 flags 0
[00:00:22.997,000] <dbg> bt_att: bt_att_connected: chan 0x2020f108 cid 0x0004
[00:00:22.997,000] <dbg> bt_att: bt_att_status: chan 0x2020f108 status 0x2020f118
[00:00:23.224,000] <dbg> bt_att: bt_att_recv: Received ATT chan 0x2020f100 code 0x10 len 6
[00:00:23.224,000] <dbg> bt_att: att_read_group_req: start_handle 0x0001 end_handle 0xffff type 2800
[00:00:23.224,000] <dbg> bt_att: read_group_cb: handle 0x0001
[00:00:23.224,000] <dbg> bt_att: read_group_cb: handle 0x000b
[00:00:23.224,000] <dbg> bt_att: read_group_cb: handle 0x0014
[00:00:23.224,000] <dbg> bt_att: chan_send: code 0x11
[00:00:23.307,000] <dbg> bt_att: att_sent: conn 0x20203108 chan 0x2020f108
[00:00:23.307,000] <dbg> bt_att: bt_att_sent: chan 0x2020f100
[00:00:23.307,000] <dbg> bt_att: chan_rsp_sent: chan 0x2020f100
[00:00:23.308,000] <dbg> bt_att: bt_att_recv: Received ATT chan 0x2020f100 code 0x10 len 6
[00:00:23.308,000] <dbg> bt_att: att_read_group_req: start_handle 0x001c end_handle 0xffff type 2800
[00:00:23.308,000] <dbg> bt_att: chan_send: code 0x01
[00:00:23.397,000] <dbg> bt_att: bt_att_recv: Received ATT chan 0x2020f100 code 0x08 len 6
[00:00:23.397,000] <wrn> bt_att: Ignoring unexpected request
[00:00:23.398,000] <dbg> bt_att: att_sent: conn 0x20203108 chan 0x2020f108
[00:00:23.398,000] <dbg> bt_att: bt_att_sent: chan 0x2020f100
[00:00:23.398,000] <dbg> bt_att: chan_rsp_sent: chan 0x2020f100
LE conn param updated: int 0x0006 lat 0 to 500
LE conn param updated: int 0x0006 lat 0 to 500
Disconnected: 5D:58:E4:2A:E2:B5 (random) (reason 0x13)
[00:00:53.378,000] <dbg> bt_att: bt_att_disconnected: chan 0x2020f108 cid 0x0004
[00:00:53.378,000] <dbg> bt_att: att_chan_detach: chan 0x2020f100
[00:00:53.378,000] <dbg> bt_att: bt_att_released: chan 0x2020f100
```

Thalley, 2024-05-22:
@dingste Please follow the bug report template. In order to understand the issue we need to know which commit you are using, what board, how to reproduce etc. 

The reason why we are asking, is that this is a looks like a weird issue that we don't see on most platforms, nor the do fix you've provided make much sense (to me at least)

dingste, 2024-05-22:
sure! thanx to your build pipeline, i see it belongs not to all hw constelations. 
i'm on it.

dingste, 2024-05-24:
would like to close the issue, after i saw so many workarounds and other specialization from our SDK, that this issue possibly occur.

closed 2024-05-24 by Thalley