zephyrproject-rtos/zephyr #73132: kconfig: Abusive/undocumented syntax

erwango, 2024-05-22:
**Describe the bug**
In the tree we can find the following undocumented Kconfig syntax:
```
config FOO
	default BAR
```
as a short for 
```
config FOO
	default y if BAR
```
with `FOO` being a bool

For instance:
```
config USB_CDC_ACM
	default SERIAL
```

This syntax is unclear as when reading it can suggest that `FOO` is selecting `BAR` option.
Also it is undocumented.

**Expected behavior**
This syntax should be prevented and existing occurrences cleaned up

**Impact**
This makes `kconfig.defconfig` files reading quite confusing.

**Logs and console output**
One example:
https://github.com/zephyrproject-rtos/zephyr/blob/c211cb347e0af0a4931e0e7af3d93577bcc7af8f/boards/nordic/nrf52840dongle/Kconfig.defconfig#L30-L48

jfischer-no, 2024-05-22:
> **Expected behavior**
> This syntax should be forbidden and existing occurrences cleaned up

Certainly not, this is a common shorthand, see also Linux kernel tree.
https://docs.zephyrproject.org/3.6.0/build/kconfig/tips.html#common-kconfig-shorthands

tejlmand, 2024-05-22:
> In the tree we can find the following undocumented Kconfig syntax:
> ...
> as a short for

no, actually that's not what this means, the construct:
```
config FOO
	default BAR
```
actually means default the value of `BAR` to the value of `FOO`.

Now with `bool`s this really doesn't change anything in practice, because at the end of the day, whether you do:
`default y if BAR`, meaning assign `y` if `BAR` is also `y`.
or you say
`default BAR`, assign the value of `BAR` to this setting, results in same value

but if starting to play with combinations of strings and bools, then the difference becomes clearer, for example:
```
config FOO_STRING
        string
        default "foo"

config BAR_BOOL
        bool
        default FOO_STRING # <-- illegal, because a string cannot be assigned to bool.

config BAR_BOOL
        bool
        default y if FOO_STRING = "foo"
```

tejlmand, 2024-05-22:
Not a bug.

erwango, 2024-05-22:
Thanks for the feedback. Then it would need to be documented in the, otherwise fairly complete, [Kconfig tips and tricks](https://docs.zephyrproject.org/latest/build/kconfig/tips.html#kconfig-tips-and-tricks) doc page.

Does it sound reasonnable ?

tejlmand, 2024-05-23:
> An example is used, but the shorthand is not explicitly documented in https://docs.zephyrproject.org/latest/build/kconfig/tips.html#common-kconfig-shorthands.

yep, seems that the docs could be improved, but that should be under setting of values and not a short-hand.
A short-hand is where the short syntax actually means the same as the long syntax.

And that is not the case with `default BAR`, as that is assigning a value.

tejlmand, 2024-05-24:
https://github.com/zephyrproject-rtos/zephyr/pull/73270

PR #73270 (2024-06-04): doc: describe Kconfig tracking symbol
Fixes: #73132

Add a subsection on tracking Kconfig symbols whose default values is obtained from another Kconfig symbol.
commit ed954bf6 (2024-06-04): doc: describe Kconfig tracking symbol

Fixes: #73132

Add a subsection on tracking Kconfig symbols whose default values is
obtained from another Kconfig symbol.
closed 2024-06-04 by MaureenHelm