zephyrproject-rtos/zephyr #73143: DOC: PDF generation is failed.

butok, 2024-05-22:
Describe the bug

Zephyr Documentation PDF generation on Windows is failed. HTML generation is fine.

To Reproduce
Steps to reproduce the behavior:

Use Windows OS.
1. Install all documentation tools and requirements according to https://docs.zephyrproject.org/latest/contribute/documentation/generation.html
2. Go to the zephyr doc directory
`cd \zephyr\doc`
3. Use cmake to configure a Ninja-based build system:
`cmake -GNinja -B_build .`
4. Enter the build directory
`cd _build`
5. To generate HTML output, run ninja on the generated build system:
ninja html
6. **HTML Generation is successful.**
`The HTML pages are in html. `
8. To generate PDF output, run ninja on the generated build system:
`ninja pdf`
9. **PDF Generation is failed** . The command log:
````
C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build>ninja pdf
[0/3] Generating Devicetree bindings documentation...
[1/3] Running Sphinx LaTeX build...Running Sphinx v7.3.7
C:\Python311\Lib\site-packages\breathe\project.py:116: RemovedInSphinx80Warning: Sphinx 8 will drop support for representing paths as strings. Use "pathlib.Path" or "os.fspath" instead.
  self._default_build_dir = os.path.dirname(app.doctreedir.rstrip(os.sep))
loading pickled environment... done
Building Kconfig database...... done
Preparing Doxyfile...
Checking if Doxygen needs to be run...
Doxygen build will be skipped (no changes)!

Traceback (most recent call last):
  File "C:\Python311\Lib\site-packages\sphinx\events.py", line 97, in emit
    results.append(listener.handler(self.app, *args))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_extensions\zephyr\api_overview.py", line 145, in sync_contents
    groups = parse_xml_dir(doxygen_xml_dir)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_extensions\zephyr\api_overview.py", line 101, in parse_xml_dir
    root = doxmlparser.index.parse(Path(dir_name) / "index.xml", True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\doxmlparser\index.py", line 1467, in parse
    doc = parsexml_(inFileName, parser)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\doxmlparser\index.py", line 63, in parsexml_
    doc = etree_.parse(infile, parser=parser, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src\lxml\etree.pyx", line 3541, in lxml.etree.parse
  File "src\lxml\parser.pxi", line 1879, in lxml.etree._parseDocument
  File "src\lxml\parser.pxi", line 1905, in lxml.etree._parseDocumentFromURL
  File "src\lxml\parser.pxi", line 1808, in lxml.etree._parseDocFromFile
  File "src\lxml\parser.pxi", line 1180, in lxml.etree._BaseParser._parseDocFromFile
  File "src\lxml\parser.pxi", line 618, in lxml.etree._ParserContext._handleParseResultDoc
  File "src\lxml\parser.pxi", line 728, in lxml.etree._handleParseResult
  File "src\lxml\parser.pxi", line 655, in lxml.etree._raiseParseError
OSError: Error reading file 'C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build\latex\doxygen\xml\index.xml': failed to load external entity "file:/C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/latex/doxygen/xml/index.xml"

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Python311\Lib\site-packages\sphinx\cmd\build.py", line 332, in build_main
    app = Sphinx(args.sourcedir, args.confdir, args.outputdir,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\sphinx\application.py", line 268, in __init__
    self._init_builder()
  File "C:\Python311\Lib\site-packages\sphinx\application.py", line 339, in _init_builder
    self.events.emit('builder-inited')
  File "C:\Python311\Lib\site-packages\sphinx\events.py", line 108, in emit
    raise ExtensionError(
sphinx.errors.ExtensionError: Handler <function sync_contents at 0x00000261540C9B20> for event 'builder-inited' threw an exception (exception: Error reading file 'C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build\latex\doxygen\xml\index.xml': failed to load external entity "file:/C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/latex/doxygen/xml/index.xml")

Extension error (zephyr.api_overview):
Handler <function sync_contents at 0x00000261540C9B20> for event 'builder-inited' threw an exception (exception: Error reading file 'C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build\latex\doxygen\xml\index.xml': failed to load external entity "file:/C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/latex/doxygen/xml/index.xml")

FAILED: CMakeFiles/latex C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/CMakeFiles/latex
C:\WINDOWS\system32\cmd.exe /C "cd /D C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build && C:\Users\nxa12205\.mcuxpressotools\cmake-3.28.1-windows-x86_64\bin\cmake.exe -E env DOXYGEN_EXECUTABLE=C:/Strawberry/c/bin/doxygen.exe DOT_EXECUTABLE=C:/ProgramData/chocolatey/bin/dot.exe C:/Python311/Scripts/sphinx-build.exe -b latex -c C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc -d C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/doctrees -w C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/latex.log -t development -t svgconvert -j auto -W --keep-going -T C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/src C:/_ddm/ZEPHYR/zephyr_vs/zephyr/doc/_build/latex"
ninja: build stopped: subcommand failed.

C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build>sphinx -?
'sphinx' is not recognized as an internal or external command,
operable program or batch file.

C:\_ddm\ZEPHYR\zephyr_vs\zephyr\doc\_build>were sphinx
'were' is not recognized as an internal or external command,
operable program or batch file.
````

Expected behavior
No error

Impact
Not able to generation the Zephyr PDF documentation.

Environment (please complete the following information):

OS: Windows 10
Sphinx v7.3.7
Commit https://github.com/zephyrproject-rtos/zephyr/commit/a2ebe49259d12c7c3375669f65ed5703ddd7f33a

butok, 2024-09-30:
BTW:
On every PR, CI sends "[zephyrproject-rtos/zephyr] Run failed: Documentation Publish (Pull Request)" e-mail

butok, 2024-12-02:
Still failed:
```
...

processing zephyr.tex: failed
sphinx-sitemap: No pages generated for sitemap.xml

Traceback (most recent call last):
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\cmd\build.py", line 514, in build_main
    app.build(args.force_all, args.filenames)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\application.py", line 381, in build
    self.builder.build_update()
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\builders\__init__.py", line 355, in build_update
    self.build(['__all__'], to_build)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\builders\__init__.py", line 437, in build
    self.write(docnames, list(updated_docnames), method)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\builders\__init__.py", line 711, in write
    self.write_documents(docnames)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\builders\latex\__init__.py", line 335, in write_documents
    doctree = self.assemble_doctree(
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\builders\latex\__init__.py", line 408, in assemble_doctree
    self.env.resolve_references(largetree, indexfile, self)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\environment\__init__.py", line 717, in resolve_references
    self.apply_post_transforms(doctree, fromdocname)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\environment\__init__.py", line 729, in apply_post_transforms
    transformer.apply_transforms()
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\transforms\__init__.py", line 94, in apply_transforms
    super().apply_transforms()  # type: ignore[misc]
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\docutils\transforms\__init__.py", line 182, in apply_transforms
    transform.apply(**kwargs)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx\transforms\post_transforms\__init__.py", line 44, in apply
    self.run(**kwargs)
  File "C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_extensions\zephyr\domain\__init__.py", line 431, in run
    category_node = search.find(
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\anytree\search.py", line 165, in find
    return _find(node, filter_=filter_, stop=stop, maxlevel=maxlevel)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\anytree\search.py", line 216, in _find
    items = _findall(node, filter_, stop=stop, maxlevel=maxlevel, maxcount=1)
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\anytree\search.py", line 228, in _findall
    raise CountError(msg % (maxcount, resultlen), result)
anytree.search.CountError: Expecting 1 elements at maximum, but found 2. (Node('/samples/net\\cloud', category={'id': 'cloud', 'name': 'IoT Cloud', 'docname': 'samples/net/cloud/README'}), Node('/samples/net/cloud', category={'id': 'cloud', 'name': 'IoT Cloud', 'docname': 'samples/net/cloud/README'}))

Exception occurred:
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\anytree\search.py", line 228, in _findall
    raise CountError(msg % (maxcount, resultlen), result)
anytree.search.CountError: Expecting 1 elements at maximum, but found 2. (Node('/samples/net\\cloud', category={'id': 'cloud', 'name': 'IoT Cloud', 'docname': 'samples/net/cloud/README'}), Node('/samples/net/cloud', category={'id': 'cloud', 'name': 'IoT Cloud', 'docname': 'samples/net/cloud/README'}))
The full traceback has been saved in C:\Users\nxa12205\AppData\Local\Temp\sphinx-err-0m76raee.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!

FAILED: CMakeFiles/latex C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/CMakeFiles/latex
C:\Windows\system32\cmd.exe /C "cd /D C:\_ddm\ZEPHYR\zephyr_main\zephyr\doc\_build && C:\Strawberry\c\bin\cmake.exe -E env "DOXYGEN_EXECUTABLE=C:/Program Files/doxygen/bin/doxygen.exe" DOT_EXECUTABLE=C:/ProgramData/chocolatey/bin/dot.exe DOCS_HTML_DIR=C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/html OUTPUT_DIR=C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/latex C:/Users/nxa12205/AppData/Local/Programs/Python/Python310/Scripts/sphinx-build.exe -b latex -c C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc -d C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/doctrees -w C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/latex.log -t development -t convertimages -j auto -W --keep-going -T C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/src C:/_ddm/ZEPHYR/zephyr_main/zephyr/doc/_build/latex"
ninja: build stopped: subcommand failed.
```

fabiobaltieri, 2025-03-06:
@kartben should we even try to support pdf generation on windows? It's already a miracle that it works on Linux...

kartben, 2025-03-06:
It works fine for me as long as I disable "fail on warnings" (the -W flag in doc/Makefile).
I would recommend making sure you have up-to-date dependencies https://docs.zephyrproject.org/latest/contribute/documentation/generation.html#installing-the-documentation-processors

fabiobaltieri, 2025-03-06:
Ok tentatively closing then

closed 2025-03-06 by fabiobaltieri