zephyrproject-rtos/zephyr #73203: drivers: i2c: i2c_ram: drivers.i2c.ram{.rtio} fail

katgiadla, 2024-05-23:
**Describe the bug**
Tests:
- `drivers/i2c/i2c_ram/drivers.i2c.ram`
- `drivers/i2c/i2c_ram/drivers.i2c.ram.rtio`
fails. 

Observed for `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/i2c/i2c_ram -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-4458-g17a5beb34124 ***
Running TESTSUITE i2c_ram
===================================================================
START - test_ram_transfer
ram using i2c_transfer from thread 0x20000108 addr 7
Assertion failed at ../../../../../../../zephyr/tests/drivers/i2c/i2c_ram/src/test_i2c_ram.c:98: i2c_ram_test_ram_transfer: (i2c_transfer(i2c_dev, msgs, 1, RAM_ADDR) is non-zero)
I2C write to fram failed
FAIL - test_ram_transfer in 0.524 seconds
===================================================================
START - test_ram_write_read
ram using i2c_write and i2c_write_read from thread 0x20000108 addr e
Assertion failed at ../../../../../../../zephyr/tests/drivers/i2c/i2c_ram/src/test_i2c_ram.c:121: i2c_ram_test_ram_write_read: (i2c_write(i2c_dev, tx_data, ARRAY_SIZE(tx_data), RAM_ADDR) is non-zero)
I2C write to fram failed
FAIL - test_ram_write_read in 0.027 seconds
===================================================================
TESTSUITE i2c_ram failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [i2c_ram]: pass = 0, fail = 2, skip = 0, total = 2 duration = 0.551 seconds
- FAIL - [i2c_ram.test_ram_transfer] duration = 0.524 seconds
- FAIL - [i2c_ram.test_ram_write_read] duration = 0.027 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 2a8ef7dddeb19365d97206d8f7e0cd9f
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: `v3.6.0-4458-g17a5beb34124`

teburd, 2024-05-23:
This test has requirements around hardware connected to the device, and likely I need to add a test fixture to signal that properly. Let me deal with that.

katgiadla, 2024-05-29:
@teburd any updates?

PR #73476 (2024-06-03): tests: i2c_ram: Add fixture for i2c_ram testing
i2c_ram expects that a ram-like part is connected to an i2c bus and therefore requires a hardware fixture setup much like spi_loopback.

Adds a fixture that must be designated when running twister tests on a board with -X i2c_ram.

Fixes #73203