zephyrproject-rtos/zephyr #73216: How to set  the values of CONFIG_ROM_END_OFFSET and CONFIG_MCUBOOT_UPDATE_FOOTER_SIZE?

NamrataGagare-eaton, 2024-05-23:
Hello All,

I am compiling zephyr 3.6 with my application code with below compilation cmd:
west -v build -b nucleo_h743zi -p --sysbuild -- -DCONFIG_BLR_TEST_NV_RESET_KEY=1701686850 -DCONFIG_TEST_FUNC_BLR_WEBSERVER=y  -DCONFIG_BLR_CRYPT_CERT_MCUBOOT_SHARED_REGION=y -DCONFIG_BLR_TEST_FUNC_MODBUS_RTU_SERVER_SOFT_DMA=y -DCONFIG_BLR_IP_ALLOCATION_ALWAYS_DHCP=y -DCONF_FILE="prj_functional.conf;certificate.conf" -DCONFIG_NEWLIB_LIBC_NANO=y

I see the value for CONFIG_ROM_END_OFFSET and CONFIG_MCUBOOT_UPDATE_FOOTER_SIZE flags are taken from CMakeCache.list which are calculated below in C:\blr_deps\zephyr-v3.6.0\blr_modules\mcuboot\boot\zephyr\CMakeLists.txt 
  set(mcuboot_image_footer_size ${required_size} CACHE INTERNAL "Estimated MCUboot image trailer size" FORCE) 
  set(mcuboot_image_upgrade_footer_size ${required_upgrade_size} CACHE INTERNAL "Estimated MCUboot up

I wanted to keep CONFIG_ROM_END_OFFSET and CONFIG_MCUBOOT_UPDATE_FOOTER_SIZE values to be configured through Kconfig or prj.conf, but it is not taking any effect when I set these value in prj.conf or set them using -D in compilation cmd.

I kindly request your inputs on above issue.

Thanks in advance.

RamPingle-eaton, 2024-05-24:
@michaeljones It would great if you could help me on this?

michaeljones, 2024-05-24:
I can only assume you mean someone else as this means nothing to me.