zephyrproject-rtos/zephyr #73224: Handle LwM2M requests asynchronously

koalatux, 2024-05-23:
**Is your enhancement proposal related to a problem? Please describe.**
Handling of LwM2M requests is done in a blocking way. While handling a request no other requests can be handled at that time. To build a response we would like to query information from an external component, which can take some seconds. We would like to handle other LwM2M requests during that time.

**Describe the solution you'd like**
In a request callback we would like to store the LwM2M request context away, send an empty CoAP ACK and tell the LwM2M engine it should not send a response upon returning of the callback. Later we want to send the response separately from a different thread using the previously stored request context.

**Describe alternatives you've considered**
We considered calling `lwm2m_acknowledge()` to send an empty CoAP ACK and then block the LwM2M thread until we have the final response. While this somewhat lowers the pain it still does not allow handling of other LwM2M messages nor does it send repeated acknowledge messages for receiving duplicate requests in case the first acknowledge got lost.  

**Additional context**
None

boaks, 2024-08-06:
At least in the earlier LwM2M TS (1.0) the recommendation was to use "always an empty token", which will not work with parallel pending requests from the same LwM2M server. I guess, pretty much are using none-empty tokens, so in practice that may work. I would consider to add then some notes on that.

boaks, 2025-07-14:
As I wrote above, my lwm2m time was long ago.

"Academically", defining "(for example if it has a non-empty token)" may be considered as non RFC 7252 compliant. 

[5.3.1. Token](https://www.rfc-editor.org/rfc/rfc7252#section-5.3.1)

>    An endpoint receiving a token it did not generate MUST treat the
>    token as opaque and make no assumptions about its content or
>    structure.

Also, if the device trigger a time consuming function, I would at least mention/recommend a timeout for that.

SeppoTakalo, 2025-07-15:
@rlubos my thoughts on this.

First, I think the API conceptually looks OK. Only the application really knows if certain resources require longer processing time. If we do this, maybe `lwm2m_acknowledge()` should be dropped from the public API as it aims to do the same, but not proberly. Requesting a message to be handled asynchronously should automatically trigger the CoAP Ack internally and postpone further processing. 

Secondly, all callbacks registered to a resource, should be able to pause the processing of the message. So the functionality should be equivalent for pre-write, validate, post-write, read, delete and execute callbacks. Any of those might require a longer processing time if they trigger an event on external system.

Thirdly, request-async and send-response pair sounds a bit off, as this postponing the processing of certain incoming request from server, until the application is ready to continue. However, withing the callback, the application (or engine) does not know the remaining content of the message. So if there is write command for multiple resources, all of them might request pause on processing. So the API that instructs the engine to continue on this message should probably be like `lwm2m_continue_something()`. I'm not yet sure of proper naming, but I hope you get the point. Pause-resume or postpone-continue, or similar.

Last point, If we allow any callbacks to request a processing delay, should it be and known return code for callbacks? For example, returning a negative error, is now automatically sending a CoAP error as a response and returning a zero is OK. But we could define for example that returning a positive `EINPROGRESS` or similar, would stop the processing of this message. Maybe easier to refactor the `lwm2m_message_handling.c` this way, but I don't how to deal with the given message handle then. Maybe it should be kept in the `struct lwm2m_ctx *` so only a one new API would be required `lwm2m_continue_processing()`.

rlubos, 2025-07-15:
Thanks @SeppoTakalo for the feedback, I need to rethink my initial proposal based on this - I'll investigate how viable it would be to pause/resume message processing.

> First, I think the API conceptually looks OK. Only the application really knows if certain resources require longer processing time. If we do this, maybe lwm2m_acknowledge() should be dropped from [...]

I personally wouldn't be so hasty on removing `lwm2m_acknowledge()`, it should generally be easier to use than the new mechanism, so if some app does not care about delays in message processing it might be preferable option.

> Secondly, all callbacks registered to a resource, should be able to pause the processing of the message. So the functionality should be equivalent for pre-write, validate, post-write, read, delete [...]

Ok, I had initially only reading in mind, but it seems reasonable to extend this to other ops as well. Will keep that in mind.

> Thirdly, request-async and send-response pair sounds a bit off, as this postponing the processing of certain incoming request from server, until the application is ready to continue. However, [...]

Yeah, I'd need to use different naming then. Originally I was thinking (with read in mind) that the engine would do a double-pass on the message, but that wouldn't be a good idea for writes for instance. I'll investigate further how viable it is to pause in the middle of processing instead.

> Last point, If we allow any callbacks to request a processing delay, should it be and known return code for callbacks? For example, returning a negative error, is now automatically sending a CoAP [...]

The thing is, some callback return codes (ints) and some return pointers - that's why I needed the function to "request async processing". The idea was to simply ignore the result from the callback in case processing was halted.

rlubos, 2025-07-15:
> > The thing is, some callback return codes (ints) and some return pointers - that's why I needed the function to "request async processing".
> 
> This is true, however, I would prefer that those callbacks that are supposed to give you a memory pointer, would still be able to fail and return an error code (instead of NULL).
> 
> So maybe it is a time to refactor a bit.
> 
> And, if we refactor, maybe it is a time to change callbacks to have `struct lwm2m_obj_path *path` instead of one to three uint16_t values to indicate which resource instance the call is pointed to.

After some thinking I think we'd still need some function to request async processing, so that the application has some handle it can use to resume the processing. So far the idea was to return a pointer to the `lwm2m_msg` struct (which would be opaque to the app), which would then be passed to the engine by the app when resuming. Otherwise I think we'd be limited to process only a single message asynchronously.