zephyrproject-rtos/zephyr #73234: mimxrt1170_evk/mimxrt1176/cm7 Flashing bug !!

mellila-abd, 2024-05-23:
**Describe the bug**

Zephyr version 3.6 puts the card in a state where it can no longer be flashed a second time. 
When the card is flashed a first time with this new version, the code works correctly, but this 
puts the card in a state where a second flash becomes impossible.

- message error from pyocd gdb server : 
**Unhandled exception in handle_message (b'v'): flash init failure (result code 0x1) [gdbserver]**

- message error from gdb : 
**Error finishing flash operation**

More details  : 
**arm-none-eabi-gdb -q  build/zephyr/zephyr.elf** 
Reading symbols from build/zephyr/zephyr.elf...
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
__enable_irq () at /home/abderram/zephyr3.6/modules/hal/cmsis/CMSIS/Core/Include/cmsis_gcc.h:951
Python Exception <class 'ModuleNotFoundError'>: No module named 'gdb.styling'
951	  __ASM volatile ("cpsie i" : : : "memory");
(gdb) load
Loading section rom_start, size 0x23a8 lma 0x30000000
Loading section text, size 0x4858 lma 0x300023a8
Loading section .ARM.exidx, size 0x8 lma 0x30006c00
Loading section initlevel, size 0xb8 lma 0x30006c08
Loading section device_area, size 0x154 lma 0x30006cc0
Loading section sw_isr_table, size 0x6d0 lma 0x30006e14
Loading section rodata, size 0x2828 lma 0x300074e4
Loading section datas, size 0x1c lma 0x30009d0c
Loading section device_states, size 0x22 lma 0x30009d28
Loading section .last_section, size 0x4 lma 0x30009d4a
**Error finishing flash operation**
**(gdb) bt**
#0  __enable_irq ()
    at /home/abderram/zephyr3.6/modules/hal/cmsis/CMSIS/Core/Include/cmsis_gcc.h:951
#1  arch_cpu_idle () at /home/abderram/zephyr3.6/zephyr/arch/arm/core/cortex_m/cpu_idle.c:96
#2  0x300069c6 in k_cpu_idle () at /home/abderram/zephyr3.6/zephyr/include/zephyr/kernel.h:5868
#3  idle (unused1=<optimized out>, unused2=<optimized out>, unused3=<optimized out>)
    at /home/abderram/zephyr3.6/zephyr/kernel/idle.c:89
#4  0xa28a4f2a in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

 - What target platform are you using?
 mimxrt1170_evk/mimxrt1176/cm7 
 
 - What have you tried to diagnose or workaround this issue?
The changing of the device tree  

**To Reproduce**

1. build a sample programme from zephyr : 
" west build -p always -b mimxrt1170_evk/mimxrt1176/cm7 samples/hello_world/ "
2. open a server with pyocd : 
"pyocd gdbserver"
3. flasg the programme 
 - arm-none-eabi-gdb -q  build/zephyr/zephyr.elf
 - target remote localhost:3333
 - load
4. it will work with no problem, then try to re flash the board again : 
5. See error : 
Error finishing flash operation. in gdb 

to reuse the card : 
Change the boot mode by changing the switch values according to the attached table. Flash an uncorrupted program : " return to the old version of zephyr " the 3.5". Then reset the initial boot mode and you can reflash your card normally using this old version with no problems.

dleach02, 2024-05-28:
This appears to be a duplicate of #70271, if you disagree please reopen the issue

closed 2024-05-28 by dleach02

mellila-abd, 2024-05-30:
This is indeed a distinct issue:

In the previously mentioned issue, although an error is reported during the verification phase, the image is successfully loaded. However, in the issue I've opened, the problem is that after flashing the board once with Zephyr version 3.6, subsequent flashes become impossible. Even though the last flashed image still works, it's not feasible to load a new program. It's as if communication with the CPU is disrupted, possibly due to JTAG deactivation or interference.

Regarding the issue I've raised:

**Describe the Bug:**

Zephyr version 3.6 causes the board to enter a state where it cannot be flashed a second time. Initially, flashing with this version works fine, but subsequent attempts fail. 

**Error Messages:**

**From pyocd gdb server**
```shell
Unhandled exception in handle_message (b'v'): flash init failure (result code 0x1) [gdbserver]
``` 
  **From gdb:**
```shell
Error finishing flash operation
``` 

**Workaround:**
To reuse the card I should :
- Change the boot mode by adjusting the switch values.
- Flash an uncorrupted program using an older version of Zephyr (e.g., version 3.5).
- Reset the initial boot mode.
- Now, I can reflash the card normally using the older version without any problems.

However, this workaround doesn't work with Zephyr 3.6.

dleach02, 2024-05-30:
@danieldegrasse can you review the details here based on the additional information @mellila-abd provided?

danieldegrasse, 2024-05-31:
> Workaround:
> To reuse the card I should :
>
>     Change the boot mode by adjusting the switch values.
>     Flash an uncorrupted program using an older version of Zephyr (e.g., version 3.5).
>     Reset [...]

To be clear here, can you reflash another Zephyr 3.6 image when you adjust the boot mode using the DIP switches on the board, and have that image run fine? Futhermore, are you running Zephyr 3.6, or the latest revision of `main` to reproduce this issue? Specifically, is the commit e4e463af812e6 present in your Zephyr tree? This is the commit that causes the flashing issue with JLink I was referring to, so if you are working on a version of Zephyr before that commit this would not be the root cause.

DerekSnell, 2024-05-31:
Hi @mellila-abd and @danieldegrasse ,
I can replicate the issue with pyOCD, and it is indeed the same issue as reported in https://github.com/zephyrproject-rtos/zephyr/issues/70271.  The HAL update to v2.15 in https://github.com/zephyrproject-rtos/zephyr/commit/e4e463af812e61bb5bfb4b1087d48e1dbcd9d8d5 breaks pyOCD.  The symptoms are different because of the different debug probes: pyOCD or JLink.  But it is the same root cause of the FCB update for booting from flash.

This issue is not specific to Zephyr.  I can replicate the same issue with pyOCD using an example app from the [MCUXpresso SDK](https://github.com/nxp-mcuxpresso/mcux-sdk) v2.15.  pyOCD needs to be updated to support the new FCB.  I created this pyOCD issue https://github.com/pyocd/pyOCD/issues/1698.

I am removing the bug label for this Zephyr issue.  And I think we should close this Zephyr issue here, since this is not specific to Zephyr, and requires an update in pyOCD.

Thanks

dleach02, 2024-05-31:
not sure how we are going to get an update to pyOCD?

danieldegrasse, 2024-05-31:
> not sure how we are going to get an update to pyOCD?

We can change the version required in [requirements.txt](https://github.com/zephyrproject-rtos/zephyr/blob/a4b37c7947fa93c80111baff778c1bb6ea88cfa9/scripts/requirements-run-test.txt#L6)

dleach02, 2024-06-03:
what I meant was who would make the change to pyOCD