zephyrproject-rtos/zephyr #73258: tests/modules/uoscore/libraries.uoscore fails to build

aescolar, 2024-05-24:
**Describe the bug**
tests/modules/uoscore/libraries.uoscore fails to build in multiple platforms causing main's CI job to fail

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=native_sim ../tests/modules/uoscore/ 
3. ninja
4. See error

**Expected behavior**
No build errors or warnings

**Impact**
main CI failing. PRs triggering this test blocked

**Logs and console output**
```
/__w/zephyr/modules/crypto/mbedtls/library/psa_crypto_cipher.c: In function 'mbedtls_cipher_info_from_psa':
/__w/zephyr/modules/crypto/mbedtls/library/psa_crypto_cipher.c:273:20: error: 'cipher_id_tmp' may be used uninitialized [-Werror=maybe-uninitialized]
  273 |         *cipher_id = cipher_id_tmp;
      |         ~~~~~~~~~~~^~~~~~~~~~~~~~~
/__w/zephyr/modules/crypto/mbedtls/library/psa_crypto_cipher.c:266:25: note: 'cipher_id_tmp' declared here
  266 |     mbedtls_cipher_id_t cipher_id_tmp;
      |                         ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```
```
In file included from modules/crypto/mbedtls/library/pk_wrap.c:14:
modules/crypto/mbedtls/library/pk_internal.h: In function ‘mbedtls_pk_get_ec_group_id’:
modules/crypto/mbedtls/library/pk_internal.h:101:14: warning: implicit declaration of function ‘mbedtls_ecc_group_from_psa’; did you mean ‘mbedtls_pk_copy_from_psa’? [-Wimplicit-function-declaration]
  101 |         id = mbedtls_ecc_group_from_psa(curve, psa_get_key_bits(&opaque_attrs));
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |              mbedtls_pk_copy_from_psa
In file included from modules/crypto/mbedtls/library/pk_internal.h:23:
modules/crypto/mbedtls/library/pk_wrap.c: In function ‘rsa_verify_wrap’:
modules/crypto/mbedtls/library/pk_internal.h:26:67: error: ‘psa_to_pk_rsa_errors’ undeclared (first use in this function); did you mean ‘psa_to_ssl_errors’?
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro ‘PSA_TO_MBEDTLS_ERR_LIST’
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c:116:15: note: in expansion of macro ‘PSA_PK_RSA_TO_MBEDTLS_ERR’
  116 |         ret = PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/pk_internal.h:26:67: note: each undeclared identifier is reported only once for each function it appears in
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro ‘PSA_TO_MBEDTLS_ERR_LIST’
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c:116:15: note: in expansion of macro ‘PSA_PK_RSA_TO_MBEDTLS_ERR’
  116 |         ret = PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c: In function ‘mbedtls_pk_psa_rsa_sign_ext’:
modules/crypto/mbedtls/library/pk_internal.h:26:67: error: ‘psa_to_pk_rsa_errors’ undeclared (first use in this function); did you mean ‘psa_to_ssl_errors’?
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro ‘PSA_TO_MBEDTLS_ERR_LIST’
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c:213:15: note: in expansion of macro ‘PSA_PK_RSA_TO_MBEDTLS_ERR’
  213 |         ret = PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c: In function ‘rsa_decrypt_wrap’:
modules/crypto/mbedtls/library/pk_internal.h:26:67: error: ‘psa_to_pk_rsa_errors’ undeclared (first use in this function); did you mean ‘psa_to_ssl_errors’?
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro ‘PSA_TO_MBEDTLS_ERR_LIST’
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c:329:15: note: in expansion of macro ‘PSA_PK_RSA_TO_MBEDTLS_ERR’
  329 |         ret = PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c: In function ‘rsa_encrypt_wrap’:
modules/crypto/mbedtls/library/pk_internal.h:26:67: error: ‘psa_to_pk_rsa_errors’ undeclared (first use in this function); did you mean ‘psa_to_ssl_errors’?
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro ‘PSA_TO_MBEDTLS_ERR_LIST’
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c:412:15: note: in expansion of macro ‘PSA_PK_RSA_TO_MBEDTLS_ERR’
  412 |         ret = PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c: In function ‘eckey_check_pair_psa’:
modules/crypto/mbedtls/library/pk_wrap.c:967:9: warning: implicit declaration of function ‘mbedtls_ecc_group_to_psa’; did you mean ‘mbedtls_ecp_group_copy’? [-Wimplicit-function-declaration]
  967 |         mbedtls_ecc_group_to_psa(mbedtls_pk_ec_ro(*prv)->grp.id, &curve_bits);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
      |         mbedtls_ecp_group_copy
modules/crypto/mbedtls/library/pk_wrap.c: In function ‘rsa_opaque_sign_wrap’:
modules/crypto/mbedtls/library/pk_internal.h:26:67: error: ‘psa_to_pk_rsa_errors’ undeclared (first use in this function); did you mean ‘psa_to_ssl_errors’?
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro ‘PSA_TO_MBEDTLS_ERR_LIST’
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
modules/crypto/mbedtls/library/pk_wrap.c:1536:20: note: in expansion of macro ‘PSA_PK_RSA_TO_MBEDTLS_ERR’
 1536 |             return PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
[218/232] Building C object modules/mbedtls/CMakeFiles/modules__mbedtls.dirmodules/crypto/mbedtls/library/ssl_tls.c.obj
In file included from modules/crypto/mbedtls/library/x509_internal.h:18,
                 from modules/crypto/mbedtls/library/ssl_misc.h:48,
                 from modules/crypto/mbedtls/library/ssl_client.h:14,
                 from modules/crypto/mbedtls/library/ssl_tls.c:19:
modules/crypto/mbedtls/library/pk_internal.h: In function ‘mbedtls_pk_get_ec_group_id’:
modules/crypto/mbedtls/library/pk_internal.h:101:14: warning: implicit declaration of function ‘mbedtls_ecc_group_from_psa’; did you mean ‘mbedtls_pk_copy_from_psa’? [-Wimplicit-function-declaration]
  101 |         id = mbedtls_ecc_group_from_psa(curve, psa_get_key_bits(&opaque_attrs));
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |              mbedtls_pk_copy_from_psa
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
 - OS: Linux CI (Ubuntu 22.04)
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.16.5 or host gcc 12
 - Commit SHA or Version used: Main as of now 8c8e2f6243f147953ce3fadb999c4e2a5faf4f5e

**Additional context**

aescolar, 2024-05-24:
CC @valeriosetti @ceolin

valeriosetti, 2024-05-24:
#73243 looks like a fix. Does it?

PR #73243 (2024-05-24): tests: uoscore: Set necessary PSA options
Set necessary psa options to the test execution.

Edit by @aescolar 
Fixes #73258
closed 2024-05-24 by aescolar