zephyrproject-rtos/zephyr #73281: mqtt_azure example uses the Baltimore certificate

uriel-kluk, 2024-05-24:
### Summary
_Azure IoT Hub and Device Provisioning Service (DPS) use TLS certificates issued by the Baltimore CyberTrust Root, which expires in 2025. Starting in February 2023, all IoT hubs in the global Azure cloud will migrate to a new TLS certificate issued by the DigiCert Global Root G2._

### Details
_The example provided conatins a file named `digicert.cer`, but it contains the `Baltimore CyberTrust Root`. The solution should include `DigiCert Global Root G2` instead._

### Reproduce

1. Run the example as is, entering the username, password, and hostname in the configuration.
2. Observe the following error:

```text
[00:00:11.336,000] <dbg> mqtt_azure: try_to_connect: attempting to connect...
[00:00:11.401,000] <err> net_sock_tls: Failed to parse CA certificate, err: -0xfffffffe
```

3. Replace the certificate using the certificates provided [here](https://github.com/Azure/azure-iot-sdk-c/blob/main/certs/certs.c).
4. Observe that the same error persists.
5. Change the certificate to include only the DigiCert Global Root G2.
6. Observe the new error:
```text
[00:00:16.308,000] <dbg> mqtt_azure: try_to_connect: attempting to connect...
[00:00:16.420,000] <err> net_sock_tls: TLS handshake error: -0x7280
[00:00:16.430,000] <err> mqtt_azure: mqtt_connect failed -113
```

### Impact
It might be my configuration that is failing, but if developers base their code on the example provided, devices will stop connecting in 2025.

uriel-kluk, 2024-05-28:
> Thanks for the report, it seems that for the time being it'd be most reasonable to support both certificates.
>
> As for the issues with Baltimore CyberTrust Root certificate registration, there's been a lot of happening recently in mbed TLS area, and it seems one of the PRs (#72078<https://github.com/zephyrproject-rtos/zephyr/pull/72078> specifically) might've affected the sample. Baltimore CyberTrust Root uses sha1WithRSAEncryption as a signature algorithm, and SHA1 is no longer enabled by default. Can you check if it works for you after enabling CONFIG_MBEDTLS_MAC_SHA1_ENABLED=y in the prj.conf file?
>
> As for the failures with DigiCert Global Root G2, I guess the Azure transition is not complete, so it could be the case that we'd still need Baltimore for your case?
>
> Feel free to send a PR if it helps in your case. We could also add the second certificate if such a configuration works for you.

Thanks Robert,

Here is the PR, but heads up, it is having trouble parsing the CA certificates:
Add SHA1 to project configuration and added new certificates to conne… by uriel-kluk · Pull Request #73426 · zephyrproject-rtos/zephyr (github.com)<https://github.com/zephyrproject-rtos/zephyr/pull/73426>

You should reject the PR

rlubos, 2024-05-31:
I've opened https://github.com/zephyrproject-rtos/zephyr/pull/73557 with the fixes discussed.

uriel-kluk, 2024-05-31:
Great job! Thanks

Sent from my Verizon, Samsung Galaxy smartphone

PR #73557 (2024-06-04): samples: net: cloud: mqtt_azure: Update sample certificates 
Baltimore certificate expires in 2025 and Azure is in the transition period for the time being to the new certificates. Add support for all certificates needed in the sample.

Update mbed TLS configuration as needed.

Fixes #73281
closed 2024-06-04 by henrikbrixandersen