zephyrproject-rtos/zephyr #73287: After enabling CONFIG_SHELL=y And CONFIG_SHELL_BACKEND_RPMSG=y in prj.conf, I get the Flash overflow error

neuberfran, 2024-05-25:
Hi,

I stay trying this: https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/ipc/openamp_rsc_table in my imx7d-pico (m4 side, board in zephyr = pico_pi_m4)

https://github.com/zephyrproject-rtos/zephyr/issues/69433#issuecomment-2016598955
https://www.youtube.com/watch?v=V06RC2G_3M8

I'm migrating from Version 3.4 of Zephyr to 3.6.99 so I can access /dev/ttyRPMSG via Linux/Yocto through minicom -D /dev/ttyRPMSG0. Unsuccessful because I need to implement CONFIG_SHELL=y And CONFIG_SHELL_BACKEND_RPMSG=y

When I enable these options in my prj.conf I get the following error:

```
neuberfran@neuberfran2204:~/zephyrproject/zephyr$ west build --sysbuild -b pico_pi_m4 samples/subsys/ipc/openamp_rsc_table
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): extensions sysbuild_extensions python west root zephyr_module boards shields hwm_v2 sysbuild_kconfig native_simulator_sb_extensions
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/neuberfran/.cache/zephyr
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")

-- Board: pico_pi, qualifiers: mcimx7d/m4
Parsing /home/neuberfran/zephyrproject/zephyr/share/sysbuild/Kconfig
Loaded configuration '/home/neuberfran/zephyrproject/zephyr/build/empty.conf'
Merged configuration '/home/neuberfran/zephyrproject/zephyr/build/empty.conf'
Merged configuration '/home/neuberfran/zephyrproject/zephyr/build/empty.conf'
Configuration saved to '/home/neuberfran/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/neuberfran/zephyrproject/zephyr/build/autoconf.h'
-- 
   ***************************************
   * Running CMake for openamp_rsc_table *
   ***************************************

Loading Zephyr default modules (Zephyr base).
-- Application: /home/neuberfran/zephyrproject/zephyr/samples/subsys/ipc/openamp_rsc_table
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/neuberfran/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/neuberfran/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: pico_pi, qualifiers: mcimx7d/m4
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/neuberfran/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (/home/neuberfran/zephyr-sdk-0.16.5-1)
-- Found Dtc: /home/neuberfran/zephyr-sdk-0.16.5-1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/neuberfran/zephyrproject/zephyr/boards/technexion/pico_pi/pico_pi_mcimx7d_m4.dts
-- Found devicetree overlay: /home/neuberfran/zephyrproject/zephyr/samples/subsys/ipc/openamp_rsc_table/boards/pico_pi_m4.overlay
'label' is marked as deprecated in 'properties:' in /home/neuberfran/zephyrproject/zephyr/dts/bindings/ipm/nxp,imx-mu.yaml for node /soc/mu@30ab0000.
-- Generated zephyr.dts: /home/neuberfran/zephyrproject/zephyr/build/openamp_rsc_table/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/neuberfran/zephyrproject/zephyr/build/openamp_rsc_table/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/neuberfran/zephyrproject/zephyr/build/openamp_rsc_table/zephyr/dts.cmake

warning: OPENAMP_WITH_DCACHE (defined at modules/Kconfig.open-amp:35) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: CACHE_MANAGEMENT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_OPENAMP_WITH_DCACHE and/or look up
OPENAMP_WITH_DCACHE in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: PRINTK (defined at subsys/debug/Kconfig:204) was assigned the value 'n' but got the value
'y'. See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_PRINTK and/or look up PRINTK in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: LOG_PRINTK (defined at subsys/logging/Kconfig.processing:8) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: LOG (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_LOG_PRINTK and/or look up LOG_PRINTK in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

Parsing /home/neuberfran/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/neuberfran/zephyrproject/zephyr/boards/technexion/pico_pi/pico_pi_mcimx7d_m4_defconfig'
[... 87 lines cut ...]
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_CONSOLE=y
CONFIG_LOG_PRINTK=y
CONFIG_IPM_IMX_MAX_DATA_SIZE_16=n
CONFIG_IPM_IMX_MAX_DATA_SIZE_4=y

CONFIG_OPENAMP_WITH_DCACHE=y
```
**prj.conf**

```
CONFIG_KERNEL_BIN_NAME="zephyr_openamp_rsc_table"
CONFIG_PRINTK=n
CONFIG_IPM=y
CONFIG_MAIN_STACK_SIZE=1024
CONFIG_HEAP_MEM_POOL_SIZE=1024
CONFIG_OPENAMP=y
CONFIG_OPENAMP_RSC_TABLE_NUM_RPMSG_BUFF=32
CONFIG_OPENAMP_RSC_TABLE=y
CONFIG_OPENAMP_MASTER=n

CONFIG_SHELL=y
CONFIG_SHELL_BACKEND_RPMSG=y
```
How can I solve it  (Pls)

neuberfran, 2024-05-27:
@pdgendt  https://discord.com/channels/720317445772017664/1243652018086613082

**Why do you build for sysbuild?**
I still don't know, but I always used this tag

aescolar, 2024-05-28:
Converting to a discussion as this is not a bug