zephyrproject-rtos/zephyr #73296: ICM42688 is not configurable from DTS and gyro does not work with sensor shell

bperseghetti, 2024-05-25:
**Describe the bug**

Please also mention any information which could help others to understand
the problem you're facing:
 - Using platform VMU_RT1170
 - Unable to set params for ICM42688 from dts (just overrides with fixed values at lowest settings that clip in most use cases): https://github.com/zephyrproject-rtos/zephyr/blob/68bd4da4c64ca0c19cee0334bee6496ec8cde2b9/drivers/sensor/tdk/icm42688/icm42688.c#L274-L286

Also sensor shell does not work properly for gyro:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to set sensor attributes with DTS for ICM42688
2. Test shaking and see it locks to 125 deg/sec (hard coded gyro FS)
3. Run sensor shell and see it does not output gyro 

**Expected behavior**
Sensor should be configurable and sensor shell should work for the gyro.

**Impact**
Makes sensor not work as expected.

teburd, 2024-05-25:
Thanks for the report, I’ll try and take a look next week

teburd, 2024-06-13:
The shell issue seems tied to the way the sensor_shell currently works, and @ycsin noted issues as well in #72841 and related PRs/issues. I'm looking at this, and it is a bug but I don't think its solely tied to this driver. Notably if I force using the fetch + channel_get API with some Kconfig hacks the gyro works mostly fine

ycsin, 2024-06-14:
Should be the same as #72840

teburd, 2024-06-18:
Not quite fixed, needs #74267 still

teburd, 2024-06-26:
Fixed as #74267 is merged

PR #74335 (2024-06-18): sensors: Fix shell missing channel types
The shell tries to decode all channel types and sequential channel indexes. Unfortunately when swapping to channel specs we forgot to reset the channel index for the next channel type to look at.

Fixes #73296
closed 2024-06-26 by teburd