zephyrproject-rtos/zephyr #73301: mcuboot: nordic development kits fail to compile

JordanYates, 2024-05-26:
**Describe the bug**

The following Nordic development kits fail to compile the upstream mcuboot application.
 * nrf9161dk@0.7.0/nrf9161
 * thingy52/nrf52832
 * thingy53/nrf5340/cpuapp

Enabling multithreading by removing the following line lets them all compile:
https://github.com/zephyrproject-rtos/mcuboot/blob/d4394c2f9b76e0a7b758441cea3a8ceb896f66c8/boot/zephyr/Kconfig#L751

**To Reproduce**

```
west build -b nrf9161dk@0.7.0/nrf9161 bootloader/mcuboot/boot/zephyr/ -p
west build -b thingy52/nrf52832 bootloader/mcuboot/boot/zephyr/ -p
west build -b thingy53/nrf5340/cpuapp bootloader/mcuboot/boot/zephyr/ -p
```

**Expected behavior**

Development kits should successfully compile the standard Zephyr bootloader

**Environment (please complete the following information):**
 - Zephyr v3.6.99 ea29ef36bbedeacd171199d6feea5e699fa64447

nashif, 2024-06-04:
why is this a zephyr bug? Please report at nordic NCS.

JordanYates, 2024-06-05:
> why is this a zephyr bug? Please report at nordic NCS.

Why is this a NCS bug? It's a Zephyr bug because these are Zephyr supported platforms failing to compile the Zephyr bootloader.

nordicjm, 2024-06-21:
Fix: https://github.com/mcu-tools/mcuboot/pull/1984

closed 2024-07-11 by aescolar

JordanYates, 2024-07-19:
I am now discovering the situation is worse for me with the fix than it was without it :)
Before I could simply set `mcuboot_CONFIG_MULTITHREADING=y` in my sysbuild config and everything would be fine.
Now I have no board agnostic way to enable the external flash drivers in MCUboot...

nordicjm, 2024-07-19:
> I am now discovering the situation is worse for me with the fix than it was without it :) Before I could simply set `mcuboot_CONFIG_MULTITHREADING=y` in my sysbuild config and everything would be fine. Now I have no board agnostic way to enable the external flash drivers in MCUboot...

it needs to be done on a board by board basis, if you want to do that then I would suggest using sysbuild with your application and having a `sysbuild/mcuboot` folder with your MCUboot files, you can then create a `boards` folder there and have your overlays for each board, or alternatively setting `mcuboot_APPLICATION_CONFIG_DIR` to a common place if you need it across multiple repos/applications.

Example: https://github.com/nrfconnect/sdk-nrf/tree/main/applications/machine_learning

JordanYates, 2024-07-19:
That's just not a scalable solution. Copy pasting files into every application is a poor solution when better solutions exist IMO.
Happy to continue the conversation on this PR and why you think its a bad idea :)
https://github.com/mcu-tools/mcuboot/pull/2015