zephyrproject-rtos/zephyr #73311: infineon: flash driver failing CI

JordanYates, 2024-05-27:
**Describe the bug**

The following test is failing due to HAL misconfiguration, causing CI failures in #73302
```
west build -b cyw920829m2evk_02 tests/drivers/flash/common -T drivers.flash.common.default
```

Cause appears to be that `CYHAL_DRIVER_AVAILABLE_NVM` is not defined, I don't know enough (anything) about Infineon HAL to diagnose further.

**Impact**
CI failures

**Logs and console output**
```
/home/jordan/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ifx_cat1.c.obj): in function `ifx_cat1_flash_erase':
/home/jordan/code/zephyr/zephyr/drivers/flash/flash_ifx_cat1.c:130: undefined reference to `cyhal_nvm_erase'
/home/jordan/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ifx_cat1.c.obj): in function `ifx_cat1_flash_init':
/home/jordan/code/zephyr/zephyr/drivers/flash/flash_ifx_cat1.c:176: undefined reference to `cyhal_nvm_init'
/home/jordan/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ifx_cat1.c.obj): in function `ifx_cat1_flash_write':
/home/jordan/code/zephyr/zephyr/drivers/flash/flash_ifx_cat1.c:96: undefined reference to `cyhal_nvm_write'
/home/jordan/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(flash_ifx_cat1.c.obj): in function `ifx_cat1_flash_read':
/home/jordan/code/zephyr/zephyr/drivers/flash/flash_ifx_cat1.c:63: undefined reference to `cyhal_nvm_read'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/jordan/code/zephyr/zephyr/build
```

**Environment (please complete the following information):**
 - Zephyr 3a3f25c9a683562dcf09232eb754fa544affac3b

aescolar, 2024-05-28:
This problem has been present since the introduction of this target in f96e6ccbc065eef90852ec730bf2b3d440907298

aescolar, 2024-05-28:
CC @sreeramIfx

sreeramIfx, 2024-05-28:
I have pushed a fix (https://github.com/zephyrproject-rtos/zephyr/pull/73377) and closed it since its being updated in https://github.com/zephyrproject-rtos/zephyr/pull/73267 as well. Since this PR is approved it should me merged soon..

JordanYates, 2024-05-28:
> I have pushed a fix (#73377) and closed it since its being updated in #73267 as well. Since this PR is approved it should me merged soon..

That is not going to resolve this issue as it is a different test application.
Simply excluding the platform from individual tests is not going to solve the root cause.

sreeramIfx, 2024-05-29:
I am working on a PR to address the root cause, Please let me know if there is a better way to unblock other PR's in the meanwhile.

tomi-font, 2024-05-30:
Great. For my part just make sure to revert the excluding of the platform in https://github.com/zephyrproject-rtos/zephyr/pull/73267.

aescolar, 2024-06-11:
@sreeramIfx @ifyall What is the status on this issue?

sreeramIfx, 2024-06-11:
There was another issue created for the same error(https://github.com/zephyrproject-rtos/zephyr/issues/73525), This is fixed @ https://github.com/zephyrproject-rtos/zephyr/pull/73552

aescolar, 2024-06-11:
Thanks @sreeramIfx . Closing as duplicate (and fixed)

closed 2024-06-11 by aescolar