zephyrproject-rtos/zephyr #73313: bt_id_reset always fails with CONFIG_BT_SMP disabled

papatel05, 2024-05-27:
With BT_SMP disabled: when creating a new identity with bt_id_create(NULL,NULL) and trying to rotate the private random address using bt_id_reset(<id>,NULL, NULL) the current implementation errors out because bt_unpair will always return -EINVAL

BT_SMP uses a large chunk of flash memory and some implementations may want the ability to manually rotate a static random address for privacy without having the full SMP stack.

If BT_SMP is disabled, could the easy fix just to always return success from bt_unpair since there can't be any possible pairings?

papatel05, 2024-05-27:
Upon further thinking, I suggest just checking for null or BT_ADDR_LE_ANY equivalence before executing unpair and not erroring in the case that it is null, in hci_core.c:

Original:
```
int bt_unpair(uint8_t id, const bt_addr_le_t *addr)
{
	if (id >= CONFIG_BT_ID_MAX) {
		return -EINVAL;
	}

	if (IS_ENABLED(CONFIG_BT_SMP)) {
		if (!addr || bt_addr_le_eq(addr, BT_ADDR_LE_ANY)) {
			bt_foreach_bond(id, unpair_remote, &id);
		} else {
			unpair(id, addr);
		}
	} else {
		CHECKIF(addr == NULL) {
			LOG_DBG("addr is NULL");
			return -EINVAL;
		}

		unpair(id, addr);
	}

	return 0;
}
```

Proposed:
```
int bt_unpair(uint8_t id, const bt_addr_le_t *addr)
{
	if (id >= CONFIG_BT_ID_MAX) {
		return -EINVAL;
	}

	if (IS_ENABLED(CONFIG_BT_SMP)) {
		if (!addr || bt_addr_le_eq(addr, BT_ADDR_LE_ANY)) {
			bt_foreach_bond(id, unpair_remote, &id);
		} else {
			unpair(id, addr);
		}
	} else {
		if(addr && !bt_addr_le_eq(addr, BT_ADDR_LE_ANY)){
			unpair(id, addr);
		}
	}

	return 0;
}
```

This mirrors the case when CONFIG_BT_SMP is enabled...

jhedberg, 2024-05-27:
@papatel05 wouldn't it make more sense to have the condition inside `id.c`, i.e.:
```c
if (IS_ENABLED(CONFIG_BT_SMP)) {
	bt_unpair(...);
}
```

Do you want to submit a pull request for that?

papatel05, 2024-05-27:
Yep! that works too.  Please submit pull request and thank you!!

jhedberg, 2024-05-27:
>> Do you want to submit a pull request for that?

> Yep! that works too. Please submit pull request and thank you!!

So you don't want to submit a pull request? This would be a good opportunity get started with code contributions, if you haven't submitted code previously to Zephyr :)

papatel05, 2024-05-27:
Sorry I misread your initial response.  Yes, I'll submit a pull request tomorrow and thank you for the encouragement :)

jhedberg, 2024-05-27:
> One note: `bt_unpair()` is also supported for Bluetooth Classic (BR/EDR), so theoretically you could have SMP disabled but still have a valid case to call `bt_unpair()`. That said, the identity management is completely tied to LE, so I think it should be safe to make the generalization in that code that we only care about `bt_unpair()` specifically if SMP has been enabled.

...meaning, I think you can simply change the `CONFIG_BT_CONN` references to `CONFIG_BT_SMP` in the conditions prior to calling `bt_unpair()`

PR #73374 (2024-05-31): Bluetooth: Fix bt_id_reset and bt_id_del without SMP
Allows BT ID resets without having the BT_SMP stack enabled. Fixes #73313
commit 131a0fad (2024-05-31): Bluetooth: host: Update id.c to support id rst/del for CONFIG_BT_SMP=n

Currently calls to these two functions fail unnecessarily when
CONFIG_BT_SMP is disabled. This fix allows identity resets
without having the BT_SMP stack enabled. The primary use case
is enabling random mac address rotation for privacy in memory
constrained SOCs. Fixes #73313
closed 2024-05-31 by dleach02