zephyrproject-rtos/zephyr #73337: net: socket: register: net.socket.register.tls fails

katgiadla, 2024-05-27:
**Describe the bug**
The test `net/socket/register/net.socket.register.tls` fails.

Observed for:
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf9160dk/nrf9160/ns` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/net/socket/register -p nrf9160dk/nrf9160/ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
Logs from `build.log`: 

```
arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 25716 bytes
arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(ssl_tls.c.obj): in function `ssl_update_checksum_sha256':
/zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/net/socket/register/net.socket.register.tls/modules/crypto/mbedtls/library/ssl_tls.c:944: undefined reference to `mbedtls_md_error_from_psa'
opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(ssl_tls.c.obj): in function `ssl_update_checksum_sha384':
zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/net/socket/register/net.socket.register.tls/modules/crypto/mbedtls/library/ssl_tls.c:957: undefined reference to `mbedtls_md_error_from_psa'
arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(ssl_tls.c.obj): in function `ssl_calc_finished_tls_generic':
zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/net/socket/register/net.socket.register.tls/modules/crypto/mbedtls/library/ssl_tls.c:8400: undefined reference to `mbedtls_md_error_from_psa'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(ssl_tls.c.obj): in function `ssl_calc_verify_tls_psa':
zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/net/socket/register/net.socket.register.tls/modules/crypto/mbedtls/library/ssl_tls.c:7314: undefined reference to `mbedtls_md_error_from_psa'
opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(ssl_tls.c.obj): in function `ssl_update_checksum_start':
zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/net/socket/register/net.socket.register.tls/modules/crypto/mbedtls/library/ssl_tls.c:927: undefined reference to `mbedtls_md_error_from_psa'
opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(ssl_tls.c.obj):zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/net/socket/register/net.socket.register.tls/modules/crypto/mbedtls/library/ssl_tls.c:876: more undefined references to `mbedtls_md_error_from_psa' follow
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information)**
- OS: Ubuntu 20.04.1 LTS
- Toolchain: Zephyr SDK 0.16.5-1
- Commit SHA or Version used: `v3.6.0-4561-g8c8e2f6243f1`

jukkar, 2024-05-27:
This is a regression after #72243 is merged. @valeriosetti could you take a look why the memory consumption increased and we have undefined symbols in the compilation?

valeriosetti, 2024-05-27:
Sure! Meanwhile can you assign me this issue? So it will be easier to find this in case I accidentally close the page

katgiadla, 2024-05-27:
> Sure! Meanwhile can you assign me this issue? So it will be easier to find this in case I accidentally close the page

I cannot do that. @jukkar, could you assign him?

valeriosetti, 2024-05-28:
I have pushed a fix in #73364. Can you please take a look and let me know it that's OK for you?

katgiadla, 2024-05-28:
> I have pushed a fix in #73364. Can you please take a look and let me know it that's OK for you?

it works on-target.

PR #73364 (2024-05-29): test: net: socket: do not enable USE_PSA and all PSA features
This commit fixes #73337.

Before #72243 Mbed TLS was not using
USE_PSA and all PSA features were not enabled. After #72243 if BUILD_WITH_TFM is set:
- USE_PSA in Mbed TLS is enabled by default and
- all PSA features are enabled.

This commits reverts both changes for `net.socket.register.tls` test case.

**Note**: I was only able to run the test on the `nrf9160dk` board because I don't have the `nrf5340dk` mentioned on the original issue. On `nrf9160dk` the `net.socket.register.tls` is passing now. I would really appreciate if some [... cut]
commit 979ede88 (2024-05-29): test: net: socket: do not enable USE_PSA and all PSA features

This commit fixes #73337.

Before #72243 Mbed TLS was not using
USE_PSA and all PSA features were not enabled. After #72243
if BUILD_WITH_TFM is set:
- USE_PSA in Mbed TLS is enabled by default and
- all PSA features are enabled.

This commits reverts both changes for net.socket.register.tls
test case.
closed 2024-05-29 by carlescufi