zephyrproject-rtos/zephyr #73341: sysbuild: Out Of Tree Board Fails with Sysbuild

vi200031, 2024-05-27:
**Describe the bug**
Project fails to build using Sysbuild with an out of tree board.

This issue was posted and closed in #59114 and #59130

Please also mention any information which could help others to understand
the problem you're facing:
 - This issue was posted and closed in #59114 and #59130, but now I am facing it again.
 - Issue can be bypassed if I copy the custom board to `zephyr/boards/[arm]/ direcotry.
 - Issue occurs only when building with sysbuild. Without `--sysbuild` it works just fine.

**To Reproduce**
Steps to reproduce the behavior:

1. `git checkout main` in `zephyrproject/zephyr` directory
2. Go to `boards/st` copy and paste `nucleo_h743zi` to `samples/hello_world`, and then delete it from `boards/st` to force custom board compilation
3. Go to `samples/hello_world`
4. Run `west build -b nucleo_h743zi -p always --no-sysbuild ./ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=./`
5. See that it compiles. Yay!
6. Run `west build -b nucleo_h743zi -p always --sysbuild ./ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=./`
7.  See error
> If there is a `nucleo_h743zi` in `boards/st` though, the step 6 will compile.

**Expected behavior**
It should compile!

**Impact**
- This prevents mcuboot and application from being built using sysbuild without copying the custom board contents to the zephyr directory. 
- If you are developing an open-source application using zephyr and the user wants to use your software, it is really disturbing to tell them to copy the board directory to some zephyr directory. I also find it not clean to tamper with the zephyr's repository files. Also, the CI will be not clean. 
- A lot of time wasted trying and researching how to fix this on my own.  
- Feeling that the code is not clean, if not resolved, when using sysbuild.

**Logs and console output**
```cpp
(.venv) vic@Super-ThinkPad-X1-Carbon-6th:~/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world$ west build -b nucleo_h743zi -p always --sysbuild ./ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=./
-- west build: making build dir /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world/build pristine
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): extensions sysbuild_extensions python west root zephyr_module boards shields hwm_v2 sysbuild_kconfig native_simulator_sb_extensions
-- Found Python3: /usr/bin/python3 (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/vic/.cache/zephyr
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
CMake Warning at /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/cmake/modules/boards.cmake:136 (message):
  BOARD_ROOT element without a 'boards' subdirectory:

  /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/share/sysbuild/./

  Hints:

    - if your board directory is '/foo/bar/boards/<ARCH>/my_board' then add '/foo/bar' to BOARD_ROOT, not the entire board directory
    - if in doubt, use absolute paths
Call Stack (most recent call first):
  /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:26 (find_package)

No board named 'nucleo_h743zi' found.

Please choose one of the following boards:

.
.
.
segger_trb_stm32f407
esp32s2_franzininho
warp7
scobc_module1
CMake Error at /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/cmake/modules/boards.cmake:244 (message):
  Invalid BOARD; see above.
Call Stack (most recent call first):
  /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world/build -GNinja -DBOARD=nucleo_h743zi -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=./ -S/home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/share/sysbuild -DAPP_DIR:PATH=/home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world

```

While with --no-sysbuild 

```cpp
(.venv) vic@Super-ThinkPad-X1-Carbon-6th:~/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world$ west build -b nucleo_h743zi -p always --no-sysbuild ./ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=./
-- west build: making build dir /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.28.3
-- Found Python3: /usr/bin/python3 (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/vic/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nucleo_h743zi, qualifiers: stm32h743xx
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/home/vic/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (/home/vic/zephyr-sdk-0.16.1)
-- Found Dtc: /home/vic/zephyr-sdk-0.16.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world/boards/arm/nucleo_h743zi/nucleo_h743zi.dts
.
.
.
Generating files from /home/vic/zephyr-sdk-0.16.1/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.elf for board: nucleo_h743zi
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 24.04 LTS
 - Toolchain: Zephyr SDK 0.16.1
 - Zephyr 3.6.99.0

tejlmand, 2024-05-27:
duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/73066

Please note, using absolute paths works.
Only relative paths are failing.

closed 2024-05-27 by tejlmand

tejlmand, 2024-05-28:
Fixed here: #73390