zephyrproject-rtos/zephyr #73345: Stack overflow in 'HCI uart TX' when using 'hci_uart' sample on 'nucleo_wba55cg' board

pepe2k, 2024-05-27:
**Describe the bug**
With latest Zephyr (`08417b4d73d`), the `hci_uart` sample on `nucleo_wba55cg` doesn't work as expected. Shortly after issuing `btattach` on host's side, the controller stops responding. Enabling console drivers and logging on `lpuart1` shows stack overflow in `HCI uart TX` thread:

```
*** Booting Zephyr OS build v3.6.0-4764-g08417b4d73d5 ***
[00:00:00.031,000] <inf> bt_hci_raw: Bluetooth enabled in RAW mode
[00:00:17.806,000] <err> os: ***** USAGE FAULT *****
[00:00:17.806,000] <err> os:   Stack overflow (context area not valid)
[00:00:17.806,000] <err> os: r0/a1:  0x20007080  r1/a2:  0x2000717f  r2/a3:  0xffffffff
[00:00:17.806,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x20007097 r14/lr:  0x0801c77b
[00:00:17.806,000] <err> os:  xpsr:  0x81000200
[00:00:17.806,000] <err> os: Faulting instruction address (r15/pc): 0x080386ba
[00:00:17.806,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:17.806,000] <err> os: Current thread: 0x20006290 (HCI uart TX)
[00:00:17.865,000] <err> os: Halting system
```

**To Reproduce**
Steps to reproduce the behavior:
1. `west update`
2. `west blobs fetch stm32`
3. `west build -b nucleo_wba55cg samples/bluetooth/hci_uart`
4. `west flash`
5. `btattach -B /dev/ttyACM0 -N -S 115200` (on host's side)

**Environment (please complete the following information):**
 - OS: Linux (kernel 6.8.10)
 - Toolchain: Zephyr SDK 0.16.6
 - BlueZ 5.75

**Possible fix**
Doubling BT HCI TX stack size (to 2048 from default 1024) seems to solve the problem. So a possible fix is to set below options for the board/SOC/sample (as I don't know root cause of the problem, I'm not entirely sure about the correct fix):
```
CONFIG_BT_HCI_TX_STACK_SIZE_WITH_PROMPT=y
CONFIG_BT_HCI_TX_STACK_SIZE=2048
```

erwango, 2024-05-27:
@asm5878 ^^

asm5878, 2024-05-28:
We have already seen this overflow in BT TX stack in other samples.
Indeed the proposed solution is fixing the issue also for all other samples 
We will investigate to understand if it's a real bug or a constraint coming from STM32WBA55 integration.

aescolar, 2024-05-28:
Closing as running out of stack for a given platform is not necessarily a bug we want to fix upstream as we may just then run out of RAM for another platform. Please just adjust the stack size for your platform.

closed 2024-05-28 by aescolar