zephyrproject-rtos/zephyr #73347: crypto: mbedtls: crypto.mbedtls fails

katgiadla, 2024-05-27:
**Describe the bug**
The test `crypto.mbedtls` fails.

Observed for:
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf9160dk/nrf9160/ns` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/crypto/mbedtls -p nrf9160dk/nrf9160/ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/.arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 40230 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain: Zephyr SDK 0.16.5-1
- Commit SHA or Version used: `v3.6.0-4561-g8c8e2f6243f1`

aescolar, 2024-05-28:
CC @valeriosetti @anangl

valeriosetti, 2024-05-28:
I was only able to test #73410 on nrf9160dk because I don't have nrf5340dk. I would really appreciate if reviewers can verify this for me.

PR #73410 (2024-05-31): test: mbedtls: do not enable USE_PSA and all PSA features
This PR fixes #73347.

Before #72243 Mbed TLS was not using
USE_PSA and all PSA features were not enabled. After #72243 if BUILD_WITH_TFM is set:
- USE_PSA in Mbed TLS is enabled by default and
- all PSA features are enabled.

This commits reverts both changes for tests/crypto/mbedtls test case.

**note** = I was only able to test #73410 on nrf9160dk because I don't have nrf5340dk. I would really appreciate if reviewers can verify this for me.
closed 2024-05-31 by henrikbrixandersen