zephyrproject-rtos/zephyr #73387: ESP32S3 Add --sysbuild compilation condition to smp_svr compilation errors

sam131208, 2024-05-28:
**Describe the bug**
zephyr\samples\subsys\mgmt\mcumgr\smp_svr add the compilation condition --sysbuild, the compilation cannot be successful.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr
2. git pull
3. west update
4. west build -b esp32s3_devkitc/esp32s3/procpu samples/subsys/mgmt/mcumgr/smp_svr --sysbuild -- -DEXTRA_CONF_FILE=overlay-bt.conf

**Expected behavior**
Compilation succeeded..

**Logs and console output**
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): sysbuild_default
-- Found Python3: C:/Python310/python.exe (found suitable version "3.10.11", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: D:/zephyr/zephyr/.cache
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32s3_devkitc, qualifiers: esp32s3/procpu
Parsing D:/zephyr/zephyr/share/sysbuild/Kconfig
Loaded configuration 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/sysbuild/application/empty.conf'
Merged configuration 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/sysbuild.conf'
Merged configuration 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/sysbuild/application/empty.conf'
Configuration saved to 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/sysbuild/application/zephyr/.config'
Kconfig header saved to 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/sysbuild/application/autoconf.h'
--
   *****************************
   * Running CMake for mcuboot *
   *****************************

Loading Zephyr default modules (Zephyr base).
-- Application: D:/zephyr/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.27.9
-- Found Python3: C:/Python310/python.exe (found suitable version "3.10.11", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: D:/zephyr/zephyr/.cache
-- Zephyr version: 3.6.99 (D:/zephyr/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32s3_devkitc, qualifiers: esp32s3/procpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (C:/Users/sam/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (C:/Users/sam/zephyr-sdk-0.16.5-1)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: D:/zephyr/zephyr/boards/espressif/esp32s3_devkitc/esp32s3_devkitc_procpu.dts
-- Found devicetree overlay: D:/zephyr/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/zephyr.dts
-- Generated devicetree_generated.h: D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/dts.cmake
Parsing D:/zephyr/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration 'D:/zephyr/zephyr/boards/espressif/esp32s3_devkitc/esp32s3_devkitc_procpu_defconfig'
Merged configuration 'D:/zephyr/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration 'D:/zephyr/bootloader/mcuboot/boot/zephyr/socs/esp32s3_procpu.conf'
Merged configuration 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/.config.sysbuild'
Configuration saved to 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/.config'
Kconfig header saved to 'D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/include/generated/autoconf.h'
-- Found GnuLd: c:/users/sam/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/sam/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc.exe
esptool path: D:/zephyr/modules/hal/espressif/tools/esptool_py/esptool.py
-- Configuring done (5.9s)
-- Generating done (0.2s)
-- Build files have been written to: D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot
--
   *****************************
   * Running CMake for smp_svr *
   *****************************

Loading Zephyr default modules (Zephyr base).
[... 63 lines cut ...]
     irom0_0_seg:      180790 B         8 MB      2.16%
     drom0_0_seg:      229676 B         8 MB      2.74%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
Generating files from D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/smp_svr/zephyr/zephyr.elf for board: esp32s3_devkitc

[2/233] Generating ../../../../include/generated/xtensa_handlers_tmp.c
cc1.exe: warning:  is shorter than expected
[7/233] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (D:/zephyr/zephyr), build: v3.6.0-4822-g0b220dd6a6d1
[228/233] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/zephyr_pre0.map
cmd.exe /C "cd . && C:\Users\sam\zephyr-sdk-0.16.5-1\xtensa-espressif_esp32s3_zephyr-elf\bin\xtensa-espressif_esp32s3_zephyr-elf-gcc.exe  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=D:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/timer/libdrivers__timer.a  modules/mcuboot/libmcuboot_util.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:/users/sam/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0"  -LD:/zephyr/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/mcuboot/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -lgcc  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -TD:/zephyr/modules/hal/espressif/zephyr/esp32s3/src/linker/esp32s3.rom.alias.ld  -TD:/zephyr/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.ld  -TD:/zephyr/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld  -TD:/zephyr/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld  -TD:/zephyr/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld  -TD:/zephyr/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld  -TD:/zephyr/modules/hal/espressif/zephyr/esp32s3/../../components/soc/esp32s3/ld/esp32s3.peripherals.ld  -Wl,--wrap=longjmp && cmd.exe /C "cd /D D:\zephyr\zephyr\samples\subsys\mgmt\mcumgr\smp_svr\build\mcuboot\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
c:/users/sam/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd.exe: zephyr\zephyr_pre0.elf section `.iram0.text' will not fit in region `iram_seg'
c:/users/sam/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd.exe: section .iram0.loader_text LMA [00000000403bec00,00000000403c046f] overlaps section .iram0.text LMA [00000000403b6000,00000000403bf383]
c:/users/sam/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd.exe: region `iram_seg' overflowed by 1924 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

**Environment (please complete the following information):**
 - OS:  Windows10
 - Toolchain： zephyr-sdk-0.16.5-1
 - zephyr last version

aescolar, 2024-05-28:
@sam131208 Seems there is just not enough flash

sylvioalves, 2024-05-28:
@sam131208, thanks for point this out. This is mcuboot build issue.  Until we fix/update this, you can manually change the IRAM length provided in [esp32s3/memory.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/espressif/esp32s3/memory.h#L50)
Increase it from 0x1000 to 0xA000

Build shall work properly. I have not tested besides loading the sample code.
```c
*** Booting Zephyr OS build v3.6.0-4794-ga70bf0e488c5 ***
[00:00:00.251,000] <inf> littlefs: LittleFS version 2.8, disk version 2.1
[00:00:00.251,000] <inf> littlefs: FS at flash-controller@60002000:0x250000 is 6 0x1000-byte blocks with 512 cycle
[00:00:00.251,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.252,000] <inf> esp32_bt_adapter: BT controller compile version [963cad4]
[00:00:00.284,000] <inf> bt_hci_core: Identity: 34:B4:72:70:01:3C (public)
[00:00:00.284,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.284,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
[00:00:00.286,000] <inf> smp_bt_sample: Advertising successfully started
[00:00:00.286,000] <inf> smp_sample: build time: May 28 2024 14:23:59
```

kpyush, 2024-06-01:
Hi I am facing the same issue with `esp32c3_devkitm`, it works if we increase the `BOOTLOADER_IRAM_SEG_LEN` from 0x8000 to 0xA000

midaczy, 2024-06-04:
Hi, the same problem on esp32s3_devkitm.

sam131208, 2024-06-08:
> @sam131208, thanks for point this out. This is mcuboot build issue. Until we fix/update this, you can manually change the IRAM length provided in [esp32s3/memory.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/espressif/esp32s3/memory.h#L50) Increase it from 0x9000 to 0xA000
> 
> Build shall work properly. I have not tested besides loading the sample code.
> 
> ```c
> *** Booting Zephyr OS build v3.6.0-4794-ga70bf0e488c5 ***
> [00:00:00.251,000] <inf> littlefs: LittleFS version 2.8, disk version 2.1
> [00:00:00.251,000] <inf> littlefs: FS at flash-controller@60002000:0x250000 is 6 0x1000-byte blocks with 512 cycle
> [00:00:00.251,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
> [00:00:00.252,000] <inf> esp32_bt_adapter: BT controller compile version [963cad4]
> [00:00:00.284,000] <inf> bt_hci_core: Identity: 34:B4:72:70:01:3C (public)
> [00:00:00.284,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
> [00:00:00.284,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
> [00:00:00.286,000] <inf> smp_bt_sample: Advertising successfully started
> [00:00:00.286,000] <inf> smp_sample: build time: May 28 2024 14:23:59
> ```

 @sylvioalves    This issue has been fixed after the modification. However, the program cannot boot to Zephyr OS.
 The development board model is esp32s3_devkitc（N32R8）.
Logs and console output

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:2
load:0x3fcd1800,len:0x299c
load:0x403b4c00,len:0x9798
load:0x403bec00,len:0x1870
entry 0x403b77f8
I (88) boot: MCUboot 2nd stage bootloader
I (88) boot: compile time Jun  7 2024 20:47:33
W (88) boot: Unicore bootloader
W (88) spi_flash: Octal flash chip is using but dio mode is selected, will automatically swich to Octal mode
I (97) spi_flash: detected chip: mxic (opi)
I (101) spi_flash: flash io: opi_str
W (104) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (116) boot: chip revision: v0.1
I (119) boot.esp32s3: Boot SPI Speed : 40MHz
I (123) boot.esp32s3: SPI Mode       : SLOW READ
I (127) boot.esp32s3: SPI Flash Size : 8MB
I (131) boot: Enabling RNG early entropy source...
W (170) spi_flash: Octal flash chip is using but dio mode is selected, will automatically swich to Octal mode
I (170) spi_flash: flash io: opi_str
[esp32s3] [INF] Image index: 0, Swap type: none

sylvioalves, 2024-07-10:
Issue already fixed.

closed 2024-07-10 by sylvioalves