zephyrproject-rtos/zephyr #73398: STM32 SD/MMC 4-bit bus not working

attie-argentum, 2024-05-28:
**Describe the bug**
I/O fails when using an STM32L476, with a Micro SD card, configured for 4-bit bus mode.

Consider the following `nucleo_l476rg.overlay` paired with `BOARD=nucleo_l476rg` and `CONFIG_FILE_SYSTEM_SHELL=y`.
```dts
/delete-node/ &spi3;
&sdmmc1 {
    bus-width = <1>;
    clk-div = <0>;
    pinctrl-0 = <
        &sdmmc1_ck_pc12
        &sdmmc1_cmd_pd2
        &sdmmc1_d0_pc8
        &sdmmc1_d1_pc9
        &sdmmc1_d2_pc10
        &sdmmc1_d3_pc11
    >;
    pinctrl-names = "default";
    status = "okay";
};

&clk_msi {
    msi-range = <11>; /* ~48 MHz */
    status = "okay";
};
```

This works fine:
```
uart:~$ fs mount fat /SD:
Successfully mounted fat fs:/SD:
uart:~$ fs erase_write_test /SD:/test.bin 0x100000 1
Loop #1 done in 498ms.
Total: 498ms, Per loop: ~498ms, Speed: ~2.0MiBps
uart:~$ fs read_test /SD:/test.bin 1
File size: 1.0MiB
Loop #1 done in 384ms.
Total: 384ms, Per loop: ~384ms, Speed: ~2.6MiBps
```

While changing to `bus-width = <4>;` results in an error:
```
uart:~$ fs mount fat /SD:
Successfully mounted fat fs:/SD:
uart:~$ fs erase_write_test /SD:/test.bin 0x100000 1
Failed to truncate /SD:/test.bin (-5)
[00:00:04.498,000] <err> stm32_sdmmc: sd write error 16
[00:00:04.498,000] <err> fs: file truncate error (-5)
[00:00:04.502,000] <err> fs: file close error (-9)
```

I originally tracked this down to an issue in the STM32 HAL, and the team have produced [a patch](https://github.com/STMicroelectronics/stm32l4xx_hal_driver/issues/17) that they believe works (I'm unable to verify directly at this moment).

**To Reproduce**
Steps to reproduce the behavior:
1. Use `nucleo_l476rg.overlay` (below)
2. `cmake -B ./build-sd -GNinja ./zephyr/samples/subsys/shell/shell_module -D BOARD=nucleo_l476rg`
3. `ninja -C build-sd/`
4. `st-flash --connect-under-reset write ./build-sd/zephyr/zephyr.bin 0x08000000`
5. Execute `fs mount fat /SD:` in the device's shell
6. Execute `fs erase_write_test /SD:/test.bin 0x100000 1`, observe error

`./zephyr/samples/subsys/shell/shell_module/boards/nucleo_l476rg.overlay`:
```dts
/delete-node/ &spi3;
&sdmmc1 {
    bus-width = <4>;
    clk-div = <0>;
    pinctrl-0 = <
        &sdmmc1_ck_pc12
        &sdmmc1_cmd_pd2
        &sdmmc1_d0_pc8
        &sdmmc1_d1_pc9
        &sdmmc1_d2_pc10
        &sdmmc1_d3_pc11
    >;
    pinctrl-names = "default";
    status = "okay";
};

&clk_msi {
    msi-range = <11>; /* ~48 MHz */
    status = "okay";
};
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: GCC 13.2.rel1
- Zephyr @ f51c8ee7390
- STM32 HAL @ 22925907 (with 4-bit patch applied manually)

erwango, 2024-05-28:
Thanks @attie-argentum. Then, what we need is to add the proposed patch to hal_stm32.
Would you have the possibility to do so ?

attie-argentum, 2024-05-28:
Yes, sure - I can prepare a PR for `hal_stm32` (though to clarify, I don't expect that to "_resolve_" this issue, as above)

erwango, 2024-09-18:
Fixed merged as part of https://github.com/zephyrproject-rtos/zephyr/pull/76868

closed 2024-09-18 by erwango