zephyrproject-rtos/zephyr #73411: LE Audio: Tester: Broadcast too eager to reconfigure

Thalley, 2024-05-28:
**Describe the bug**
In tests like `BAP/BSRC/SCC/BV-34-C` PTS requests us to reconfigure the broadcast source. This is currently done by calling 
1) btp_bap_broadcast_adv_stop
2) btp_bap_broadcast_source_stop
3) btp_bap_broadcast_source_setup

The issue is that bttester assume that the broadcast source goes into the `BT_BAP_EP_STATE_QOS_CONFIGURED` state immediately after calling `btp_bap_broadcast_source_stop`, and thus it calls `btp_bap_broadcast_source_setup` as soon as possible. In reality, the broadcast source does not enter the `BT_BAP_EP_STATE_QOS_CONFIGURED` state until the `HCI_LE_Terminate_BIG_Complete` event has been received by the controller. 

This makes this, and other similar tests, subject to timing and may fail on other controllers than the Zephyr one. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run the `BAP/BSRC/SCC/BV-34-C` test with AutoPTS with the Soft Device controller

**Expected behavior**
When `btp_bap_broadcast_source_stop` is called, it should wait for the `HCI_LE_Terminate_BIG_Complete` event, which triggers the ISO disconnected function which triggers the `btp_bap_broadcast:stream_stopped` function. We should not attempt to call `btp_bap_broadcast_source_setup` before that has been called. 

This can be solved either in the bttester itself, or in the python scripts that instruct the bttester (by adding a disconnected event similar to `btp_send_bis_syced_ev`). 

**Impact**
Prevents several tests from passing on non-Zephyr controllers. 

**Logs and console output**
See attached files
[BAP_BSRC_SCC_BV-34-C_2024_05_27_16_14_30.log](https://github.com/zephyrproject-rtos/zephyr/files/15470720/BAP_BSRC_SCC_BV-34-C_2024_05_27_16_14_30.log)
[autopts-iutctl-zephyr.log](https://github.com/zephyrproject-rtos/zephyr/files/15470721/autopts-iutctl-zephyr.log)
[BAP_BSRC_SCC_BV-34-C_iutctl.log](https://github.com/zephyrproject-rtos/zephyr/files/15470722/BAP_BSRC_SCC_BV-34-C_iutctl.log)

**Environment (please complete the following information):**
 - Commit SHA or Version used: 64de50f05235e5e681810510edbf8694e2b6a482

**Additional context**
N/A

Frodevan, 2024-05-28:
It can be mentioned, for the proposed steps to reproduce, #73361 needs to be applied first.

nashif, 2024-06-05:
@Thalley  any update on this one? Any fix coming?

Thalley, 2024-06-05:
It's assigned to @sjanc, so let's see if they have any updates on this (I'm not aware of any)

carlescufi, 2024-07-11:
Downgrading to Low because there are other qualification issues not tracked on GitHub right now.

closed 2024-10-07 by carlescufi