zephyrproject-rtos/zephyr #73419: doc: Documentation issue in 'kernel/services/data_passing/pipes'

Xcross74, 2024-05-28:
**No mention of CONFIG_PIPE=y in documentation**

<< This page doesn't mention that the prj.conf should enable CONFIG_PIPE=y to allow pipes to be used.  >>

**Environment**

* Page: `kernel/services/data_passing/pipes`
* Version: main
* SHA-1: 3fab7624eb4b31dd6caa33ad50adb18b51cc9a60

andyross, 2024-05-29:
Calling not-it and bouncing to @peter-mitsis as he's touched pipes most recently :)

peter-mitsis, 2024-05-29:
Although the page in question does list the CONFIG_PIPES Kconfig option, it does not provide a link to its description. I expect that fixing the documentation to generate the link ought to be sufficient as it will then bring the pipes page into alignment with other kernel documentation pages whose features need a Kconfig option to be enabled to use the feature (e.g. polling, events).

PR #73481 (2024-06-04): doc: kernel: Tweak pipes documentation
Updates the pipes documentation to provide a link to the CONFIG_PIPES Kconfig option instead of merely just listing it. This brings the pipes documentation into alignment with other kernel service documentation (such as polling and events) whose features require a Kconfig option to be enabled before the feature can be used.

Fixes #73419
closed 2024-06-04 by nashif