zephyrproject-rtos/zephyr #73443: apollo4p_blue_kxr_evb: Multiple samples/tests fails to build in CI due to warnings

aescolar, 2024-05-29:
**Describe the bug**

* samples/subsys/llext/shell_loader/
* tests/kernel/cache/*
* tests/subsys/llext/simple/*
* samples/subsys/llext/shell_loader/

Build with warnings due to a missing CMSIS header , which are promoted to errors in CI resulting in CI failures for this platform (currently 12 samples/tests).

**To Reproduce**
Steps to reproduce the behavior:
1. twister -T samples/subsys/llext/shell_loader/  -p apollo4p_blue_kxr_evb
4. See error

**Expected behavior**
No build warnings/errors

**Impact**
CI failing when this target x sample combination is triggerd.

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/9240178517/job/25420194674#step:13:4995

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): 0.16.5
 - Commit SHA or Version used: ea29ef36bbedeacd171199d6feea5e699fa64447

aescolar, 2024-05-29:
CC @aaronyegx @AlessandroLuo @RichardSWheatley

aaronyegx, 2024-05-29:
@RichardSWheatley I see you are looking at the issue (maybe another similar ticket) and have some fix in the HAL?

RichardSWheatley, 2024-05-29:
I have an AmbiqHal branch updated to include the folders. Let's speak at Ambiq to determine best path.

nashif, 2024-06-04:
> I have an AmbiqHal branch updated to include the folders. Let's speak at Ambiq to determine best path.

any updates?

pillo79, 2024-06-07:
Previous ticket for the same issue: https://github.com/zephyrproject-rtos/zephyr/issues/72775

RichardSWheatley, 2024-06-14:
> Previous ticket for the same issue: https://github.com/zephyrproject-rtos/zephyr/issues/72775

Just a question for reference, how come this failure wasn't noticed until months after the hal addition that causes this failure?

aescolar, 2024-06-14:
> Just a question for reference, how come this failure wasn't noticed until months after the hal addition that causes this failure?

The CI run in a PR is a small subset of tests. What tests on what targets is run is based on what files are changed. Sometimes the subset does not cover enough and some issue gets in.
Other PRs which later affect a test may trigger the test being built/run on the affected targets revealing the issue.
The runs in main merge jobs are bigger, but still, do not cover everything. Though when something is detected there it tends to be reported quite soon though.
We also have a weekly job running on a lot of tests x targets combinations, but due to the amount of tests failures are too normal, so only when a pattern is clear it is seen and reported. That is what led to this issue being opened.

RichardSWheatley, 2024-06-14:
> The CI run in a PR is a small subset of tests. What tests on what targets is run is based on what files are changed. Sometimes the subset does not cover enough and some issue gets in.
>
> Other PRs [...]

Did you see the PR my colleague made for this?

PR #74232

If you approve, I will approve.

aescolar, 2024-06-14:
https://github.com/zephyrproject-rtos/zephyr/pull/74275 ?
That one is not building yet

RichardSWheatley, 2024-06-14:
> https://github.com/zephyrproject-rtos/zephyr/pull/74275 ?
> 
> That one is not building yet

They mad to add HAL changes to support the kconfig changes.

I am asking about whether that will work for you.

aescolar, 2024-06-14:
> I am asking about whether that will work for you.

If the tests pass again, sure. I only disabled it to unblock CI for others.

PR #74275 (2024-06-25): boards: ambiq: apollo4p_evb: fix multiple CI failure
This PR is trying to fix multiple CI failures which were introduced from https://github.com/zephyrproject-rtos/zephyr/pull/70842.
Adds the missing flash controller node in apollo4p_evb to address the CI failure https://github.com/zephyrproject-rtos/zephyr/issues/74212. 
Remove the untestable cache kconfig from apollo4x boards to address the CI failure https://github.com/zephyrproject-rtos/zephyr/issues/72775, https://github.com/zephyrproject-rtos/zephyr/issues/73443 at this moment.
And enable the twister for apollo4x platforms. Refer to http [... cut]
closed 2024-06-25 by nashif