zephyrproject-rtos/zephyr #73468: adc_cc13xx_cc26xx fails to build for beagleconnect freedom

Ayush1325, 2024-05-29:
**Describe the bug**
Ti adc_cc13xx_cc26xx driver fails to build for beagleconnect freedom with cc1352p7. I am trying to enable adc support in [Arduino Core API module for Zephyr](https://github.com/zephyrproject-rtos/gsoc-2022-arduino-core).

Here is my device tree overlay:
```dts
/ {
 zephyr,user {
  	io-channels = <&adc0 23>, <&adc0 24>, <&adc0 25>,
				 <&adc0 26>, <&adc0 27>, <&adc0 28>;
 };
};

&adc0 {
	status = "okay";
};
```

The error I am getting is as follows:
```shell
[187/192] Linking CXX executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/ayush/Documents/Programming/BeagleBoard/zephyr-arduino/build/zephyr/zephyr_pre0.map
: && ccache /home/ayush/.local/opt/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/ayush/Documents/Programming/BeagleBoard/zephyr-arduino/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/boards/boards/arm/beagleconnect_freedom/libboards__beagle__beagleconnect_freedom.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/pwm/libdrivers__pwm.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/ti/simplelink/source/ti/devices/cc13x2x7_cc26x2x7/lib..__modules__hal__ti__simplelink__source__ti__devices__cc13x2x7_cc26x2x7.a  modules/ti/simplelink/lib..__modules__hal__ti__simplelink.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/var/home/ayush/dotfiles/.local/opt/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -L/home/ayush/Documents/Programming/BeagleBoard/zephyr-arduino/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_DOUBLE_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /home/ayush/Documents/Programming/BeagleBoard/zephyr-arduino/build/zephyr && /usr/bin/cmake -E true
/var/home/ayush/dotfiles/.local/opt/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/adc/libdrivers__adc.a(adc_cc13xx_cc26xx.c.obj): in function `adc_context_start_sampling':
/home/ayush/Documents/Programming/BeagleBoard/zephyr-arduino/zephyr/drivers/adc/adc_cc13xx_cc26xx.c:87: undefined reference to `NOROM_AUXADCEnableSync'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build 
```

For some reason, `ROM_AUXADCEnableSync` is not defined in ti hal and thus, `AUXADCEnableSync` is set to `NOROM_AUXADCEnableSync`. Not sure if this is expected or not.

**Impact**
Cannot use `ti,cc13xx-cc26xx-adc` on beagleconnect freedom

**Environment (please complete the following information):**
 - OS: Fedora Sway Atomic 40
 - Toolchain: Zephyr SDK 0.16.5
 - Commit SHA or Version used: 34c84eccec0508b16f5001b20fa369f387d856df

aescolar, 2024-06-04:
CC @jadonk @ayush1325 @con-pax @vaishnavachath

Ayush1325, 2024-06-04:
I have found the problem: https://github.com/zephyrproject-rtos/hal_ti/pull/54

closed 2024-07-29 by aescolar