zephyrproject-rtos/zephyr #73485: Can't boot same70 based soc

yperess, 2024-05-29:
**Describe the bug**
Taking a board such as the TDK Robokit1, I've built and flashed the hello_world sample. The sample fails during `z_arm_platform_init()`. In testing I found that commenting out https://github.com/zephyrproject-rtos/zephyr/blame/main/soc/atmel/sam/same70/soc.c#L125 fixes the issue. When I run the app via `west debug` I get this backtrace:

```
(gdb) bt
#0  arch_system_halt (reason=26) at /home/peress/demo/third_party/zephyr/kernel/fatal.c:29
#1  0x0040298e in k_sys_fatal_error_handler (reason=<optimized out>, esf=<optimized out>) at /home/peress/demo/third_party/zephyr/kernel/fatal.c:44
#2  0x004012d8 in z_fatal_error (reason=<optimized out>, esf=<optimized out>) at /home/peress/demo/third_party/zephyr/kernel/fatal.c:118
#3  0x00402446 in z_arm_fatal_error (reason=<optimized out>, esf=esf@entry=0x20401010 <z_main_stack+912>) at /home/peress/demo/third_party/zephyr/arch/arm/core/fatal.c:86
#4  0x00400674 in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /home/peress/demo/third_party/zephyr/arch/arm/core/cortex_m/fault.c:1157
#5  0x00400734 in z_arm_usage_fault () at /home/peress/demo/third_party/zephyr/arch/arm/core/cortex_m/fault_s.S:102
#6  <signal handler called>
#7  SCB_DisableDCache () at /home/peress/demo/modules/hal/cmsis/CMSIS/Core/Include/cachel1_armv7.h:199
#8  z_arm_platform_init () at /home/peress/demo/third_party/zephyr/soc/atmel/sam/same70/soc.c:125
#9  0x00400746 in z_arm_reset () at /home/peress/demo/third_party/zephyr/arch/arm/core/cortex_m/reset.S:97
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b robokit1 samples/hello_world
2. west debug
3. See error

**Expected behavior**
App should print 
```
** Booting Zephyr OS build v3.6.0-4983-g3c072f917cf8 ***
Hello World! robokit1/same70q21b
```

**Impact**
Braking hardware testing

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.6
 - Commit SHA ToT #19f645edd40b38e54f505135beced1919fdc7715

tswaehn, 2024-06-01:
what I did is comment the line `SCB_DisableDCache()` ins` sam/same70/soc.c`

to be honest I didnt figure out yet, why and what it does.

strangely after booting once, it is possible to comment int the line. but cold-reboot is stuck again, ...

benediktibk, 2024-06-01:
Maybe it is already necessary at this point to invalidate the cache with SCB_InvalidateDCache (https://www.keil.com/pack/doc/CMSIS_Dev/Core/html/group__Dcache__functions__m7.html#ga6468170f90d270caab8116e7a4f0b5fe)?
Could you please give it a try, because unfortunately I do not have this hardware available.

tswaehn, 2024-06-01:
> Could you please give it a try, because unfortunately I do not have this hardware available.

just tested. works like charm.

`sam/same70/soc.c` looks like this now:

```
	/*
	 * DTCM is enabled by default at reset, therefore we have to disable
	 * it first to get the caches into a state where then the
	 * sys_cache*-functions can enable them, if requested by the
	 * configuration.
	 */
        SCB_InvalidateDCache();
	SCB_DisableDCache();
```

benediktibk, 2024-06-03:
@tswaehn Thanks for testing and providing the PR!

closed 2024-06-05 by nashif