zephyrproject-rtos/zephyr #73529: LVGL: rotary encoder push button indev

lorc-dev, 2024-05-30:
**Describe the bug**
When the rotary encoder pushbutton is pressed, the input event is not properly passed to LVGL.
This makes it impossible to go into the edit mode of a widget.

When LVLG processes the encoder input, it expects that the key is set to `LV_KEY_ENTER`.
Before LVGL calls the `read_cb` function, it will [set](https://github.com/zephyrproject-rtos/lvgl/blob/a76caad0aa99838bb7a3315aaaf07e50d595c284/src/hal/lv_hal_indev.c#L181) the `key` member of `lv_indev_data_t` to `LV_KEY_ENTER`.

But in the read_cb [read_cb](https://github.com/zephyrproject-rtos/zephyr/blob/3ea5f799dd7978b7baf2edd7d2458e0c36e431fc/modules/lvgl/input/lvgl_common_input.c#L41) function, the passed `lv_indev_data_t` struct is overwritten by the one that it got from the message queue. 
This struct has the `key`member set to 0, which is not a predefined LVGL key.

A second problem is that when the pushbutton is pressed, LVGL generates a `LV_KEY_LEFT/RIGHT` event.
This seems to be caused by that the `enc_dif` field is not cleared when there is a button event ([lvgl_encoder_input.c#L30](https://github.com/zephyrproject-rtos/zephyr/blob/3ea5f799dd7978b7baf2edd7d2458e0c36e431fc/modules/lvgl/input/lvgl_encoder_input.c#L30)).

**Expected behavior**
When the encoder button is pressed, it should send a `LV_KEY_ENTER` instead of `LV_KEY_LEFT/RIGHT` event

**Environment:**

-  Zephyr sdk 0.16.5
-  Zephyr v3.6.0
-  LVGL: 2b76c641749725ac90c6ac7959ca7718804cf356

faxe1008, 2024-05-30:
I have checked the LVGL documentation ([https://docs.lvgl.io/8.4/porting/indev.html#encoder](https://docs.lvgl.io/8.4/porting/indev.html#encoder)) and also the documentation on the `lv_indev_data_t` type: [https://github.com/lvgl/lvgl/blob/release/v8.4/src/hal/lv_hal_indev.h#L75](https://github.com/lvgl/lvgl/blob/release/v8.4/src/hal/lv_hal_indev.h#L75) and the `key` field is intended to be used with the keypad device. Yes there is a paragraph about using keys to emulate the behaviour of the encoder. You can even see this in the indev `indev_encoder_proc` method: 
[https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L1016](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L1016). Further up in the method the `key` is used to manipulate the `enc_diff` field. And on the line that I linked you you can see that if the enc_diff is set the corresponding key is forwarded to the group. The only issue that you might experience is the fact that you did not put the group into editing mode. When using an encoder you have to longpress the button to enter editing mode or leave it.

The porting of LVGL its self also only makes use of those fields [https://github.com/lvgl/lvgl/blob/master/examples/porting/lv_port_indev_template.c#L329](https://github.com/lvgl/lvgl/blob/master/examples/porting/lv_port_indev_template.c#L329)

> But in the read_cb [read_cb](https://github.com/zephyrproject-rtos/zephyr/blob/3ea5f799dd7978b7baf2edd7d2458e0c36e431fc/modules/lvgl/input/lvgl_common_input.c#L41) function, the passed [...]

This is the intended behaviour, LVGL fetches the data this way. And 0 is essentially used as "none" value, since there is no other way of expressing that no key was pressed/released since the last call of the input device handler timer:
[https://github.com/lvgl/lvgl/blob/master/examples/porting/lv_port_indev_template.c#L279](https://github.com/lvgl/lvgl/blob/master/examples/porting/lv_port_indev_template.c#L279)

> A second problem is that when the pushbutton is pressed, LVGL generates a LV_KEY_LEFT/RIGHT event.

Can you maybe show me where this is happening? I have a hard time locating this.
Also can you maybe provide your DTS, maybe you are reusing input event codes accidentally?

lorc-dev, 2024-05-30:
> The only issue that you might experience is the fact that you did not put the group into editing mode. 
> When using an encoder you have to longpress the button to enter editing mode or leave it.

Even when you longpress the button, it won't go into editing mode. It will set the group in/out of editing mode [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L921) and [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L1006), but both only when `data->key == LV_KEY_ENTER`.

Of course, it is possible to enter editing mode manually in the application itself, but the [indev_encoder_proc](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L831) is imo clearly intended to do it automatically.
The [LVGL documentation](https://lvgl.io/docs/latest/en/html/porting/indev.html#encoder) says that pressing the encoder button should put the object into edit mode, which it currently does not.
 

> The porting of LVGL its self also only makes use of those fields https://github.com/lvgl/lvgl/blob/master/examples/porting/lv_port_indev_template.c#L329

It sets indeed only the enc_diff and state field, but it also doesn't clear the key field which is already set to `LV_KEY_ENTER`.
When the input device is an encoder, LVGL will explicitly set the `key` field right before calling the read_cb function (see: [lv_hal_indev.c#L180-L182](https://github.com/lvgl/lvgl/blob/release/v8.4/src/hal/lv_hal_indev.c#L180-L182)).

> > A second problem is that when the pushbutton is pressed, LVGL generates a LV_KEY_LEFT/RIGHT event.
>
> Can you maybe show me where this is happening? I have a hard time locating this. Also can you maybe provide your DTS, maybe you are reusing input event codes accidentally?

Yeah, I forgot to mention that this only happens when the encoder is rotated first.
In the [lvgl_encoder_process_event()](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/lvgl/input/lvgl_encoder_input.c#L22) function the same struct is reused for each event from the same input device (saved in `dev->data`).
Thus, when first a rotation event occurs, the enc_diff field will for example be set to `-1`. When subsequently a button event occurs, the `enc_diff` field will still be set to `-1`. 
This causes LVGL to think that a encoder has rotated again.

Given how LVGL handles the encoder input, it makes imo more sense to set the default value of the `key` field to `LV_KEY_ENTER` than `0`.
Clearing the `enc_diff` field and setting the `key` field to `LV_KEY_ENTER` in [lvgl_encoder_process_event()](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/lvgl/input/lvgl_encoder_input.c#L22) would solve both problems.
``` c
struct lvgl_common_input_data *data = dev->data;
const struct lvgl_encoder_input_config *cfg = dev->config;

if (evt->code == cfg->rotation_input_code) {
    data->pending_event.enc_diff = evt->value;
} else if (evt->code == cfg->button_input_code) {
    data->pending_event.state = evt->value ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;
    data->pending_event.enc_diff = 0;
    data->pending_event.key = LV_KEY_ENTER;
} else {
  ...
```

faxe1008, 2024-05-31:
> Even when you longpress the button, it won't go into editing mode. It will set the group in/out of editing mode [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L921) and [...]

Ok now I understand what you meant, you are correct. This was the missing piece for my understanding:

> When the input device is an encoder, LVGL will explicitly set the key field right before calling the read_cb function

> Yeah, I forgot to mention that this only happens when the encoder is rotated first.
> In the [...]

That is also correct.
Your proposed change seems good to me, do you want to raise a PR for this :^) ?