zephyrproject-rtos/zephyr #73548: Ambiq Apollo3 MSPI uses Apollo4 specific functions and defines

Lukas1811, 2024-05-31:
I'm currently trying to get a Sparkfun Artemis Nano with an Apollo3 to talk to an SPI device.
I setup the environment today using west, so it should be on the current Zephyr release.

My device tree entry looks like this:
```devicetree
&mspi0 {
    status = "okay";
    cs-gpios = <&gpio32_63 8 GPIO_ACTIVE_HIGH>;
    pinctrl-0 = <&mspi0_default>;
    pinctrl-names = "default";
    clock-frequency = <6000000>;
    as2933: as2933@0 {
            compatible = "vnd,spi-device";
            reg = <0>;
            spi-max-frequency = <6000000>;
            label = "as2933";
    };
};
```

And the code like this:
```C
#define SPIOP      SPI_WORD_SET(8) | SPI_TRANSFER_MSB
struct spi_dt_spec spispec = SPI_DT_SPEC_GET(DT_NODELABEL(as2933), SPIOP, 0);
```

Compiling it with `west build -b apollo3_evb  -- -DDTC_OVERLAY_FILE=./boards/apollo3_evb.overlay` leads to the following results:
```bash
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:46:14: error: 'AM_HAL_MSPI_CLK_96MHZ' undeclared (first use in this function); did you mean 'AM_HAL_MSPI_CLK_6MHZ'?
   46 |         case AM_HAL_MSPI_CLK_96MHZ:
      |              ^~~~~~~~~~~~~~~~~~~~~
      |              AM_HAL_MSPI_CLK_6MHZ
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:46:14: note: each undeclared identifier is reported only once for each function it appears in
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:48:14: error: 'AM_HAL_MSPI_CLK_32MHZ' undeclared (first use in this function); did you mean 'AM_HAL_MSPI_CLK_3MHZ'?
   48 |         case AM_HAL_MSPI_CLK_32MHZ:
      |              ^~~~~~~~~~~~~~~~~~~~~
      |              AM_HAL_MSPI_CLK_3MHZ
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:59:21: error: 'AM_HAL_MSPI_CLK_INVALID' undeclared (first use in this function); did you mean 'AM_HAL_MSPI_CLK_1P5MHZ'?
   59 |                 d = AM_HAL_MSPI_CLK_INVALID;
      |                     ^~~~~~~~~~~~~~~~~~~~~~~
      |                     AM_HAL_MSPI_CLK_1P5MHZ
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c: In function 'mspi_config':
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:110:35: error: 'AM_HAL_MSPI_CLK_INVALID' undeclared (first use in this function); did you mean 'AM_HAL_MSPI_CLK_1P5MHZ'?
  110 |         if (mspicfg.eClockFreq == AM_HAL_MSPI_CLK_INVALID) {
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~
      |                                   AM_HAL_MSPI_CLK_1P5MHZ
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c: In function 'mspi_ambiq_init':
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:209:9: error: unknown type name 'am_hal_mspi_config_t'; did you mean 'am_hal_tpiu_config_t'?
  209 |         am_hal_mspi_config_t mspiCfg = {0};
      |         ^~~~~~~~~~~~~~~~~~~~
      |         am_hal_tpiu_config_t
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:211:16: error: request for member 'pTCB' in something not a structure or union
  211 |         mspiCfg.pTCB = NULL;
      |                ^
/home/lukas/zephyrproject/zephyr/drivers/spi/mspi_ambiq.c:221:15: warning: implicit declaration of function 'am_hal_mspi_configure'; did you mean 'am_hal_tpiu_configure'? [-Wimplicit-function-declaration]
  221 |         ret = am_hal_mspi_configure(data->mspiHandle, &mspiCfg);
      |               ^~~~~~~~~~~~~~~~~~~~~
      |               am_hal_tpiu_configure
```

I've looked into the [Ambiq HAL](https://github.com/zephyrproject-rtos/hal_ambiq/tree/main) and it seems like the missing struct, function and defines are all defined for the Apollo4 but not for the Apollo3.

For me it does not seem like a user error (it may is, always possible that I've overlooked something).

henrikbrixandersen, 2024-06-03:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

aescolar, 2024-06-04:
CC @aaronyegx @AlessandroLuo @RichardSWheatley

Lukas1811, 2024-06-05:
Hi @AlessandroLuo thank you for your feedback.
I use a simple single line SPI device (it's a wakeup receiver for RF applications). I'm somewhat new to device trees and still trying to get the hang of hit, do you have an example for a device tree entry to add a SPI device? For me it seems like there is no entry for the normal SPI lines in the apollo3_evb.dts files, that's why I've used the mspi one.

AlessandroLuo, 2024-06-05:
@Lukas1811 don't use MSPI, there's no MSPI pinout on your board, if you are at the latest main branch, there's a spi0 defined in apollo3_evb.dts, you can start from there.

Lukas1811, 2024-06-05:
Ok you commited it yesterday, that explains why I didn't found it, my last pull was 3 days ago. I will give it a try when I'm back home, thank you for your support.

AlessandroLuo, 2024-06-28:
issue solved in https://github.com/zephyrproject-rtos/zephyr/pull/71769, close this ticket

closed 2024-06-28 by AlessandroLuo