zephyrproject-rtos/zephyr #73555: Failure to Establish L4 Connection with SIM7000E on FRDM-K64F (cellular_modem sample)

nmpluta, 2024-05-31:
**Describe the bug:**
The application builds and flashes properly, and modem initialization completes successfully. However, the connection to `L4` always fails due to a timeout.

- **Target Platform:**
  - Board: NXP FRDM-K64F
  - LTE Module: Waveshare SIM7000E NB-IoT HAT

- **Diagnosis attempts:** Increased log verbosity

**Module connections:**
Connected the Waveshare SIM7000E NB-IoT HAT to the NXP FRDM-K64F as follows:
- FRDM-K64F Pin 3.3V/5V → SIM7000E VCC
- FRDM-K64F GND → SIM7000E GND
- FRDM-K64F UART TX (PTC3) → SIM7000E RX
- FRDM-K64F UART RX (PTC4) → SIM7000E TX
[image]

**Configuration Files:**
Added config files for target in sample directory:
- **`frdm_k64f.overlay`:**
```dts
/*
 * Copyright (c) 2022-2024 Natalia Pluta
 *
 * SPDX-License-Identifier: Apache-2.0
 */

 / {
    aliases {
        modem-uart = &arduino_serial;
        modem = &modem;
    };
};

&arduino_serial {
    status = "okay";
    current-speed = <115200>;

    modem: modem {
        compatible = "simcom,sim7080";
        mdm-power-gpios = <&gpiob 2 GPIO_ACTIVE_HIGH>;
        status = "okay";
    };
};

&enet{
    status = "disabled";
};
```
- **`frdm_k64f.conf`:**
```conf
CONFIG_MODEM_CELLULAR_APN="internet"
CONFIG_UART_ASYNC_API=y
```

**To Reproduce:**
1. Copy provided `.conf` and `.overlay` files to the `cellular_modem` sample directory.
2. Run `west build -b frdm_k64f zephyr/samples/net/cellular_modem -p`.
3. Connect the LTE module to the board as described.
4. Run `west flash`.
5. Open the terminal and observe the output.

**Expected behavior:**
The modem should establish a connection to `L4` without timing out.

**Impact:**
This issue is a showstopper for using the cellular modem on the NXP FRDM-K64F platform.

**Logs and console output:**
```
[... 103 lines cut ...]
[00:00:14.565,000] <dbg> modem_cmux: modem_cmux_log_frame: data:
```

2. Sanity of LTE module 
The sanity of LTE module itself as well as the ability to connect to the network was tested and seems to work fine.
```
> AT+CSTT="internet"
AT+CSTT="internet"
OK
> AT+CIICR
AT+CIICR
OK
> AT+CIFSR
AT+CIFSR
10.72.205.48
> AT+CIPPING="www.google.com"
AT+CIPPING="www.google.com"
+CIPPING: 2,"142.250.203.196",160,114
+CIPPING: 3,"142.250.203.196",156,114
+CIPPING: 4,"142.250.203.196",161,114

OK
```

Please advise on potential solutions or further debugging steps.

bjarki-trackunit, 2024-05-31:
The modem is not accepting CMUX frames after enabling CMUX, this is usually due to a timing error, from the modem receiving `AT+CMUX`.
Things to check:
* Does the modem disable and reenable the UART during switch over from AT to CMUX mode? (the TX pin will go low for a period if this is the case). Make sure to close the UART pipe before it is disabled, and opening it after, by adjusting https://github.com/zephyrproject-rtos/zephyr/blob/4b6d3f1e16ea8ccccbf85f1eaf7efd0caea11766/drivers/modem/modem_cellular.c#L1720 and https://github.com/zephyrproject-rtos/zephyr/blob/4b6d3f1e16ea8ccccbf85f1eaf7efd0caea11766/drivers/modem/modem_cellular.c#L750
* The modem may deadlock if the CMUX connect frame (SABM on channel 0) is sent before the modem is ready for it, increase the delay after sending AT+CMUX, maybe the modem will respond as expected
* The modem may have timed out, if the time after AT+CMUX to receiving the connect CMUX frame, decrese the delay after sending AT+CMUX, optimally wait for "OK", by changing the AT+CMUX chat script command to https://github.com/zephyrproject-rtos/zephyr/blob/4b6d3f1e16ea8ccccbf85f1eaf7efd0caea11766/drivers/modem/modem_cellular.c#L1765

nmpluta, 2024-06-03:
@bjarki-trackunit thank you very much for the quick feedback.

I have tried all the suggested changes and tests in the code, but unfortunately, they didn't help resolve the issue. 

After debugging with a logic analyzer, I found that the modem responds to the AT command `AT+CMUX=0,0,5,127` with `OK`, but afterward, it does not respond to the CMUX frames at all.
[image]
[
[image]

Although there is no official information from the manufacturer stating that CMUX is not supported on the SIMCOM 7000E, a short research led me to discover that the same problem was encountered in the Espressif project. As mentioned in the following GitHub issue: https://github.com/espressif/esp-protocols/issues/187

Given this, is it possible to use the new modem subsystem (and this sample) with modems that do not support the CMUX feature? If so, could you please point out what modifications might be required?

Thank you for your assistance.

bjarki-trackunit, 2024-06-03:
It is indeed possible, but you will have to modify the cellular modem driver a bit, essentially:

* remove the states dealing with setting up cmux, and just use the bus_pipe (UART) directly
* remove the periodic script
* change the state machine to react to L2 PPP loosing its carrier to switch from PPP back to waiting for the modem to register to a network.

You will loose the ability to check things like RSRQ while the modem is in PPP mode, but networking will work just fine :)

nmpluta, 2024-06-09:
@bjarki-trackunit thanks for all the clarification, I'm closing the issue. I would probably go then with one of the quectel modules since they, according to the documentation, support CMUX functionality.

closed 2024-06-09 by nmpluta