zephyrproject-rtos/zephyr #73577: weak attribute on the z_sys_init_run_level(enum init_level level)

trampas, 2024-05-31:
I have a problem where I need to customize the initialization of a driver, specifically not initialize the driver until I need it in my code to keep the pins in a known state.  

I do know that the latest main branch has methods for delay initialization.  

However I was wondering if the the z_sys_init_run_level(enum init_level level) could be made a weak function? 
Then in my code I could redefine the function to do my custom initialization as needed. 

Basically this would allow developers who need customization of the initialization process to have complete control, without having to modify the zephyr source code.  For the 'average' user they would never know the difference.

henrikbrixandersen, 2024-06-02:
> I do know that the latest main branch has methods for delay initialization.

As discussed in Discord multiple times, this is the way your use-case will be supported from Zephyr v3.7 and forward. 

> However I was wondering if the the z_sys_init_run_level(enum init_level level) could be made a weak function?
> Then in my code I could redefine the function to do my custom initialization as needed.

In your downstream fork, you can do anything that solves the problem for you. If you are asking for this change to be implemented and backported to older releases, not likely.

gmarull, 2024-06-04:
This usecase sounds like something that you should maintain in a downstream fork. You can also have your custom out of tree driver where you can implement the behavior you want.

closed 2024-06-04 by gmarull