zephyrproject-rtos/zephyr #73580: entropy_stm32_rng_get_entropy() sometimes block forever

zagor, 2024-05-31:
**Describe the bug**
I have a rare but serious bug where entropy_stm32.c ends up in a state where no more RNG interrupts are serviced. The entropy pool then runs empty and `entropy_stm32_rng_get_entropy()` blocks forever, causing any thread calling `sys_rand32_get()` to hang.

I don't know what triggers this. But now I have found myself being able to trigger it fairly reliably within minutes. Unless, that is, I add too much console logging or run it through the debugger. Then it can take hours to reproduce, if possible at all. This behaviour suggests a race condition, but I can't find it.

After long sessions of printk() debugging, I have managed to narrow the problem down somewhat. But I need help understanding what I see.

What appears to happen is that, if `start_pool_filling()` is called and `already_filling` becomes true, `stm32_rng_isr()` is never executed again.

Here is a console dump when it occurs, which some printk() output that I've added. I'll explain in detail what we see:

```
«......»«.....»[01:06:47.478,000] <inf> http_request: Connecting HTTPS client
«......»«.....»[01:06:48.488,000] <inf> http_request: Connecting HTTPS client
«......»«.....»[01:06:49.490,000] <inf> http_request: Connecting HTTPS client
«......»«415 irq 1 cecs 0 seis 0 drdy 1 ie 1
[01:06:50.512,000] <inf> http_request: Connecting HTTPS client
415 irq 1 cecs 0 seis 0 drdy 1 ie 1
415 irq 1 cecs 0 seis 0 drdy 1 ie 1
600 count 1 irq 1 cecs 0 seis 0 drdy 1 ie 1
608
415 irq 1 cecs 0 seis 0 drdy 1 ie 1
600 count 0 irq 1 cecs 0 seis 0 drdy 1 ie 1
```

The first three lines show the application's normal activity. The program is creating a TLS socket using `zsock_socket(AF_INET6, SOCK_STREAM, IPPROTO_TLS_1_2)` and tries to `zsock_connect()` it to a remote server. (Note that the application log is output using deferred LOG_INF() so it is not perfectly synchronous with the printk output, but it doesn't matter here.)

The characters printed at the beginning of the lines are from printk() calls in entropy_stm32.c:
 - « is printed at the end of `acquire_rng()`
 - » is printed at the end of `release_rng()`
 - . is printed in `stm32_rng_isr()`, after the error check. (If there was an error, a "!" would be printed, which does not happen here.)

These three lines show the expected cyclic behaviour from the entropy driver:
 1. The rng is acquired,
 2. A number of ISR calls are executed,
 3. The rng is suspended.

In the fourth line, the anomaly starts. We see that the rng is acquired, but the isr is never called. The text starting with 415 is printed in `start_pool_filling()`, just after the line `if (unlikely(already_filling)) {`. The printk line looks like this:
```c
	printk("%d irq %d cecs %d seis %d drdy %d ie %d\n", __LINE__,
	       !!irq_is_enabled(IRQN),
	       LL_RNG_IsActiveFlag_CECS(entropy_stm32_rng_data.rng),
	       LL_RNG_IsActiveFlag_SEIS(entropy_stm32_rng_data.rng),
	       LL_RNG_IsActiveFlag_DRDY(entropy_stm32_rng_data.rng),
	       LL_RNG_IsEnabledIT(entropy_stm32_rng_data.rng));
```

The output of these "415" lines tell us that:
 - The interrupt is enabled in Zephyr
 - The CECS and SEIS error bits of the RNG are both 0, indicating no error
 - the DRDY bit is 1, indicating there is entropy data to be read
 - the IE bit is 1, indicating interrupts are enabled on the RNG device

So, everything looks fine, yet the isr is never called again. How? What am I missing?

We then see more "415" lines, indicating further calls to `start_pool_filling()`, each time aborted due to "already_filling".

The "600" line is printed in `entropy_stm32_rng_get_entropy()`, just after the line `if (bytes == 0U) {'. It looks like this:
```c
	int count = k_sem_count_get(&entropy_stm32_rng_data.sem_sync);
	printk("%d count %d irq %d cecs %d seis %d drdy %d ie %d\n", __LINE__,
	       count, !!irq_is_enabled(IRQN),
	       LL_RNG_IsActiveFlag_CECS(entropy_stm32_rng_data.rng),
	       LL_RNG_IsActiveFlag_SEIS(entropy_stm32_rng_data.rng),
	       LL_RNG_IsActiveFlag_DRDY(entropy_stm32_rng_data.rng),
	       LL_RNG_IsEnabledIT(entropy_stm32_rng_data.rng));
```
It contains the same information as the 415 lines, adding just the semaphore count.

The "608" line is printed just after the semaphore is taken. We can see that the semaphore is successfully taken when count is 1 but we never reach the 608 line when count is 0.

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr v0.16.6
 - Zephyr version: v3.6.0

**Additional context**
This happens on an STM32H742.

Some config settings that might be relevant:
```
CONFIG_ENTROPY_STM32_CLK_CHECK=y
# CONFIG_MBEDTLS_ENTROPY_ENABLED is not set
# CONFIG_MBEDTLS_ZEPHYR_ENTROPY is not set
CONFIG_ENTROPY_GENERATOR=y
# CONFIG_ENTROPY_LOG_LEVEL_OFF is not set
# CONFIG_ENTROPY_LOG_LEVEL_ERR is not set
# CONFIG_ENTROPY_LOG_LEVEL_WRN is not set
# CONFIG_ENTROPY_LOG_LEVEL_INF is not set
# CONFIG_ENTROPY_LOG_LEVEL_DBG is not set
CONFIG_ENTROPY_LOG_LEVEL_DEFAULT=y
CONFIG_ENTROPY_LOG_LEVEL=3
CONFIG_ENTROPY_INIT_PRIORITY=50
CONFIG_ENTROPY_STM32_RNG=y
CONFIG_ENTROPY_STM32_THR_POOL_SIZE=8
CONFIG_ENTROPY_STM32_THR_THRESHOLD=4
CONFIG_ENTROPY_STM32_ISR_POOL_SIZE=16
CONFIG_ENTROPY_STM32_ISR_THRESHOLD=12
CONFIG_ENTROPY_HAS_DRIVER=y
CONFIG_ENTROPY_DEVICE_RANDOM_GENERATOR=y
CONFIG_DT_HAS_ST_STM32_RNG_ENABLED=y
CONFIG_USE_STM32_LL_RNG=y
CONFIG_CSPRNG_ENABLED=y
# CONFIG_PM_DEVICE is not set
```

Possibly relevant excerpts from the generated device tree:
```
		rcc: rcc@58024400 {
			compatible = "st,stm32h7-rcc";
			#clock-cells = < 0x2 >;
			reg = < 0x58024400 0x400 >;
			clock-frequency = < 0xe4e1c00 >;
			clocks = < &pll >;
			d1cpre = < 0x1 >;
			hpre = < 0x2 >;
			d1ppre = < 0x2 >;
			d2ppre1 = < 0x2 >;
			d2ppre2 = < 0x2 >;
			d3ppre = < 0x2 >;
			phandle = < 0x2 >;
			rctl: reset-controller {
				compatible = "st,stm32-rcc-rctl";
				#reset-cells = < 0x1 >;
				phandle = < 0x4 >;
			};
		};
		rng: rng@48021800 {
			compatible = "st,stm32-rng";
			reg = < 0x48021800 0x400 >;
			clocks = < &rcc 0xdc 0x40 >;
			interrupts = < 0x50 0x0 >;
			status = "okay";
		};

```

I have tried to create a small isolated test application, but I can't get it to trigger this behaviour since I don't know what causes it. We have other boards with the same STM32H742 and the same Zephyr version where this does not happen either, so I am stuck doing minimal printk() modifications to this one application where I am able to repeat the issue.

(Ping @erwango and @taltenbach who from file history look like you might have some ideas what to look for.)

erwango, 2024-05-31:
Can you check https://github.com/zephyrproject-rtos/zephyr/pull/72882 ?

zagor, 2024-05-31:
> Can you check #72882 ?

I looked at it, but since I'm using neither #define in that patch it doesn't apply to my board.

zagor, 2024-05-31:
It occurred to me I don't print `LL_RNG_IsEnabled()` and that turns out to be interesting.

The problem occurs because RNGEN is suddenly 0 _after_ the call to `entropy_stm32_resume()`!

I am looking in RM0433 but I can't find that scenario described.

zagor, 2024-05-31:
Updated console output:
```
[00:04:19.167,000] <inf> http_request: Connecting HTTP client
[00:04:20.189,000] <inf> http_request: Connecting HTTP client
[00:04:21.190,000] <inf> http_request: Connecting HTTP client
[00:04:22.192,000] <inf> http_request: Connecting HTTP client
415 irq 1 cecs 0 seis 0 drdy 1 ie 1 en 0
[00:04:23.214,000] <inf> http_request: Connecting HTTP client
415 irq 1 cecs 0 seis 0 drdy 1 ie 1 en 0
415 irq 1 cecs 0 seis 0 drdy 1 ie 1 en 0
601 count 1
604
415 irq 1 cecs 0 seis 0 drdy 1 ie 1 en 0
601 count 0
```

Line 415 is now:
```c
		printk("%d irq %d cecs %d seis %d drdy %d ie %d en %d\n", __LINE__,
		       !!irq_is_enabled(IRQN),
		       LL_RNG_IsActiveFlag_CECS(entropy_stm32_rng_data.rng),
		       LL_RNG_IsActiveFlag_SEIS(entropy_stm32_rng_data.rng),
		       LL_RNG_IsActiveFlag_DRDY(entropy_stm32_rng_data.rng),
		       LL_RNG_IsEnabledIT(entropy_stm32_rng_data.rng),
		       LL_RNG_IsEnabled(entropy_stm32_rng_data.rng));
```

zagor, 2024-05-31:
I tried adding this to the end of `entropy_stm32_resume`:

```c
	while (!LL_RNG_IsEnabled(rng)) {
		printk("^");
		LL_RNG_Enable(rng);
	}
```

And I get the occasional output of ^ but only one each time. And it stops the hanging. But it's weird to have to set it twice...

Should the order of setting EN vs IE bits matter?

zagor, 2024-05-31:
If I swap the order of setting IE/EN, I get the exact same problem but for the IE bit:
```
420 irq 1 cecs 0 seis 0 drdy 1 ie 0 en 1
603 count 1
606
420 irq 1 cecs 0 seis 0 drdy 1 ie 0 en 1
603 count 0
```

taltenbach, 2024-06-01:
@zagor From what you describes, it seems the RNG peripheral is not ready immediately after `clock_control_on` has returned: I suspect the peripheral not to be properly clocked at the time the first write to one of its register is made in `entropy_stm32_resume`. This would cause the write to be ignored, as explained at the end of RM0433, section 8.5.10:

> When the bus interface clock is not active, read or write accesses to the peripheral registers 
are not supported. A read access will return invalid data. **A write access will be ignored and 
will not create any bus errors**

That might explain why this is impacting the IE bit instead of the EN bit when you swap the two writes.

According to RM0433, section 8.5.10, and assuming the D2 domain in which the RNG peripheral is located is in DRun mode, the recommended sequence to enable a peripheral's clock is:
1. Set the corresponding `PERxEN` bit in the `RCC_xxxxENR` register.
2. Read back the `RCC_xxxxENR` register.
3. Perform a dummy read operation to any register of the enabled peripheral.

From what I see, the implementation of the `clock_control_on` (see [here](https://github.com/zephyrproject-rtos/zephyr/blob/fe03e0c4bb3f8a658efe69b04c430d1a15474c6c/drivers/clock_control/clock_stm32_ll_h7.c#L359)) on STM32H7 MCUs only performs 1 and 2, which means the RNG peripheral might not be clocked yet at the time the `clock_control_on` returns.

Could you try adding a dummy read, by for example calling `LL_RNG_IsEnabled`, between `clock_control_on` and `LL_RNG_Enable` to see if that solves the issue?

erwango, 2024-06-03:
@zagor Can you checkout https://github.com/zephyrproject-rtos/zephyr/pull/70540 ?

zagor, 2024-06-04:
Problem identified as already fixed in main, closing this bug.

closed 2024-06-04 by zagor

taltenbach, 2024-06-04:
> It appears @taltenbach's analysis is spot on. I have tested both remedies, and they both make the problem go away. Since #70540 is already merged to main I will use that.
> 
> However, I wonder, is it formally correct?
> 
> RM0433 8.5.10 says _"Perform a dummy read operation **into a register of the enabled peripheral**"_. But as I read #70540 it always reads a register of the RCC (area 0x58024400) instead of the enabled peripheral (in my case RNG, area 0x48021800)?
> 
> Can the RCC be clocked faster than the peripheral so that the read of RCC completes before the peripheral is ready? Or is there something I am unaware of that guarantees the delay will always be long enough?

@zagor Glad that solved your issue. However, as you pointed out, I suspect the dummy read performed to the RCC register is not enough to guarantee the peripheral is properly clocked when the `clock_control_on` returns. RM0433 8.5.10 clearly indicates to _both_ read back the RCC register _and_ perform a dummy read to a register of the enabled peripheral. This might be especially important if the peripheral's clock frequency is low, in comparison to the CPU clock.

@erwango Are you aware of any specific reason not to follow strictly what the reference manual recommends on this point? If not, I guess it would probably be good to create another issue regarding this topic