zephyrproject-rtos/zephyr #73599: Blinky build fails for STM32C0116_DK due to parse error in device tree.

bobcagle, 2024-05-31:
**Describe the bug**
After a clean checkout of Zephyr [origin/main] (v3.6.0-4041-g4efc035209), I tried to build the blinky app for stm32c0116_dk. The build fails with a parse error on line 56 of the device tree. It looks like it has a problem with INPUT_KEY_SELECT. If I change that symbol to INPUT_KEY_0, it compiles fine. 

 - What target platform are you using?
 Linux container, ubuntu 22.04 lts. XFCE desktop.

 - What have you tried to diagnose or workaround this issue?
I tried deleting the line, but the binding is required, so I changed it to INPUT_KEY_0 and it worked.

**To Reproduce**
Steps to reproduce the behavior:
west build -b stm32c0116_dk samples/basic/blinky
You should fail with a parse error on stm32c0116_dk on line 56.

...zephyr/boards/st/stm32c0116_dk/stm32c0116_dk.dts:56 (column 19): parse error: expected number or parenthesized expression

**Expected behavior**
Build completes with no errors.

**Impact**
I cannot figure out how to avoid this with an overlay, so I presume I have to choose between modifying the zephyr source until this is fixed, or copy and rename the entire BSP to my own board directory.

**Logs and console output**

(.venv) rcagle@zdev:~/zephyrproject/zephyr$ lsb_release  -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
(.venv) rcagle@zdev:~/zephyrproject/zephyr$ west build -b stm32c0116_dk samples/basic/blinky
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/rcagle/zephyrproject/zephyr/samples/basic/blinky
-- CMake version: 3.22.1
-- Cache files will be written to: /home/rcagle/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/rcagle/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: stm32c0116_dk, qualifiers: stm32c011xx
-- Found host-tools: zephyr 0.16.5 (/home/rcagle/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (/home/rcagle/zephyr-sdk-0.16.5-1)
-- Found BOARD.dts: /home/rcagle/zephyrproject/zephyr/boards/st/stm32c0116_dk/stm32c0116_dk.dts
devicetree error: /home/rcagle/zephyrproject/zephyr/boards/st/stm32c0116_dk/stm32c0116_dk.dts:56 (column 19): parse error: expected number or parenthesized expression

**Environment (please complete the following information):**
Linux
Zephyr-sdk-0.16.5-1
v3.6.0-4041-g4efc035209

**Additional context**

FRASTM, 2024-06-03:
It looks like the select_key node of the **stm32c0116_dk** is using the 
`zephyr,code = <INPUT_KEY_SELECT>;`
instead of 
`zephyr,code = <INPUT_KEY_ENTER>;`

ajarmouni-st, 2024-06-03:
Yeah there is no `INPUT_KEY_SELECT` in `include/zephyr/dt-bindings/input/input-event-codes.h`.

closed 2024-06-03 by nashif