zephyrproject-rtos/zephyr #73601: Add SPI support for stm32c011 and stm32c031

bobcagle, 2024-06-01:
**Is your enhancement proposal related to a problem? Please describe.**
Not a problem, but I need spi for a project and the stm32c0 series processors have some useful properties if I can use them.

**Describe the solution you'd like**
I would like to be able to define my spi devices in the device tree under the typical spi1, spi2, ... hierarchy and have them connect to my custom spi-device drivers while using the stm32c0 series processors. Granted, spi would take up 20% of the pins on these chips, which may be why spi is not currently supported.

**Describe alternatives you've considered**
- I've considered adding this myself thinking it might just be an oversight, but I am not an expert at adding SoC enhancements in Zephyr and I do still have the time to choose another processor that is better supported.
- I've also considered using the STMicro HAL libraries directly inside Zephyr. I've done that in the past with other drivers, but would prefer to use the more zephyr-like interfaces. It appears stm32cubemx/stm32cubeide do support spi on these devices, so there should be a good chance this approach would work.

**Additional context**
I guess I would mainly like to understand if this omission was because the initial stm32c0 SoC contributors didn't need spi but the feature might still get added someday, or if there is some restriction in the chip that prevents spi from being effectively used.

ajarmouni-st, 2024-06-03:
Hi @bobcagle, I guess SPI wasn't added bcz the `tests/drivers/spi/spi_loopback/src/spi.c` test won't run with only 32KB of flash & 12KB of RAM that `nucleo_c031c6` has, & we only add peripherals that we are able to test.

You can add it for your use case like this:
```
diff --git a/dts/arm/st/c0/stm32c011.dtsi b/dts/arm/st/c0/stm32c011.dtsi
index a3196138478..e162aaf7737 100644
--- a/dts/arm/st/c0/stm32c011.dtsi
+++ b/dts/arm/st/c0/stm32c011.dtsi
@@ -9,5 +9,15 @@
 / {
        soc {
                compatible = "st,stm32c011", "st,stm32c0", "simple-bus";
+
+               spi1: spi@40013000 {
+                       compatible = "st,stm32-spi-fifo", "st,stm32-spi";
+                       #address-cells = <1>;
+                       #size-cells = <0>;
+                       reg = <0x40013000 0x400>;
+                       interrupts = <25 0>;
+                       clocks = <&rcc STM32_CLOCK_BUS_APB1_2 0x00001000>;
+                       status = "disabled";
+               };
        };
 };
```

bobcagle, 2024-06-04:
Thanks for the help. I put your definition in an overlay along with pinctrl entries and it compiled. Still trying to verify it works, but hopefully soon.

closed 2024-06-04 by bobcagle