zephyrproject-rtos/zephyr #73603: CI BT mesh tests fail to build due to MBEDTLS kconfig

aescolar, 2024-06-01:
**Describe the bug**
All BT mesh tests are currently failing in main due to kconfig issues

**To Reproduce**
Steps to reproduce the behavior:

0. If you don't have Babblesim, follow the steps in https://docs.zephyrproject.org/latest/boards/native/nrf_bsim/doc/nrf52_bsim.html#building-and-running

AND, either:

1. tests/bsim/bluetooth/mesh/compile.sh
2. See error

OR, for just one image

1. mkdir build ; cd build
1. cmake -GNinja -DBOARD=nrf52_bsim '-DOVERLAY_CONFIG=overlay_pst.conf;overlay_psa.conf' -DCONFIG_COMPILER_WARNINGS_AS_ERRORS=y ../tests/bsim/bluetooth/mesh
2. ninja
3. See error

**Expected behavior**
No kconfig warnings that result in build failures in CI

**Impact**
All BT related PRs blocked by failing CI

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/9327423640/job/25677281338?pr=73040#step:13:183

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 22.04
 - Toolchain: gcc 12
 - Commit SHA or Version used: main as of now c23132afdf33dc474bc814d4ccd576e76ee6618c

**Additional context**
Introduced by c1342b3aa9fabc39f83036b8adc68dd50e3e3c1a

aescolar, 2024-06-01:
CC @ceolin @valeriosetti

aescolar, 2024-06-01:
CC @PavelVPV @alxelax @cvinayak @Thalley @jori-nordic @jhedberg

hakehuang, 2024-06-02:
tests/crypto/mbedtls
tests/subsys/jwt
fails on all NXP platforms due to this

valeriosetti, 2024-06-03:
> CC @ceolin @valeriosetti

@tomi-font is on vacation. I will take care of this immediately

hakehuang, 2024-06-03:
@valeriosetti  this fix only works for bt cases, but for all other applications that uses the MBEDTLS_CIPHER_AES_ENABLED this issue still

valeriosetti, 2024-06-03:
> @valeriosetti this fix only works for bt cases, but for all other applications that uses the MBEDTLS_CIPHER_AES_ENABLED this issue still

They seem not to be tested by the PR's CI because #73627 has CI green. Can you point me to some failing test case so that I can have a look at it?

hakehuang, 2024-06-03:
> > @valeriosetti this fix only works for bt cases, but for all other applications that uses the MBEDTLS_CIPHER_AES_ENABLED this issue still
> 
> They seem not to be tested by the PR's CI because #73627 has CI green. Can you point me to some failing test case so that I can have a look at it?

tests/crypto/mbedtls try below command
```
scripts/twister -p mimxrt1060_evk -T tests/crypto/mbedtls  -s crypto.mbedtls
```

valeriosetti, 2024-06-03:
> tests/crypto/mbedtls try below command
> 
> ```
> scripts/twister -p mimxrt1060_evk -T tests/crypto/mbedtls  -s crypto.mbedtls
> ```

@hakehuang thanks for the tip! #73627 was updated accordingly. If there's any other failing case please let me know.

hakehuang, 2024-06-03:
and 

<html>

<body>

CASE_NAME | ZEPHYR_VERSION | RESULT | PLATFORM | TEST_PATH
-- | -- | -- | -- | --
kernel.memory_protection.stackprot | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/kernel/mem_protect/stackprot
net.conn_mgr.conn | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/conn_mgr_conn
net.iface | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/iface
net.mqtt.packet | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/lib/mqtt_packet
net.conn_mgr.nodad | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/conn_mgr_monitor
net.conn_mgr.dad | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/conn_mgr_monitor
net.socket.af_packet.ipproto_raw | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/af_packet_ipproto_raw
net.socket.udp | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/udp
net.socket.udp.preempt | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/udp
net.socket.udp.ipv6_fragment | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/udp
net.socket.udp.pktinfo | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/udp
net.socket.udp.ttl | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/udp
net.socket.get_name_info | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/getnameinfo
net.socket.socketpair | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/socketpair
net.socket.socketpair.newlib | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/socketpair
net.socket.socketpair.picolibc | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/socketpair
net.socket.socketpair.high_mem | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/socketpair
net.socket.reuseaddr_reuseport.userspace | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/reuseaddr_reuseport
net.socket.misc.userspace | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/misc
net.socket.misc.v4_mapping_to_v6_enabled | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/misc
net.socket.misc.v4_mapping_to_v6_disabled | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/misc
net.socket.mgmt | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/net_mgmt
net.socket.mgmt.preempt | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/net_mgmt
net.socket.select | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/select
net.socket.select.preempt | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/select
net.socket.get_addr_info | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/getaddrinfo
net.socket.get_addr_info.timeout | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/getaddrinfo
net.socket.af_packet | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/af_packet
net.socket.tcp | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/tcp
net.socket.tcp.preempt | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/socket/tcp
net.offloaded_netdev | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/offloaded_netdev
net.util | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/net/utils
crypto.mbedtls | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/crypto/mbedtls
crypto.rng | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/subsys/random/rng
crypto.rng.random_sw_systimer | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/subsys/random/rng
crypto.rng.random_hw_xoshiro | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/subsys/random/rng
crypto.rng.random_ctr_drbg | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/subsys/random/rng
libraries.encoding.jwt | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/subsys/jwt
testing.ztest.base.verbose_0_userspace | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/ztest/base
net.ethernet.build.default | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | tests/drivers/build_all/ethernet
sample.net.sockets.net_mgmt.usermode | v3.6.0-5117-gfe03e0c4bb3f | Build failure | mimxrt1060_evkb | samples/net/sockets/net_mgmt

</body>

</html>

aescolar, 2024-06-03:
@valeriosetti for info, in the weekly run we had an increase of approx. 600 failures, and most seem to be related to mbedtls (many are just the same test in different platforms)
https://github.com/zephyrproject-rtos/zephyr/runs/25711794410

aescolar, 2024-06-03:
> @aescolar is there a way to ask a full CI check on #73662 so that we are sure there is no corner case left?

A CI run like a weekly takes a really long time. So I don't have any good idea for this beyond running all tests in 1 or 2 platforms locally.

PR #73627 (2024-06-03): [fix] CI BT mesh tests fail to build due to MBEDTLS kconfig
PR #72475 disabled default enabling of most Mbed TLS features. This means that:

- `CONFIG_MBEDTLS_CIPHER_AES_ENABLED` needs to be manually enabled if AES support is required;
- `CONFIG_PSA_WANT_KEY_TYPE_RSA_KEY_PAIR_BASIC` does not need to be (almost) always added because there is no default RSA key-exchange enabled, so PSA can be built without RSA support.

Resolves #73603

As suggested on the original issue I tested this change with
- `tests/bsim/bluetooth/mesh/compile.sh` + `SEARCH_PATH="tests/bsim/bluetooth/mesh/" tests/bsim/run_ [... cut]
PR #73662 (2024-06-03): [fix] CI many tests fail to build due to MBEDTLS kconfig (part 2)
Following #72475 AES is no more enabled by default in Mbed TLS so it must be explicitly enabled when required. This commit is a follow up of #73627 in trying to add this dependency where required.

This PR was tested against the following failing test cases:
- https://github.com/zephyrproject-rtos/zephyr/issues/73603#issuecomment-2145122484
- https://github.com/zephyrproject-rtos/zephyr/issues/73603#issuecomment-2145281476
They all build successfully now.

Resolves #73603 ([part 2](https://github.com/zephyrproject-rtos/zephyr/is [... cut]
closed 2024-06-03 by nashif