zephyrproject-rtos/zephyr #73616: Commit 012c9a28470 is causing error: unknown type name 'wint_t'

williamtcdns, 2024-06-02:
Following commit is causing `zephyr/build/modules/picolibc/picolibc/include/sys/_types.h:218:5: error: unknown type name 'wint_t'`:
012c9a28470 libc/picolibc: Support Picolibc for toolchains without bundled Picolibc

We are seeing the error when building xtensa target using following command:
`west build -b xt-sim samples/hello_world`

nashif, 2024-06-03:
@williamtcdns  trying the above west command, the sample builds fine for me, can you share more information about your environment?

nashif, 2024-06-03:
are you build with gcc or the xtensa toolchain?

williamtcdns, 2024-06-03:
We use patches that allows us to build with recent xtensa toolchain which uses clang.

dcpleung, 2024-06-03:
Does your toolchain built with wide character support?

dcpleung, 2024-06-03:
Note that I have built with xt-clang (RI-2022.10) where picolibc was built from source, and there was no errors related to `wint_t`.

williamtcdns, 2024-06-03:
> Does your toolchain built with wide character support?

How can I tell whether toolchain was built with wide character support ?

dcpleung, 2024-06-04:
I am not sure. You will need to look at the configurations that are used to build that toolchain.

SzymonRichert, 2024-06-25:
@williamtcdns  any update?

williamtcdns, 2024-08-20:
The toolchain I am using (ie: RJ-2024.3) does have the type `wint_t`, so I will also assume that it has wide character support.
`wint_t` is defined in `stddef.h`, but to use it macro `__need_wint_t` must be defined.

@nashif are you using xt-clang or gcc ?

dcpleung, 2024-08-20:
xt-clang RI-2022.10. It builds for audio DSP platforms without issues.

williamtcdns, 2024-08-20:
Perhaps something different between `stddef.h` in RI-2022.10 and RJ-2024.3 ?

I now need to revert following 2 commits for a successful build:
git revert 89bf4b2ad9b
git revert 012c9a28470

Alternatively, instead of reverting above 2 commits, if I use `default n` with `config PICOLIBC_SUPPORTED` in `zephyr/lib/libc/Kconfig` , I also get a successful build.

Does `config PICOLIBC_SUPPORTED` has to be `y` by default ? Why not use it only if needed ?

dcpleung, 2024-08-21:
Picolibc can be built from source so it is supported. The audio DSP platforms are actually doing that with xt-clang.

williamtcdns, 2024-08-28:
I am now using RI-2022.10 and I am still seeing this failure.

@dcpleung I am building the config `sample_controller` using `west build -b xt-sim samples/hello_world`.

Please, could you also try on whether it is passing for you ?

dcpleung, 2024-08-28:
There is something wrong with the RJ-2024.3 toolchain. I got a hold of RJ-2024.3 and it was giving the same error. However, if you look at `sys/_types.h`, it does `#define __need_wint_t` before including `stddef.h`. I think the issue here is that the `sample_controller` comes with its own `stddef.h` (under `c99/stddef.h`), and it is being used instead of the LLVM one that comes with the main tool package (`clang-15.0.7/include/stddef.h` which should really be used with `xt-clang`).

FYI, I tried building with host installed clang 15.0.7 in Ubuntu 24.04 using `qemu_x86`. It built without any issue and it also required building picolibc from source. The audio DSP toolchains only target `xt-clang` and not `xcc`, and I guess that's why they don't have `stddef.h` in the core definition directories. So they use the LLVM's `stddef.h` and have no issue building picolibc from source. So from my perspective, it is an issue with the Xtensa toolchain due to incorrect header file being used and has nothing to do with Zephyr or picolibc.

williamtcdns, 2024-08-28:
But I am also seeing the same failure when using RI-2022.10  , which is the release you are using to successfully build for audio DSP platforms.
Please note that I am seeing this failure building `sample/hello_world` using config `sample_controller`; ie: `west build -b xt-sim samples/hello_world`.

dcpleung, 2024-08-29:
As mentioned above in the second paragraph, the audio DSP core packages for RI-2022.10 only target `xt-clang` and not `xcc`. They don't have `c99/stddef.h` in their core packages. The only `stddef.h` they are using are from the main tool directory (`clang-15.0.7/include/stddef.h`).

Let me rephrase it a bit:
* The `sample_controller` core package contains its own `stddef.h` (as `c99/stddef.h`)...
  * This C99 `stddef.h` does not have the correct typedef for `wint_t`.
  * Actually it only has typedef for `wint_t` when building under cygwin.
* This problematic `stddef.h` is erroneously being used for builds using `xt-clang`...
  * Where the LLVM's `stddef.h` should be used instead (and it has the correct typedef for `wint_t`).
* This points to incorrect include paths being used internally in the toolchain.
  * The Zephyr build system does not know where the `c99/stddef.h` file is so it cannot add that incorrect include path during build.
  * The incorrect include path is being inserted by the Xtensa toolchain.

You can do `VERBOSE=1 west build ...` and grab any command line compiling a source file, and add `-E` after `<...>/xt-clang` to force it into preprocessing mode. You can then take a look at the output file and see which `stddef.h` is included. This will allow you to look at the `stddef.h` to see if `wint_t` has been correctly typedef-ed.

williamtcdns, 2024-09-05:
Thank you @dcpleung , we are now trying to fix the issues with the header files.

dcpleung, 2024-09-17:
Closing it for now as this is not an issue in Zephyr. Re-open if needed.

closed 2024-09-17 by dcpleung