zephyrproject-rtos/zephyr #73621: bluetooth: nordic: fault after 6.5 day uptime with radio not ready assert

thedjnK, 2024-06-02:
**Describe the bug**
The following code is used: https://github.com/thedjnK/central_esp which has an nrf52840 (nrf52840 dongle) run in central bluetooth mode, with a single sensor and PWM instance, transmit sensor readings from the local sensor and bluetooth sensors over USB CDC. After 6.5 days of uptime, the device faults, after adding a dedicated UART (not USB CDC) and enabling logging on this UART, the following was output:
```
[13:20:22.868,103] <err> abe: [NOTIFICATION] from 0 data 0x2000b43c length 2
[13:20:22.868,164] <err> abe: temp = 20.150000c
[13:20:22.868,652] <err> abe: [NOTIFICATION] from 0 data 0x2000b43c length 2
[13:20:22.868,652] <err> abe: hum = 56.340000%
[13:20:22.869,171] <err> abe: [NOTIFICATION] from 0 data 0x2000b43c length 1
[13:20:22.869,201] <err> abe: battery = 81%
[13:21:56.515,228] <err> abe: [NOTIFICATION] from 1 data 0x2000b43c length 2
[13:21:56.515,258] <err> abe: temp = 17.770000c
[13:21:56.615,234] <err> abe: [NOTIFICATION] from 1 data 0x2000b43c length 2
[13:21:56.615,234] <err> abe: hum = 77.690000%
[13:21:56.765,228] <err> abe: [NOTIFICATION] from 1 data 0x2000b43c length 1
[13:21:56.765,228] <err> abe: battery = 100%
[13:22:03.466,644] <err> abe: [NOTIFICATION] from 2 data 0x2000b43c length 2
[13:22:03.466,674] <err> abe: temp = 17.760000c
[13:22:03.566,650] <err> abe: [NOTIFICATION] from 2 data 0x2000b43c length 2
[13:22:03.566,650] <err> abe: hum = 74.170000%
[13:22:03.666,625] <err> abe: [NOTIFICATION] from 2 data 0x2000b43c length 1
[13:22:03.666,656] <err> abe: battery = 62%
[13:22:52.868,103] <err> abe: [NOTIFICATION] from 0 data 0x2000b43c length 2
[13:22:52.868,164] <err> abe: temp = 20.160000c
[13:22:52.868,652] <err> abe: [NOTIFICATION] from 0 data 0x2000b43c length 2
[13:22:52.868,652] <err> abe: hum = 56.290000%
[13:22:52.869,171] <err> abe: [NOTIFICATION] from 0 data 0x2000b43c length 1
[13:22:52.869,201] <err> abe: battery = 82%
ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic2
[13:23:31.638,427] <err> os: r0/a1:  0x00000003  r1/a2:  0x20006060  r2/a3:  0x40011000
[13:23:31.638,427] <err> os: r3/a4:  0x00000003 r12/ip:  0x00000000 r14/lr:  0x000239cb
[13:23:31.638,458] <err> os:  xpsr:  0x01000011
[13:23:31.638,458] <err> os: s[ 0]:  0x0003020c  s[ 1]:  0x200056e8  s[ 2]:  0x00034434  s[ 3]: 0
[13:23:31.638,458] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000005  s[ 6]:  0xdfff8301  s[ 7]: 7
[13:23:31.638,458] <err> os: s[ 8]:  0x0000001e  s[ 9]:  0x00021fa3  s[10]:  0x20009c00  s[11]: d
[13:23:31.638,458] <err> os: s[12]:  0x1ab12a43  s[13]:  0xf73395d0  s[14]:  0x739422d4  s[15]: 7
[13:23:31.638,488] <err> os: fpscr:  0x8447852e
[13:23:31.638,488] <err> os: Faulting instruction address (r15/pc): 0x000239da
[13:23:31.638,519] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[13:23:31.638,519] <err> os: Fault during interrupt handling

[13:23:31.638,549] <err> os: Current thread: 0x200027b0 (unknown)
[13:23:31.746,917] <err> os: Halting system
```

The 2 lines:
```
addr2line -e zephyr/zephyr.elf 0x000239da                              /tmp/central/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:342 (discriminator 1)

addr2line -e zephyr/zephyr.elf 0x200027b0
/tmp/central/zephyr/subsys/shell/backends/shell_uart.c:519
```

**Expected behavior**
The device to not fault

**Impact**
Showstopper

**Environment (please complete the following information):**
 - Commit SHA or Version used: ea29ef36bbedeacd171199d6feea5e699fa64447

cvinayak, 2024-06-03:
Could you try use lower priorities for other SoC peripherals used in your application by having the below

`#define NRF_DEFAULT_IRQ_PRIORITY 5`

inside your application folder in `dts/arm/nordic/override.dtsi`  file.

cvinayak, 2024-06-03:
I see the below in your prj.conf, disabling interrupts will cause the assertion reported in this issue.

```
CONFIG_DHT_LOCK_IRQS=y
```

Also, use of the below in your application indicates that the Controller Scheduling already has seen latencies.

```
CONFIG_BT_CTLR_ASSERT_OVERHEAD_START=n
```

Please the github issue if you do not see the assertion when not using interrupt disable in your application.

cvinayak, 2024-06-03:
Alternatively, you can enable `CONFIG_BT_CTLR_ZLI=y`  and its dependencies.

thedjnK, 2024-06-03:
> I see the below in your prj.conf, disabling interrupts will cause the assertion reported in this issue.
> 
> ```
> CONFIG_DHT_LOCK_IRQS=y
> ```
> 
> Also, use of the below in your application indicates that the Controller Scheduling already has seen latencies.
> 
> ```
> CONFIG_BT_CTLR_ASSERT_OVERHEAD_START=n
> ```
> 
> Please close the github issue if you do not see the assertion when not using interrupt disable in your application.

The sensor is pretty terrible in that it requires a long time to read it using GPIO read functions without other things like the radio from interrupting it, so how can that be done in a way without it causing a fault? If radio messages are delayed or missed that is not a problem, but the fault is

cvinayak, 2024-06-03:
You will have to use the sensor in coexistence with the Controller, take a look at how flash driver and gpio co-ex is implemented in coexistence with the Controller.

cvinayak, 2024-06-15:
lowering issue priority, as it is more a feature request to have sensors with hard real-time requirements to co-exist with BLE Controller.

cvinayak, 2024-08-06:
@thedjnK Could we convert this to a sensor driver enhancement to co-exist with the Bluetooth Controller?

thedjnK, 2024-08-06:
Will close this issue, anyone that gets this sensor soon realises what a terrible mistake they have made

closed 2024-08-06 by thedjnK