zephyrproject-rtos/zephyr #73659: kernel: timer: cycle64: kernel.timer.cycle64 reports timeout on nrf54l15pdk/nrf54l15/cpuapp

katgiadla, 2024-06-03:
**Describe the bug**
Test `tests/kernel/timer/cycle64/kernel.timer.cycle64` reports timout.

Observed for `nrf54l15pdk/nrf54l15/cpuapp`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15pdk/nrf54l15/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/kernel/timer/cycle64 -p nrf9160dk/nrf9160/ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-5099-g127cb9edb6c6 ***
Running TESTSUITE cycle64_tests
===================================================================
START - test_32bit_wrap_around
32-bit wrap-around should occur every 4294s
[ddd:hh:mm:ss.0ms]
[000:00:03:27.029]: cycle: 000000000c5706ce
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: `v3.6.0-5099-g127cb9edb6c6`

nordic-piks, 2024-08-30:
nrf_triage_done

nordic-piks, 2024-08-30:
https://nordicsemi.atlassian.net/browse/NRFX-5564

closed 2024-11-25 by katgiadla