zephyrproject-rtos/zephyr #73667: samples: tfm_integration: psa_crypto: sample.psa_crypto reports timeout

katgiadla, 2024-06-03:
**Describe the bug**
Sample `samples/tfm_integration/psa_crypto/sample.psa_crypto`

Observed for:
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk/nrf9160/ns`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf9160dk/nrf9160` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/tfm_integration/psa_crypto -p nrf9160dk/nrf9160/ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-5099-g127cb9edb6c6 ***
[00:00:00.424,499] <inf> app: app_cfg: Creating new config file with UID 0x55CFDA7A
[00:00:02.256,561] <inf> app: att: System IAT size is: 550 bytes.
[00:00:02.256,561] <inf> app: att: Requesting IAT with 64 byte challenge.
[00:00:02.553,344] <inf> app: att: IAT data received: 550 bytes.
...
[00:00:02.818,237] <inf> app: Persisting SECP256R1 key as #1
[00:00:02.824,371] <err> app: Generic error
[00:00:02.824,401] <err> app: Function: 'crp_gen_key_secp256r1'
[00:00:02.824,432] <err> app: Failed to generate key.
[00:00:02.841,552] <inf> app: Calculating SHA-256 hash of value
...
[00:00:02.871,582] <err> app: Generic error
[00:00:02.871,612] <err> app: Function: 'crp_hash_payload'
[00:00:02.871,643] <err> app: Failed to setup hash op.
[00:00:02.888,488] <inf> app: Signing SHA-256 hash
[00:00:02.893,798] <err> app: Generic error
[00:00:02.893,829] <err> app: Function: 'crp_sign_hash'
[00:00:02.893,829] <err> app: Failed to open persistent key #1
[00:00:02.911,041] <inf> app: Verifying signature for SHA-256 hash
[00:00:02.917,724] <err> app: Generic error
[00:00:02.917,755] <err> app: Function: 'crp_verify_sign'
[00:00:02.917,755] <err> app: Failed to open persistent key #1
[00:00:02.935,241] <err> app: Generic error
[00:00:02.935,272] <err> app: Function: 'crp_dest_key'
[00:00:02.935,272] <err> app: Failed to open persistent key #1
[00:00:02.952,514] <err> app: Generic error
[00:00:02.952,545] <err> app: Function: 'crp_test_rng'
[00:00:02.952,575] <inf> app: Generating 256 bytes of random data.
...
[00:00:03.079,101] <inf> app: Initialising PSA crypto
[00:00:03.084,533] <inf> app: PSA crypto init completed
[00:00:03.090,148] <inf> app: Persisting SECP256R1 key as #1
[00:00:03.096,313] <err> app: Generic error
[00:00:03.096,343] <err> app: Function: 'crp_gen_key_secp256r1'
[00:00:03.096,374] <err> app: Failed to generate key.
[00:00:03.113,464] <err> app: Generic error
[00:00:03.113,494] <err> app: Function: 'crp_generate_csr'
[00:00:03.113,525] <err> app: Failed to create persistent key #1
[00:00:03.131,134] <inf> app: Done.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: `v3.6.0-5099-g127cb9edb6c6`

Vge0rge, 2024-06-25:
The sample seems to work for me when run it standalone. When I run it with twister it fails for me as well though.

tomi-font, 2024-06-26:
Runs just fine when adding `--west-flash=--erase` to the `twister` command. The sample expects the device to be in a blank state.

closed 2024-06-26 by tomi-font